zephyrproject-rtos/zephyr #52561: fsl_flexcan missing flags

dipakgmx, 2022-11-25:
**Describe the bug**
The CAN FD driver provided in the fsl_flexcan file seems to have a few flags of `flexcan_fd_frame_t` not being set or are set to default values.

For example, [here](https://github.com/zephyrproject-rtos/hal_nxp/blob/master/mcux/mcux-sdk/drivers/flexcan/fsl_flexcan.c#L2897) the `pTxFrame->edl` uses a default value of 1, which implies a CAN FD frame. This wouldn't allow one to send CAN 2.0 frames.

Further on, on the reception side, the EDL or the BRS flags aren't being read. which is necessary for mapping the flags to the Zephyr `can_frame`.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Cannot send CAN 2.0 frames.
On the receiver side, one is not able to detect if the frame is a FD frame or if Bit-Rate-Switch were active.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: zephyr-sdk-0.15.0
 - Zephyr version: 3.2.99

**Additional context**
Tested the following patch and it seems to fix the issue.

[image]

henrikbrixandersen, 2022-11-27:
> For example, [here](https://github.com/zephyrproject-rtos/hal_nxp/blob/master/mcux/mcux-sdk/drivers/flexcan/fsl_flexcan.c#L2897) the `pTxFrame->edl` uses a default value of 1, which implies a CAN FD frame. This wouldn't allow one to send CAN 2.0 frames.

The function is called `FLEXCAN_WriteFDTxMb()`. For transmitting classic CAN frames, you should use `FLEXCAN_WriteTxMb()`?

dipakgmx, 2022-11-27:
> The function is called `FLEXCAN_WriteFDTxMb()`. For transmitting classic CAN frames, you should use `FLEXCAN_WriteTxMb()`?

Am uncertain if the `FLEXCAN_WriteTxMb()` would be the right option. Looking into the implementation of `FLEXCAN_WriteFDTxMb()`, it   uses the `FLEXCAN_GetFDMailboxOffset` to figure out where the data needs to placed, whereas the former uses another approach in figuring out where the data needs to be placed. When CAN FD is configured, the MailBoxes are also configured differently, and hence the location where I suppose the data needs to be placed.

henrikbrixandersen, 2022-12-05:
Removing the bug label as this NXP MCUX HAL code is not even called in Zephyr (yet).

closed 2023-01-03 by dipakgmx