zephyrproject-rtos/zephyr #52575: Kconfig: excessive `select` usage

JordanYates, 2022-11-27:
**Describe the bug**

Several subsytems (mcumgr, hawkbit) have defaulted to using `select` for all their dependencies.

https://github.com/zephyrproject-rtos/zephyr/blob/5a5e15fc958a0ce299809286a2dc9826d6e4124c/subsys/mgmt/hawkbit/Kconfig#L4-L21

https://github.com/zephyrproject-rtos/zephyr/blob/5a5e15fc958a0ce299809286a2dc9826d6e4124c/subsys/mgmt/mcumgr/Kconfig#L5-L9

https://github.com/zephyrproject-rtos/zephyr/blob/5a5e15fc958a0ce299809286a2dc9826d6e4124c/subsys/mgmt/mcumgr/transport/Kconfig.bluetooth#L5-L9

This goes against practically every recommendation in https://docs.zephyrproject.org/latest/build/kconfig/tips.html#select-recommendations.
* The symbols have prompts
* The symbols have dependencies
* There is 0 expectation that the dependencies will ever stay in sync

Additionally, no effort has been made to address https://docs.zephyrproject.org/latest/build/kconfig/tips.html#select-pitfalls.
In general, mixing `depends on` and `select` for the same symbol commonly causes dependency loops.

This is damaging because these `select` statements can corrupt otherwise perfectly valid configuration setups that don't reference the offending symbol at all.
```
config MY_NVS_ABSTRACTION
	bool "NVS abstraction"
	depends on NVS

config MY_NETWORKING_FEATURE
	bool "Networking feature"
	depends on MY_NVS_ABSTRACTION

# Turn on networking if my app needs some feature
config NETWORKING
	default y if MY_NETWORKING_FEATURE

source "Kconfig.zephyr"
```
This results in the dependency loop shown below (Note the trip through config HAWKBIT):
```
/home/jordan/code/zephyr/zephyr/scripts/kconfig/kconfig.py: Dependency loop
===============

MY_NVS_ABSTRACTION (defined at /home/jordan/code/zephyr/zephyr/tests/misc/test_build/Kconfig:10), with definition...

config MY_NVS_ABSTRACTION
        bool "NVS abstraction"
        depends on NVS

...depends on NVS (defined at subsys/fs/nvs/Kconfig:6), with definition...

config NVS
        bool "Non-volatile Storage"
        select CRC
        help
          Enable support of Non-volatile Storage.

(select-related dependencies: HAWKBIT)

(imply-related dependencies: HAWKBIT)

...depends on NET_L2_OPENTHREAD (defined at subsys/net/l2/openthread/Kconfig:12), with definition...

menuconfig NET_L2_OPENTHREAD
        bool "OpenThread L2"
        select NET_L2_PHY_IEEE802154
        select OPENTHREAD
        select SETTINGS if FLASH
        select OPENTHREAD_SETTINGS_RAM if !FLASH
        select CPLUSPLUS
        select REBOOT
        select ENTROPY_GENERATOR
        imply FLASH
        imply FLASH_PAGE_LAYOUT
        imply FLASH_MAP
        imply MPU_ALLOW_FLASH_WRITE
        imply NVS
        imply NET_UDP
        imply NET_IPV6
        imply NET_CONFIG_NEED_IPV6
        depends on NETWORKING && NETWORKING

...depends on NETWORKING (defined at /home/jordan/code/zephyr/zephyr/tests/misc/test_build/Kconfig:19, subsys/net/Kconfig:62), with definition...

config NETWORKING
        bool
        default y if MY_NETWORKING_FEATURE

config NETWORKING
        bool "Link layer and networking (including IP)"
        select NET_BUF
        select POLL
        select ENTROPY_GENERATOR
        help
          This option enables generic link layer and networking support
          and is the basis for higher level protocols (e.g. IP protocols).

(select-related dependencies: (MCUMGR_SMP_UDP && MCUMGR) || HAWKBIT || UPDATEHUB)

...depends on MY_NETWORKING_FEATURE (defined at /home/jordan/code/zephyr/zephyr/tests/misc/test_build/Kconfig:14), with definition...

config MY_NETWORKING_FEATURE
        bool "Networking feature"
        depends on MY_NVS_ABSTRACTION

...depends again on MY_NVS_ABSTRACTION (defined at /home/jordan/code/zephyr/zephyr/tests/misc/test_build/Kconfig:10)
CMake Error at /home/jordan/code/zephyr/zephyr/cmake/modules/kconfig.cmake:328 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/jordan/code/zephyr/zephyr/cmake/modules/zephyr_default.cmake:108 (include)
  /home/jordan/code/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/jordan/code/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:4 (find_package)
```

**Impact**

Valid configurations are unable to be expressed in Kconfig. Working around these issues requires patching Zephyr.

tejlmand, 2023-01-11:
thanks for raising this. It's something I have often tried to point out but few people care.
Unfortunately at the stage where we are now, it becomes a huge task to clean up.

ycsin, 2024-02-20:
@JordanYates I wonder if #55297 addressed this issue?

nordicjm, 2026-01-06:
Seems both MCUmgr and hawkbit have long since been updated to use `depends on` rather than `select`

closed 2026-01-06 by nordicjm