zephyrproject-rtos/zephyr #52585: PDM event handler shouldn't stop driver on allocation failure

jefffhaynes, 2022-11-28:
I've been struggling to get a USB microphone implementation working, and it strikes me that some of the DMIC driver behavior could probably be better.

I think there are a few weird things, but probably the primary issue is that the PDM "pool" has to be drained faster than it is replenished or else the microphone will simply stop working.  Conversely, if the PDM driver is only activated when the system is ready to record, samples will be lost.

It seems like this behavior occurs because rather than go with the event-driven model implemented at the nrf layer, the event is intercepted but then can no longer be optionally discarded and it becomes a do or die situation.

Consider the following implementation for a USB mic:

-- Start PDM driver (start receiving PDM interrupts)
-- Time passes...
-- Host connects over USB and starts requesting samples
-- Start reading from PDM pool and providing samples to host

The last step is impossible without adding yet another layer of buffering because by this time the PDM event handler has given up and gone home:

https://github.com/zephyrproject-rtos/zephyr/blob/7ac69dc4ca95dcf7375a21c24aa4af22b34d6fa7/drivers/audio/dmic_nrfx_pdm.c#L61

Although not a perfect fix, probably better behavior would be to simply drop the samples.  There is no real recourse for the consumer when the driver is simply stopped, and samples will be lost either way.

anangl, 2023-04-12:
> Consider the following implementation for a USB mic:
>
> -- Start PDM driver (start receiving PDM interrupts)
> -- Time passes...
> -- Host connects over USB and starts requesting samples
> -- Start reading [...]

@jefffhaynes I don't understand this. What is the point in starting the stream when the samples are not needed yet? Couldn't this be done when the connection over USB appears? And why cannot the application drop the samples if it does not need them (although it started the stream), why you want the driver to do it instead?

jefffhaynes, 2023-04-13:
I need to go back and remind myself now but I think the issue is that this implementation _forces_ the application to keep up or else be punished by having the driver shutdown. Why should it work this way?  What is the benefit of stopping the driver when it can simply drop packets?  

Bear in mind that if the driver stops, the application now has no recourse so this is actively preventing any situation where dropped packets may be ok.

anangl, 2023-04-13:
> I need to go back and remind myself now but I think the issue is that this implementation _forces_ the application to keep up or else be punished by having the driver shutdown. Why should it work this way? What is the benefit of stopping the driver when it can simply drop packets?

Well, that's how I understand the current dmic API - after the application starts the stream, it is responsible for handling the produced audio data. If the driver just dropped some packets, the application would not be aware of that - the driver has no way to signal this.

> Bear in mind that if the driver stops, the application now has no recourse so this is actively preventing any situation where dropped packets may be ok.

The application can start the stream again. Or am I missing something?

I still don't understand why to start the stream if the samples are not needed.

carlescufi, 2023-04-21:
@jefffhaynes please close this issue if @anangl's comments have answered your questions.

jefffhaynes, 2023-04-21:
Well it may just be a difference in opinion about driver design. The way the driver is currently written creates an inherent race condition where the caller has very little time between stream start and stream processing. If you try to write a USB mic implementation, for example, this basically forces a second buffering layer, making the buffering in the driver essentially useless. 

I think a mechanism for buffering but allowing for packets to be dropped would be more effective than simply shutting down the driver. Alternatively I would remove buffering from the driver altogether, since it isn't doing much other than adding overhead. 

I would recommend writing some real-world use cases out (e.g. a USB mic) and I think you'll see what I mean.

carlescufi, 2023-05-10:
@jefffhaynes we've discussed this internally and we don't currently see the exact usecase you are trying to cover. However, we would like to improve the driver if that is indeed reasonable, so we'd ask you if you could provide a Pull Request with a draft of what you'd like to see, so we can understand better and then make a decision. As a side note, we are currently working on a new implementation of the USB Audio class in the context of the "next" USB stack that is in the tree. That should also give us the opportunity to revisit this.

jefffhaynes, 2023-05-10:
I appreciate you guys looking at it. I am a little confused by the use case comment though. Would it be possible for you to sketch out how you picture a USB mic implementation working that does _not_ involve a race condition?  If the answer is to wait for USB rework, that's fine.

carlescufi, 2023-05-17:
> I appreciate you guys looking at it. I am a little confused by the use case comment though. Would it be possible for you to sketch out how you picture a USB mic implementation working that does _not_ involve a race condition? If the answer is to wait for USB rework, that's fine.

Since it's not quite clear neither to us nor to you exactly what to fix here, I will reassign this to @tmon-nordic who is working on the audio class implementation for the device_next USB stack. As part of this work we can provide an example of PDM to USB capture.

tmon-nordic, 2024-02-05:
> I would recommend writing some real-world use cases out (e.g. a USB mic) and I think you'll see what I mean.

I don't quite observe the issue you mention, but some different maybe even more complex one. While for the I2S audio sample, it is possible to receive ±1 sample  packets based on the feedback to synchronize I2S FRAMESTART against USB SOF (see https://github.com/zephyrproject-rtos/zephyr/tree/e90613efb2ba84e99295fe88d2fe8324316b4271/samples/subsys/usb/uac2_explicit_feedback/src for implementation), I see no such possibility with dmic API.

The way new USB Audio is implemented makes the I2S buffering the only buffering - USB does not copy the samples. The implicit feedback sample (that actually includes microphone as the feedback) is still work-in-progress, but relies on the same underlying principle - controlling number of samples per block by varying the number of data sent.

With dmic API the data will always come at fixed blocks and thus there'll be either underrun or overrun as the sample clock is not locked to USB SOF. I currently don't have a clear solution in mind for the issue.

jefffhaynes, 2024-02-05:
I was able to get a stable USB mic implementation but it required using the NRFX PDM "raw" driver vs the DMIC driver, which I don't believe will support the use case because of the fast-fail behavior I mentioned.  You essentially have to have the PDM driver buffering into a queue or bucket of some sort that can drop frames if needed.  This allows you to sync with USB and start/stop as needed without the PDM source shutting down.  In my opinion, the DMIC implementation is not very useful as written.

tmon-nordic, 2024-02-05:
> You essentially have to have the PDM driver buffering into a queue or bucket of some sort that can drop frames if needed.

But this only works if the dmic is providing data faster than USB SOFs, that is when on standard compliant asynchronous endpoint implementation you would get +1 sample packets every now and then. When you get the dmic providing data slower, i.e. when in compliant implementation you would get -1 sample packets every now and then - then you cannot "drop frames if needed" but rather have to come up with something.

The problem I would rather prefer to solve is not "drop frames if needed" or "artificially create frames if needed" but rather to provide appropriate number of samples within frame so the two data rates match over the long term.

jefffhaynes, 2024-02-05:
I believe I am running synchronously and not dropping frames (as far as I know) during normal operation.  I only drop frames during startup and shutdown of the USB channel, but that is where it matters.

tmon-nordic, 2024-02-05:
> I believe I am running synchronously and not dropping frames (as far as I know) during normal operation. I only drop frames during startup and shutdown of the USB channel, but that is where it matters.

How are you adjusting the sampling clock frequency then?

jefffhaynes, 2024-02-05:
I suppose I'm not.  Are you saying that mismatches between the USB clock and the PDM clock will inevitably cause dropped frames?  You're probably right, although for my purposes it doesn't especially matter.  I suppose if you were making a production grade USB mic you would want to adjust the clock.  Or does running asynchronously solve this but using the PDM clock as reference?  I'll admit I'm out of my depth on the USB modes for this.

tmon-nordic, 2024-02-05:
> I suppose I'm not. Are you saying that mismatches between the USB clock and the PDM clock will inevitably cause dropped frames?

Yes, but depending on which clock is faster you'll either have too many (and have to drop it) or too few (and have to somehow fill the gap).

> You're probably right, although for my purposes it doesn't especially matter. I suppose if you were making a production grade USB mic you would want to adjust the clock.

Only if you want to use USB Synchronous isochronous endpoint type.

> Or does running asynchronously solve this by using the PDM clock as reference? I'll admit I'm out of my depth on the USB modes for this.

USB Asynchronous isochronous endpoints solve this by allowing ±1 sample packets. For example if you have 48 kHz sampling rate, the nominal is 48 samples per frame, but asynchronous can send 47, 48 or 49 samples per frame. The non-nominal packets are in minority, but these are essential for preventing the "timing error accumulation" that leads to full frame buffer overrun/underrun.

jefffhaynes, 2024-02-05:
Interesting, I'll have to experiment with that.  I would think it should work without too much modification?

tmon-nordic, 2024-02-05:
> Interesting, I'll have to experiment with that. I would think it should work without too much modification?

In the old USB stack that's probably impossible. With the new USB stack and new USB Audio 2 implementation (where you describe the UAC2 microphone topology in devicetree), it is up to application to come up with the data at varying length (nominal, +1 or -1) on every SOF.

jefffhaynes, 2024-02-05:
Ok, I understand now.  Thanks for the info, I'll try to spend some time with it.

nordic-piks, 2024-08-29:
nrf_triage_done

closed 2025-05-24 by github-actions[bot]