zephyrproject-rtos/zephyr #52586: doc version selection change

vChavezB, 2022-11-28:
**Is your enhancement proposal related to a problem? Please describe.**
Whenever you change the version of the zephyr docs the string "latest" is always displayed. Is this an intended behavior or should it change to the current selected version?

[image]

**Describe the solution you'd like**
To display the version of the docs I selected.

**Describe alternatives you've considered**

Alternative could be adding the version information such as 

```js
{% set prefix = current_version if is_release else "latest" %}
v: {{ prefix }}
```

Before this line

https://github.com/zephyrproject-rtos/zephyr/blob/v3.2.0/doc/_templates/zversions.html#L5

erwango, 2022-11-28:
I personally find that `latest` is quite explicit. Don't see why it should be changed.
Alternative is `3.2.99` which will confuse most new comers.

vChavezB, 2022-11-28:
Yes the latest tag is explicit, but once you change the version and  go to https://docs.zephyrproject.org/SPECIFIC_VERSION  the v:latest tag is still shown. 

What I propose is that if I click v3.20 the label "v:latest" changes to v:3.20. Thats why I asked if its an intended behavior that it always shows v:latest independent of navigation directory in the docs website.

BR
Victor

vChavezB, 2022-11-28:
Yes the latest tag is explicit, but once you change the version and  go to https://docs.zephyrproject.org/SPECIFIC_VERSION  the v:latest tag is still shown. 

What I propose is that if I click v3.20 the label "v:latest" changes to v:3.20. Thats why I asked if its an intended behavior that it always shows v:latest independent of navigation directory in the docs website.

Btw: my proposal does not use 3.2.99 for latest, it uses "latest" 
see

`{% set prefix = current_version if is_release else "latest" %}`

BR
Victor

vChavezB, 2024-02-20:
Just an additional example provided from the read the docs project itself, of how the displayed version matches

https://example-sphinx-basic.readthedocs.io/en/v1.0/

[image]