zephyrproject-rtos/zephyr #52588: ESP32c3 SPI driver DMA mode limited to 64 byte chunks

dean-at-evos, 2022-11-28:
The fix implemented in #35978 may have the unintended side effect of limiting DMA enabled transfers to 64 byte chunks as well.

My hack to test this was to add a check around the code limiting the chunk size in [spi_esp32_spim.c](https://github.com/zephyrproject-rtos/zephyr/blob/7ac69dc4ca95dcf7375a21c24aa4af22b34d6fa7/drivers/spi/spi_esp32_spim.c).
```
	size_t chunk_len = spi_context_max_continuous_chunk(&data->ctx);
	if (!cfg->dma_enabled) { 
	    chunk_len = MIN(chunk_len, SOC_SPI_MAXIMUM_BUFFER_SIZE);
	}
	size_t bit_len = chunk_len << 3;
```
With this added and declaring buffers as `DMA_ATTR` I could send a buffer larger than 64 bytes in one transaction. Without, multiple 64 byte transactions would occur when using DMA. AFAIK this is the intended use of the DMA enabled SPI hardware, I didn't test edge cases, or research if there should be a sensible chunk length for DMA transactions.

laurenmurphyx64, 2022-11-29:
CC @lgl88911

closed 2022-12-07 by fabiobaltieri

dean-at-evos, 2022-12-08:
Thanks people for fixing this up for us!

PR #52676 (2022-12-07): drivers: spi: esp32xx: Fix buffer length for DMA
Fixes #52588
commit b196edf5 (2022-12-07): drivers: spi: esp32xx: Fix buffer length for DMA

Fixes #52588