zephyrproject-rtos/zephyr #52598: esp32c3 Unable to do any timing faster than 1ms

mredp-evos, 2022-11-28:
Trying to use any of the kernel timing functions for anything less than 1ms does not work. I would have thought I just need to increase the `CONFIG_SYS_CLOCK_TICKS_PER_SEC` to `10000` to get 100us ticks. But this doesn't work.

I have tried various combinations of `CONFIG_SYS_CLOCK_TICKS_PER_SEC` and `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC`. But it always ends up wrong/very innacurate.

To be specific, sometimes 100us delay would be closer to 250us. Or sometimes 100us delay would be 10ms. But I could never get 100us to equal 100us.

Is it possible to do any real-time scheduling in the **microseconds**? Perhaps, I am missing something but 1ms (1kHz) seems a bit slow to be the timing limit for a 160MHz processor. 

Thanks,

mredp-evos, 2022-11-29:
Narrowed the issue down to `esp32c3_sys_timer.c`

A quick workaround was to set `#define MIN_DELAY` to 1. Then when you increase the `CONFIG_SYS_CLOCK_TICKS_PER_SEC` it won't be capped at 1ms.

I guess MIN_DELAY might be left over from the **esp32s2** driver where the HW CYCLES are faster perhaps? Whereas in the **esp32c3** HW CYCLE is only 1us which makes MIN_DELAY of 1000 equal 1ms.

I also set to `CONFIG_TICKLESS_KERNEL=n` but I never tested if this had any effect on the issue. I suspect that it did not.

This lead me to the other issue which is that the `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC` is fixed at 1MHz when the clock is a 16MHz clock. This is because the `esp32c3_sys_timer.c` file uses `systimer_hal` functions which use `SYSTIMER_TICKS_PER_US`. And because the `systimer_alarm` function uses `systimer_hal_get_time`.

Instead, base the timers off the `systimer_ll` functions and then in the `systimer_alarm` function use use `systimer_hal_get_counter_value` instead. 

Fixing either the MIN_DELAY issue or the CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC issue will allow you to get faster timings.

sylvioalves, 2022-12-20:
> Trying to use any of the kernel timing functions for anything less than 1ms does not work on the ESP32-C3. I would have thought I just need to increase the `CONFIG_SYS_CLOCK_TICKS_PER_SEC` to `10000` to get 100us ticks. But this doesn't work.
> 
> I have tried various combinations of `CONFIG_SYS_CLOCK_TICKS_PER_SEC` and `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC`. But it always ends up wrong/very innacurate.
> 
> To be specific, sometimes 100us delay would be closer to 250us. Or sometimes 100us delay would be 10ms. But I could never get 100us to equal 100us.
> 
> Is it possible to do any real-time scheduling in the **tens or hundreds of microseconds**? Perhaps, I am missing something but 1ms (1kHz) seems a bit slow to be the timing limit for a 160MHz processor.
> 
> Thanks,

Hi, can you describe which kernel timing functions you have been using? And how are you measuring the clock in this scenario?

PR #53453 (2023-01-04): driver: systimer: increase esp32c3 tick resolution
hal_espressif systimer HAL calls are based on 1MHz reference. This changes systimer driver to allow max clocking reference of 16MHz and increases soc tick resolution by reducing min delay interval.

This also sets all ESP32-C3 socs to 16MHz hardware cycles reference.

Fixes #52870
Fixes #52598
closed 2023-01-04 by carlescufi