zephyrproject-rtos/zephyr #52602: tests: subsys: settings: littlefs: system.settings.littlefs.raw fails

nordic-piks, 2022-11-28:
**Describe the bug**
The zephyr/tests/subsys/settings/littlefs/system.settings.littlefs.raw fails

Observed for
- nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/subsys/settings/littlefs -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.2.0-1928-g6c21c9635d3a ***
Running TESTSUITE settings_config
===================================================================
START - test_config_commit
PASS - test_config_commit in 0.001 seconds
===================================================================
START - test_config_empty_lookups
PASS - test_config_empty_lookups in 0.001 seconds
===================================================================
START - test_config_getset_int
PASS - test_config_getset_int in 0.001 seconds
===================================================================
START - test_config_getset_int64
PASS - test_config_getset_int64 in 0.001 seconds
===================================================================
START - test_config_getset_unknown
PASS - test_config_getset_unknown in 0.001 seconds
===================================================================
TESTSUITE settings_config succeeded
Running TESTSUITE settings_config_fs
===================================================================
I: LittleFS version 2.5, disk version 2.0
I: FS at mx25r6435f@0:0x0 is 16 0x1000-byte blocks with 512 cycle
I: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
E: ../../../../../../../../../../zephyr/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}
W: can't mount (LFS -84); formatting
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x00000000  r1/a2:  0x00000040  r2/a3:  0x00000000
E: r3/a4:  0x00005f0d r12/ip:  0x00000000 r14/lr:  0x20000614
E:  xpsr:  0x60000000
E: Faulting instruction address (r15/pc): 0x20000614
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200003a0 (idle)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.0
 - Commit SHA or Version used: zephyr-v3.2.0-1928-g6c21c9635d3a

nordic-piks, 2022-12-15:
Recently - https://github.com/zephyrproject-rtos/zephyr/commit/ffa622b5b9c82642157ea632790f04a6c337099c
test name was change to system.settings.file.raw

de-nordic, 2023-01-09:
Tried to reproduce issue with:
```
/zephyr/scripts/twister -T zephyr/tests/subsys/settings/file/ -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
```
on 11aa8454f01dc986eb78d663a926cb0b9168a2c4, and got successful execution:
```
*** Booting Zephyr OS build v3.1.0-rc2-9167-g11aa8454f01d ***
Running TESTSUITE settings_config
===================================================================
START - test_config_commit
PASS - test_config_commit in 0.001 seconds
===================================================================
START - test_config_empty_lookups
PASS - test_config_empty_lookups in 0.001 seconds
===================================================================
START - test_config_getset_int
PASS - test_config_getset_int in 0.001 seconds
===================================================================
START - test_config_getset_int64
PASS - test_config_getset_int64 in 0.001 seconds
===================================================================
START - test_config_getset_unknown
PASS - test_config_getset_unknown in 0.001 seconds
===================================================================
TESTSUITE settings_config succeeded
Running TESTSUITE settings_config_fs
===================================================================
I: LittleFS version 2.5, disk version 2.0
I: FS at mx25r6435f@0:0x0 is 16 0x1000-byte blocks with 512 cycle
I: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}
W: can't mount (LFS -84); formatting
I: /fs mounted
START - test_config_compress_file
E: file open error (-2)
PASS - test_config_compress_file in 6.074 seconds
===================================================================
START - test_config_empty_file
E: file open error (-2)
E: file open error (-2)
E: failed to create directory (-17)
PASS - test_config_empty_file in 0.061 seconds
===================================================================
START - test_config_multiple_in_file
PASS - test_config_multiple_in_file in 0.040 seconds
===================================================================
START - test_config_save_in_file
E: failed to create directory (-17)
E: file open error (-2)
PASS - test_config_save_in_file in 0.253 seconds
===================================================================
START - test_config_save_one_file
E: failed to create directory (-17)
PASS - test_config_save_one_file in 0.425 seconds
===================================================================
START - test_config_small_file
PASS - test_config_small_file in 0.077 seconds
===================================================================
TESTSUITE settings_config_fs succeeded
------ TESTSUITE SUMMARY START ------
SUITE PASS - 100.00% [settings_config]: pass = 5, fail = 0, skip = 0, total = 5 duration = 0.005 seconds
- PASS - [settings_config.test_config_commit] duration = 0.001 seconds
- PASS - [settings_config.test_config_empty_lookups] duration = 0.001 seconds
- PASS - [settings_config.test_config_getset_int] duration = 0.001 seconds
- PASS - [settings_config.test_config_getset_int64] duration = 0.001 seconds
- PASS - [settings_config.test_config_getset_unknown] duration = 0.001 seconds
SUITE PASS - 100.00% [settings_config_fs]: pass = 6, fail = 0, skip = 0, total = 6 duration = 6.930 seconds
- PASS - [settings_config_fs.test_config_compress_file] duration = 6.074 seconds
- PASS - [settings_config_fs.test_config_empty_file] duration = 0.061 seconds
- PASS - [settings_config_fs.test_config_multiple_in_file] duration = 0.040 seconds
- PASS - [settings_config_fs.test_config_save_in_file] duration = 0.253 seconds
- PASS - [settings_config_fs.test_config_save_one_file] duration = 0.425 seconds
- PASS - [settings_config_fs.test_config_small_file] duration = 0.077 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: a5eb8cf3ee646f9f2e0f38c577e64a12
PROJECT EXECUTION SUCCESSFUL
```

Use board: nrf52840dk_nrf52840 pca10056 0.12.0.

@nordic-piks Can you confirm the issue?

nordic-piks, 2023-01-09:
@de-nordic I was checking this today and I do not see fail since 10 days.
 Can you close it?
 When it will occur again I will raise/reopen issue.

closed 2023-01-09 by nordic-piks