zephyrproject-rtos/zephyr #52605: esp32-usb-serial tx-complete interrupt not working in interrupt mode on esp32c3

fabiobaltieri, 2022-11-28:
**Describe the bug**
Hi! I'm running into some issue when trying to use the `usb_serial` device on an ESP32C3 system. When running the `shell_module` sample, I can tell that the data is received correctly, but the output buffer is only flushed as a reaction to a receive event. For example I can type a command, but the echo is delayed and the output is not printed until something else is sent on the serial port. The behavior seems to indicate that `serial_esp32_usb_isr` is not being called as it should when the transmit buffer is empty, hence stopping the application from flushing its transmit queue.

The driver works correctly in polling mode (`CONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN=n`).

Additionally, the USB debug interface fails entirely when building the wifi sample (that is, does not enumerate, errors in dmesg, have to reset it in bootloader mode to flash it again). Could this be some interrupt management issue?

Disclaimer: I'm actually using a Seeed XIAO-ESP32-C3 board, not the official `esp32c3_devkitm`, but I think the problem should be the same since the device is builtin.

**To Reproduce**
Steps to reproduce the behavior:
1. patch `boards/riscv/esp32c3_devkitm/esp32c3_devkitm.dts`
```patch
-               zephyr,shell-uart = &uart0;
+               zephyr,shell-uart = &usb_serial;
```
2. build and flash the `shell_module` sample
``` 
west build -p -b esp32c3_devkitm samples/subsys/shell/shell_module
west flash
```
3. access the shell interface (minicom, west espressif monitor...), send some command, output is not flushing
4. build and flash with `west build -p -b esp32c3_devkitm samples/subsys/shell/shell_module -DCONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN=y`, the shell works fine

**Expected behavior**
The USB serial works as expected.

**Impact**
Annoyance

**Environment (please complete the following information):**
 - Linux, Zephyr SDK 0.15.2, main (55843e7d77), Seeed XIAO-ESP32-C3

cfriedt, 2022-11-28:
I've experienced the same problem

sylvioalves, 2022-11-30:
@fabiobaltieri, The below has been fixed already, but needs west.yml update, which depends of other PRs).  
```c
Additionally, the USB debug interface fails entirely when building the wifi sample (that is, does not enumerate, errors in dmesg, have to reset it in bootloader mode to flash it again). Could this be some interrupt management issue?
```  
I was able to confirm the same issue, will take a look. Thanks!

fabiobaltieri, 2022-12-05:
Thanks @sylvioalves, I think that the west update got merged but I still see the problem with `shell_module`, could you check it on your side? I'm running 659c2143b5 on the main repo and d53c07e949 on `modules/hal/espressif`.

fabiobaltieri, 2022-12-05:
The `usb-serial tx-complete`, that is `west build -p -b esp32c3_devkitm samples/subsys/shell/shell_module` getting the data out only in response to inputs.

sylvioalves, 2022-12-05:
> The `usb-serial tx-complete`, that is `west build -p -b esp32c3_devkitm samples/subsys/shell/shell_module` getting the data out only in response to inputs.

Sure, this still needs to be fixed.

cfriedt, 2023-03-19:
This is still quite problematic. The shell is practically unusable.

horakdavid, 2023-03-29:
Calling irq callback after enabling tx interrupt fixed this issue for me.
```
static void serial_esp32_usb_irq_tx_enable(const struct device *dev)
{
    struct serial_esp32_usb_data *data = dev->data;

    usb_serial_jtag_ll_clr_intsts_mask(USB_SERIAL_JTAG_INTR_SERIAL_IN_EMPTY);
    usb_serial_jtag_ll_ena_intr_mask(USB_SERIAL_JTAG_INTR_SERIAL_IN_EMPTY);

    if (data->irq_cb != NULL) {
        data->irq_cb(dev, data->irq_cb_data);
    }
}
```

sylvioalves, 2023-03-29:
Also confirmed it works, thanks.

fabiobaltieri, 2023-03-29:
There's a different problem with that workaround though, ran some sample application and it seems like they block until the serial port is opened on the host, guess the call is blocking. Anyway let's discuss it on a PR.

fabiobaltieri, 2023-03-29:
Yeah I tried flashing a blinky sample, run `west flash`, the led does not blink, open minicom on the serial port, it starts blinking. I think it's waiting on flushing the sample printk.

cfriedt, 2023-04-01:
> Calling irq callback after enabling tx interrupt fixed this issue for me.

@horakdavid - nice work - can you submit a PR?

PR #56483 (2023-04-04): driver: usb_serial: esp32c3: fix USB port behavior
- Fixes missing poll out events when UART interrupt is enabled
- Fixes blocking USB interface during board start up.

Fixes #52605
closed 2023-04-04 by carlescufi