zephyrproject-rtos/zephyr #52612: mcux: flexio: Add generic MCUX FlexIO driver.

MikhailSiomin, 2022-11-28:
Add generic MCUX FlexIO driver.
================================

**Zephyr** currently does not support ***mcux flexio***.

I originally implemented ***mcux flexio spi*** driver that uses FlexIO as SPI only. But @henrikbrixandersen proposed an implementation of a **generic** FlexIO driver. Now I have a proposal to split the driver into parts: 
* `zephyr\drivers\misc\mcux_flexio\mcux_flexio.c` – for initialization common resources (pinctrl, irq handler), initialization DEVICE_DT_DEFINE() for each childs (interfaces), and checking resource overlapping;
* `zephyr\drivers\<interface>\<interface>_mcux_flexio.c` – for code of specific interface (spi, i2c, uart, etc.);
* `zephyr\drivers\<interface>\<interface>_mcux_flexio.h` – for definition device data and device config structs.

***Compatible*** will define the interface type (SPI, I2C, UART, etc.).

`zephyr\drivers\<interface>\<interface>_mcux_flexio.h` will be included to `zephyr\drivers\misc\mcux_flexio\mcux_flexio.c` in the following way (for example for spi):

```c
...
#include <zephyr/drivers/spi/spi_mcux_flexio.h>
...
```
`zephyr\drivers\spi\spi_mcux_flexio.h`:
```c
...
struct spi_mcux_flexio_config {
    FLEXIO_SPI_Type *flexio_spi;
    const struct device *clock_dev;
    clock_control_subsys_t clock_subsys;
    void (*irq_config_func)(const struct device *dev);
#ifdef CONFIG_PINCTRL
    const struct pinctrl_dev_config *pincfg;
#endif /* CONFIG_PINCTRL */
};

struct spi_mcux_flexio_data {
    const struct device *dev;
    flexio_spi_master_handle_t handle;
    struct spi_context ctx;
    size_t transfer_len;
    uint8_t transfer_flags;
};
...
```

`example.overlay`:
```dts
&pinctrl {
    pinmux_flexioN: pinmux_flexioN {
        ...
    };
};

&flexioN {
    status = "okay";
    flexioN_spi0: flexioN_spi0 {
        compatible = "nxp,imx-flexio-spi";
        status = "okay";
        ...
    };
    flexioN_i2c0: flexioN_i2c0 {
        compatible = "nxp,imx-flexio-i2c";
        status = "okay";
        ...
    };
    flexioN_uart0: flexioN_uart0 {
        compatible = "nxp,imx-flexio-uart";
        status = "okay";
        ...
    };
    pinctrl-0 = <&pinmux_flexioN>;
    pinctrl-names = "default";
};
```

If the community approves this proposal, then I will implement it.

carlescufi, 2022-11-28:
@MikhailSiomin I recommend you also mention that you posted this RFC in the #NXP [Discord channel](https://chat.zephyrproject.org/)

MikhailSiomin, 2022-12-06:
@henrikbrixandersen, what do you think of the generic driver suggested above?

harristomyx, 2023-09-18:
Hi, is there any plans to move forward with using this generic FlexIO implementation?

DerekSnell, 2023-09-18:
Hi @harristomyx ,
There is an open PR to add this FlexIO driver at https://github.com/zephyrproject-rtos/zephyr/pull/53748.

Best regards

closed 2024-03-12 by aescolar