zephyrproject-rtos/zephyr #52614: Using "test item" doesn't report warnings/errors

PerMac, 2022-11-28:


marc-hb, 2022-11-28:
Thanks for the report. I transferred this to the `zephyr` repo because the `west build` command is not part of `west`. If you run `git grep test.item` in both repos you will find nothing in the `west` repo and the (relatively simple) implementation of `--test-item` in `zephyr/scripts/west_commands/build.py`. If you look at the short `_parse_test_item()` there you will see that the general logic of that function is "look everywhere and grab every CMake argument we can find". So it's not a big surprise that it does not fail when nothing is found.

You could change the logic a bit to fail when either no file is found or when no CMake argument is found (which one??) but this would unfortunately break backwards-compatibility. It's not a stretch of the imagination to think some current `--test-item` users may have come to actually rely on `--test-item` doing nothing in come cases: for instance you could use that generated some `testcase.yml` at build time and conveniently test with versus without some CMake option. Could also be convenient to compare results of tests run by CI.

mbolivar-nordic, 2022-11-30:
@carlocaione originally added support for this feature to `west build` -- can you take a look?

carlocaione, 2022-11-30:
> @carlocaione originally added support for this feature to `west build` -- can you take a look?

nope, I suggested that but @nashif actually wrote the code.

mbolivar-nordic, 2022-11-30:
> > @carlocaione originally added support for this feature to `west build` -- can you take a look?
> 
> nope, I suggested that but @nashif actually wrote the code.

Thanks for the correction; assignee applied

PR #54163 (2023-01-28): west build --test-item improvement, twister synopsis
Provide brief summary about failed tests at the end and instructions on how to reporduce them using twister or west.

- west: build: handler errors in west build with --test-item
- west: build: allow --test-item to accept twister style instance names
- twister: print synopsis at the end and instruction for reproducing

Fixes #52614
commit 10081243 (2023-01-28): west: build: handler errors in west build with --test-item

Handle errors and wrong test meta-data when using --test-item.

Fixes #52614
closed 2023-01-28 by nashif