zephyrproject-rtos/zephyr #52623: qemu_x86: thousands of timer interrupts per second

wentasah, 2022-11-29:
**Describe the bug**

When running sample applications under qemu_x86, I experience wired problems caused by overloading the system with timer interrupts.

 - **What target platform are you using?** qemu_x86
 - **What have you tried to diagnose or workaround this issue?**
 
   Tried different samples. Tried older zephyr versions.
   
 - **Is this a regression?**
   Not sure.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `(sleep 10; printf "\x01c"; echo "info irq"; sleep 1; echo "info irq"; printf "\x01x") | west build -t run -b qemu_x86 samples/basic/minimal
`

   This command runs the sample while waiting 10 seconds, switching to qemu monitor and then it prints interrupt counts twice with one second delay in between. The output is:
```
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..
*** Booting Zephyr OS build zephyr-v3.2.0-2077-g737d799660cc ***
QEMU 7.0.0 monitor - type 'help' for more information
(qemu) info irq
IRQ statistics for isa-i8259:
 0: 792144
 4: 1
IRQ statistics for ioapic:
 0: 792144
 4: 1
(qemu) info irq
IRQ statistics for isa-i8259:
 0: 880620
 4: 1
IRQ statistics for ioapic:
 0: 880620
 4: 1
(qemu) QEMU: Terminated

```
2. Observe that there were 880620 - 792144 = 88476 interrupts per second!

**Expected behavior**

The number of timer interrupts is much lower.

**Impact**

`samples/subsys/console/getchar` doesn't work.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA  737d799660ccf843c5396330fbaf254e1a322409 (zephyr-v3.2.0-2077-g737d799660). The same happens with zephyr-v3.0.0, zephyr-v3.1.0 and zephyr-v3.2.0.

wentasah, 2023-01-30:
Not stale. Still happens with current main. It seems that this is a `qemu` problem. The same code works correctly on x86 hardware.

andyross, 2023-04-15:
Coming through triaging old bugs.  This is actually just a mistake.  The default qemu_x86 configuration uses CONFIG_QEMU_ICOUNT=y, which runs the emulation in virtual time, not host/real time.  You're seeing extremely fast interrupt rates because the emulated system is running hundreds or thousands of times faster than real time.  Instead of counting interrupts in your app, it's easier to see this with e.g. a simple sleep loop, e.g.:

```c
    while (true) {
        k_msleep(1000);
        printk(".\n");
    }
```

Build this as above and you'll see it spam the console when it runs.  Build with `west build ... -- -DCONFIG_QEMU_ICOUNT=n` and you'll see the lines come out once a second.

closed 2023-04-15 by andyross