zephyrproject-rtos/zephyr #52625: USB device: mimxrt685_evk_cm33: premature ZLP during control IN transfer

desowin, 2022-11-29:
**Describe the bug**
USB CDC ACM samples does not work on NXP MIMXRT685-EVK (mimxrt685_evk_cm33).

The behavior is the same for both v3.2.0 and current main branch. The device prematurely ends GET CONFIGURATION control transfer. During Data stage, the first 64 bytes are transferred correctly, then instead of sending remaining 11 bytes, the device sends ZLP that ends the Control transfer Data stage.

**To Reproduce**
Steps to reproduce the behavior:
1. west build --pristine -b mimxrt685_evk_cm33 -d build-cdc-nxp685 ~/zephyrproject/zephyr/samples/subsys/usb/cdc_acm
2. cd build-cdc-nxp685
3. west flash
4. Connect RT685 USB to host and observe enumeration failure

**Expected behavior**
Host should receive whole CONFIGURATION descriptor, i.e. all requested 75 bytes 

**Impact**
This renders USB device support useless for all classes that use control IN transfers with wLength larger than 64 bytes.

**Logs and console output**
```
[15687.689485] usb 3-4: new high-speed USB device number 32 using xhci_hcd
[15687.836726] usb 3-4: config index 0 descriptor too short (expected 75, got 64)
[15687.836741] usb 3-4: config 1 has an invalid descriptor of length 7, skipping remainder of the config
[15687.836749] usb 3-4: config 1 interface 1 altsetting 0 has 0 endpoint descriptors, different from the interface descriptor's value: 2
[15687.837223] usb 3-4: New USB device found, idVendor=2fe3, idProduct=0001, bcdDevice= 3.02
[15687.837234] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15687.837239] usb 3-4: Product: Zephyr CDC ACM sample
[15687.837243] usb 3-4: Manufacturer: ZEPHYR
[15687.837246] usb 3-4: SerialNumber: 6A631C9D80D5EC67
[15687.839685] cdc_acm: probe of 3-4:1.0 failed with error -22
```

USB connection trace can be viewed in Wireshark 4.0 or newer.
[zephyr-cdc-acm-nxp-mimxrt685_evk_cm33.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/10112751/zephyr-cdc-acm-nxp-mimxrt685_evk_cm33.pcap.gz)

Packet 816 in attached capture contains the ZLP that prematurely ends Data stage. Instead of the ZLP, the device should have sent the trailing CONFIGURATION descriptor 11 bytes.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.1
 - zephyr-v3.2.0 tag (4256cd41df6c60f1832fd2deb14edc30ac7debab)

**Additional context**
The issue seems to be in the NXP driver and not in USB stack because the same sample works perfectly fine on Nordic targets, e.g. nrf52840dk_nrf52840

carlescufi, 2022-11-30:
cc @danieldegrasse @mmahadevan108

carlescufi, 2022-11-30:
@desowin as per @jfischer-no, can you try enabling `CONFIG_USB_DC_HAS_HS_SUPPORT`?

desowin, 2022-11-30:
> @desowin as per @jfischer-no, can you try enabling `CONFIG_USB_DC_HAS_HS_SUPPORT`?

No change at all. Note that `CONFIG_USB_DC_HAS_HS_SUPPORT` affects the bulk endpoint `wMaxPacketSize` value in CONFIGURATION descriptor. The issue here is with 64 byte **control** endpoint. The 64 bytes is the only possibility for High-Speed control endpoint, while it is one of the options for Full-Speed control endpoints (FS options are: 8, 16, 32 or 64). Zephyr USB stack assumes control endpoint max packet size is always 64 bytes.

maximevince, 2023-01-18:
Any progress here? I am encountering the same issue on an LPC55S28, and am looking for a solution, too.

jfischer-no, 2023-01-19:
The issue is _likely_ how shim driver interacts with mcux hal driver (and would not be there or easier to fix with native driver).
**usb_dc_mcux.c** is a shim driver for two different hal drivers, IP3511 and EHCI. usb_dc-ep_write() implementation in the shim driver uses intermediate buffer and transaction length clamped to MPS of an endpoint
https://github.com/zephyrproject-rtos/zephyr/blob/883e9d367f6cd489666a562c83df0f56458c94b2/drivers/usb/device/usb_dc_mcux.c#L501-L516

For the control IN transfer (e.g. data stage of GetDescriptor request) and transfer length > 64, shim driver initiate a **transaction** of 64 bytes. But for the hal driver it is a **transfer** of 64 bytes, which then ends with ZLP because it is shorter as requested wLength. Check following lines in the hal driver:

https://github.com/zephyrproject-rtos/hal_nxp/blob/3ee6020efc1f8323d0fdc85615d3477161f0aa1f/mcux/middleware/mcux-sdk-middleware-usb/device/usb_device_lpcip3511.c#L1332-L1362

closed 2023-01-30 by fabiobaltieri