zephyrproject-rtos/zephyr #52632: MQTT over WebSockets: After hours of running time receiving published messages is strongly delayed

jumo-sballing, 2022-11-29:
**Describe the bug**
I using MQTT over secure websockets. The most of the time the device is simply pushing mqtt message in an interval of 5 seconds to a single topic. But sometimes another mqtt client publishes a message to a subscribed topic. Right after the start of the device the published message from the other client is immediately send to the MQTT event callback. Thats fine. But after a several hours of running time the published message ot the other client is strongly delayed. Sometimes up to one minute or longer. I have looked at the behavior and found out, that the MQTT message gets immediately into the WebSocket buffer. But the MQTT stack does not get it right away. This behaviour stops when restarting the device.
If using MQTTS instead of MQTT over WebSockets everything works fine.

Please also mention any information which could help others to understand
the problem you're facing:
- What target platform are you using?
**STM32F4**
- What have you tried to diagnose or workaround this issue?
**Looking at the debug prints of the mqtt and websocket stack**
- Is this a regression? If yes, have you been able to "git bisect" it to a specific commit?
**Not that I'm aware of.**

**To Reproduce**
Steps to reproduce the behavior:

- Connect to a broker via MQTT_TRANSPORT_SECURE_WEBSOCKET
- Subscribe to a topic X
- Publish a message to a topic Y with QoS2 every 5 seconds
- Sometimes publish a message to topic X (QoS1) from another client
- Watch the delay between publish and receiving this message is rising over the time

**Expected behavior**
The time between publish a message and getting it through the event cb should be relatively constant.

**Impact**
I'm not able to deliver the product with this bug.

**Logs and console output**
Here is a log of this situation. Lines, starting with "!!!->" are comments added by me.

<details>
<summary>Click to open LOG</summary>

```
!!!-> Publish Message
[02:03:52.619,000] <inf> CloudClient: MQTT PUB [QoS2] 21.910614 > 6269435adc5b781b5d060c1b
[02:03:52.620,000] <dbg> net_mqtt.mqtt_publish: (MqttThread): [CID 0x200095b4]:[State 0x06]: >> Topic size 0x00000041, Data size 0x00000038
[02:03:52.620,000] <dbg> net_mqtt_enc.pack_utf8_str: (MqttThread): >> str_size:00000043 cur:0x20008bfd, end:0x20008ff8
[02:03:52.620,000] <dbg> net_mqtt_enc.pack_uint16: (MqttThread): >> val:0041 cur:0x20008bfd, end:0x20008ff8
[02:03:52.620,000] <dbg> net_mqtt_enc.pack_uint16: (MqttThread): >> val:565c cur:0x20008c40, end:0x20008ff8
[02:03:52.620,000] <dbg> net_mqtt_enc.mqtt_encode_fixed_header: (MqttThread): << msg type:0x34 length:0x0000007d
[02:03:52.620,000] <dbg> net_mqtt_enc.packet_length_encode: (MqttThread): >> length:0x0000007d cur:(nil), end:(nil)
[02:03:52.620,000] <dbg> net_mqtt_enc.mqtt_encode_fixed_header: (MqttThread): Fixed header length = 02
[02:03:52.620,000] <dbg> net_mqtt_enc.pack_uint8: (MqttThread): >> val:34 cur:0x20008bfb, end:0x20008ff8
[02:03:52.620,000] <dbg> net_mqtt_enc.packet_length_encode: (MqttThread): >> length:0x0000007d cur:0x20008bfc, end:0x20008ff8
[02:03:52.620,000] <dbg> net_mqtt.client_write_msg: (MqttThread): [0x200095b4]: Transport writing message.
[02:03:52.620,000] <dbg> net_websocket.websocket_send_msg: (MqttThread): [0x2000a2c8] Len 71 BIN/1/more
[02:03:52.622,000] <dbg> net_websocket.websocket_send_msg: (MqttThread): [0x2000a2c8] Len 56 CONT/1/final
[02:03:52.624,000] <dbg> net_mqtt.client_write_msg: (MqttThread): [0x200095b4]: Transport write complete.
[02:03:52.624,000] <dbg> net_mqtt.mqtt_publish: (MqttThread): [CID 0x200095b4]:[State 0x06]: << result 0x00000000
[02:03:52.624,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:52.624,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:52.875,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:52.875,000] <dbg> net_websocket.websocket_recv_msg: (MqttThread): There is 82 bytes of data
[02:03:52.875,000] <dbg> net_mqtt_dec.unpack_uint8: (MqttThread): >> cur:0x200087f8, end:0x200087fa
[02:03:52.875,000] <dbg> net_mqtt_dec.unpack_uint8: (MqttThread): << val:40
[02:03:52.875,000] <dbg> net_mqtt_dec.packet_length_decode: (MqttThread): length:0x00000002
[02:03:52.875,000] <dbg> net_mqtt_rx.mqtt_handle_packet: (MqttThread): [CID 0x200095b4]: Received MQTT_PKT_TYPE_PUBACK!
[02:03:52.875,000] <dbg> net_mqtt_dec.unpack_uint16: (MqttThread): >> cur:0x200087fa, end:0x200087fc
[02:03:52.875,000] <dbg> net_mqtt_dec.unpack_uint16: (MqttThread): << val:8314
[02:03:53.125,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:53.375,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:53.626,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:53.876,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:54.126,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:54.376,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:54.626,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:54.877,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:03:55.127,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[... 612 lines cut ...]
[02:05:06.513,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:06.763,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:07.013,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:07.264,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:07.514,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:07.764,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:08.014,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:08.265,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:08.515,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:08.765,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:09.015,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
[02:05:09.266,000] <dbg> net_mqtt.mqtt_input: (MqttThread): state:0x00000006
```
</details>

**Environment (please complete the following information):**
- OS: Linux
- Toolchain: toolchain-gccarmnoneeabi @ 1.80201.181220 (8.2.1), VS Code with PlatformIO
- Commit SHA or Version used: Zephyr version 2.7.2 (but also tested all mqtt and websocket related fixes up to Master-HEAD)

**Additional context**

The websocket debug print that gives the remaining amount of data is constantly rising during normal operation. 
[00:46:29.210,000] <dbg> net_websocket.websocket_recv_msg: (MqttThread): There is xx bytes of data
Right after boot there are 2 bytes. An hour later there are 30 bytes. And that number is never falling.

rlubos, 2022-12-01:
@jumo-sballing Can you perhaps share your code or provide a minimal sample which allows to reproduce the issue? I've tried to reproduce this case, according to the description, with the following setup:
* local `mosquitto` server with secure websocket port open
* slightly modified MQTT sample from Zephyr, which periodically publishes QoS2 packet to one topic, and subscribes to another one (built with `overlay-tls.conf` and `overlay-websocket.conf`.
* `mosquitto_pub` which publishes QoS1 to the subscribed topic

Even after publishing lots of packets, I see no regression in processing time, nor data accrual in internal websocket buffer.

jumo-sballing, 2022-12-05:
Thank you very much for checking this out. And thanks for the reply. I will setup a small example based on my code.

svenstorp, 2022-12-12:
@jumo-sballing @rlubos I think I just got the same issue, or at least a suspiciously similar issue. If it is the same issue your message should be delayed until there is a second message arriving?
The issue I stumbled upon is triggered when 2 consecutive websocket messages are sent before the first one is read out with websocket_recv_msg().

The issue is around zephyr/subsys/net/lib/websocket/websocket.c line 832 - 848 on main branch. When websocket_recv_msg() is entered with an already existing message in ctx->tmp_buf, you will hang in recv (or return, depending on your timeout settings) until another message is received.

rlubos, 2022-12-13:
> When websocket_recv_msg() is entered with an already existing message in ctx->tmp_buf, you will hang in recv (or return, depending on your timeout settings) until another message is received.

Ah, that would make sense... If two consecutive messages are read with a single `recv()` call (which could happen, it's stream underneath), `websocket_recv_msg()` will only process one of them. Then on the consecutive call, it ignores the fact the next message may already be present in the buffer, and tries to read it out from the socket, causing hickups. Nice find @svenstorp !

We have another WebSocket-related issue that proposes a complete rework of this `websocket_recv_msg()` function (https://github.com/zephyrproject-rtos/zephyr/issues/52327#issuecomment-1337430132). Need to make sure to get rid of this bug during the rework.

GrixaYrev, 2022-12-14:
Please check PR https://github.com/zephyrproject-rtos/zephyr/pull/53070.

rlubos, 2022-12-20:
@jumo-sballing @svenstorp Please give #53070 a try, I've briefly tested it with MQTT and it seems to work fine, the updated RX implementation should help with the issue described above as well.

carlescufi, 2022-12-21:
Hopefully fixed by #53070, please reopen if that is not the case.

closed 2022-12-21 by carlescufi

jumo-sballing, 2022-12-21:
Sorry for beeing so late at the party. I was sick for a few days. I've tested #53070 with my code and now it works fine. I've got no delay - even after hours. 
Thank you very, very much!