zephyrproject-rtos/zephyr #52667: nrf_rtc_timer: Booting application with zephyr < 3.0.0 from mcuboot with zephyr >= 3.0.0

kurddt, 2022-11-30:
**Describe the bug**
Due to change of the NRF RTC timer in zephyr v3.0.0 which enable the overflow interrupt, MCUBoot will leave this interrupt enabled which is not handled by the driver for earlier version. In practice that mean that in such use case, after ~8.5 minutes of running the interrupt will be triggered indefinitely leading to starvation of all threads.

**To Reproduce**
Steps to reproduce the behavior:
On NRF an platform:
1. Build and program MCUBoot with zephyr >= v3.0.0
2. Build and program an application with zephyr < v3.0.0
3. Wait 8 minutes and 30s

**Impact**
The impact of that issue is not that big since this configuration ("new" bootloader with "old" application) is probably not very common

carlescufi, 2022-12-21:
Perhaps this Kconfig option can help:
https://github.com/mcu-tools/mcuboot/blob/d165e9b2a52bcf01164d7e2529156a91a7a92d83/boot/zephyr/Kconfig#L146

carlescufi, 2023-02-07:
@kurddt did the Kconfig option above help?

kurddt, 2023-02-07:
@carlescufi No it doesn't help. I think `MCUBOOT_CLEANUP_ARM_CORE` only clean cleanup the core registers and the RTC is not part of them

rene-becker-setec, 2023-02-14:
We just ran into this very issue as well ...

Probably a not a very common use-case to have a newer boot-loader working with a Application that is based on an older version of Zephyr, but it does happen ...

In our case we had to start with an 'ancient' existing Application and the development schedule at the moment does not allow porting the Application to a newer version of Zephyr (although it certainly is on the roadmap). However, we were hoping to deploy a more up to date boot-loader at product introduction.

Apart from this issue we also found that with the newer boot-loader the HW Stack Protection is enabled by default. A good thing in principle, but we found that with this feature the boot-loader also left behind some 'dirty' hardware causing some grief for the Application. Easily fixable in that case by changing a config but I wonder whether it'd not in general be better practice for the boot-loader to hand over the HW to the Application FW in an as 'clean' state as possible.

carlescufi, 2023-03-28:
@kurddt and @rene-becker-setec did https://github.com/zephyrproject-rtos/zephyr/pull/55259 fix this for you?

kurddt, 2023-03-28:
I don't think it will since the application running on old zephyr won't receive this patch. To be fixed this would required mcuboot to cleanup the RTC registers before loading the app.

nordicjm, 2023-04-04:
> I think the bigger issue here is that there seems to be no 'contract' on how the handover bootloader -> application works (on any device, not specific to Nordic). Should the bootloader clean up any resources it used or does the application before doing anything else have to make sure the HW is in a well defined state.

Ideally, yes, in practice, no. The problem with this, and what is very much lacking in zephyr at present, is that there is no "disable" or "clean up" functionality for drivers, so they init at boot time, and that's their state for the lifetime of that boot.

PR #56528 (2023-04-11): drivers: timer: nrf_rtc_timer: Implement stop function
    drivers: timer: nrf_rtc_timer: Implement stop function
    
    Implements functionality to stop the nRF RTC timer clock source.

Fixes #52667
closed 2023-04-11 by carlescufi