zephyrproject-rtos/zephyr #52670: tests: subsys: usb: device: usb.device hangs

nordic-piks, 2022-11-30:
**Describe the bug**
The zephyr/tests/subsys/usb/device/usb.device hangs

Observed for
- nrf5340dk_nrf5340_cpuapp_ns

But test works fine for:
- nrf5340dk_nrf5340_cpuapp
- nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk_nrf5340_cpuapp_ns connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/subsys/usb/device  -p nrf5340dk_nrf5340_cpuapp_ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-11-30T00:11:41.998Z] [0mERROR   - D: __usb_descriptor_start 0x20040124[0m
[2022-11-30T00:11:41.998Z] [0mD: __usb_descriptor_end 0x20040195[0m
[2022-11-30T00:11:41.998Z] [0mD: Configuration descriptor 0x20040136[0m
[2022-11-30T00:11:41.998Z] [0mD: Interface descriptor 0x2004013f[0m
[2022-11-30T00:11:41.998Z] [0mD: Endpoint descriptor 0x20040148[0m
[2022-11-30T00:11:41.998Z] [0mD: ep 0x81, mps 64, type 2[0m
[2022-11-30T00:11:41.998Z] [0mD: Fixing EP address 81 -> 81[0m
[2022-11-30T00:11:41.998Z] [0mD: endpoint 0x81[0m
[2022-11-30T00:11:41.998Z] [0mD: Now the wTotalLength is 25[0m
[2022-11-30T00:11:41.998Z] [0mD: idx_max 11, ascii_idx_max 5, buf 0x20040155[0m
[2022-11-30T00:11:41.998Z] [0mD: char R : 52, idx 5 -> 11[0m
[2022-11-30T00:11:41.998Z] [0mD: char Y : 59, idx 4 -> 9[0m
[2022-11-30T00:11:41.998Z] [0mD: char H : 48, idx 3 -> 7[0m
[2022-11-30T00:11:41.998Z] [0mD: char P : 50, idx 2 -> 5[0m
[2022-11-30T00:11:41.998Z] [0mD: char E : 45, idx 1 -> 3[0m
[2022-11-30T00:11:41.998Z] [0mD: char Z : 5a, idx 0 -> 1[0m
[2022-11-30T00:11:41.998Z] [0mD: idx_max 13, ascii_idx_max 6, buf 0x20040163[0m
[2022-11-30T00:11:41.998Z] [0mD: char V : 56, idx 6 -> 13[0m
[2022-11-30T00:11:41.998Z] [0mD: char E : 45, idx 5 -> 11[0m
[2022-11-30T00:11:41.998Z] [0mD: char D : 44, idx 4 -> 9[0m
[2022-11-30T00:11:41.998Z] [0mD: char - : 2d, idx 3 -> 7[0m
[2022-11-30T00:11:41.998Z] [0mD: char B : 42, idx 2 -> 5[0m
[2022-11-30T00:11:41.998Z] [0mD: char S : 53, idx 1 -> 3[0m
[2022-11-30T00:11:41.998Z] [0mD: char U : 55, idx 0 -> 1[0m
[2022-11-30T00:11:41.998Z] [0mD: Serial Number[0m
[2022-11-30T00:11:41.998Z] [0mD: 5d d9 94 a8 62 ee 8c 1d |]...b...[0m
[2022-11-30T00:11:41.998Z] [0mD: idx_max 31, ascii_idx_max 15, buf 0x20040173[0m
[2022-11-30T00:11:41.998Z] [0mD: char D : 44, idx 15 -> 31[0m
[2022-11-30T00:11:41.998Z] [0mD: char 1 : 31, idx 14 -> 29[0m
[2022-11-30T00:11:41.998Z] [0mD: char C : 43, idx 13 -> 27[0m
[2022-11-30T00:11:41.998Z] [0mD: char 8 : 38, idx 12 -> 25[0m
[2022-11-30T00:11:41.998Z] [0mD: char E : 45, idx 11 -> 23[0m
[2022-11-30T00:11:41.998Z] [0mD: char E : 45, idx 10 -> 21[0m
[2022-11-30T00:11:41.998Z] [0mD: char 2 : 32, idx 9 -> 19[0m
[2022-11-30T00:11:41.998Z] [0mD: char 6 : 36, idx 8 -> 17[0m
[2022-11-30T00:11:41.998Z] [0mD: char 8 : 38, idx 7 -> 15[0m
[2022-11-30T00:11:41.998Z] [0mD: char A : 41, idx 6 -> 13[0m
[2022-11-30T00:11:41.998Z] [0mD: char 4 : 34, idx 5 -> 11[0m
[2022-11-30T00:11:41.998Z] [0mD: char 9 : 39, idx 4 -> 9[0m
[2022-11-30T00:11:41.998Z] [0mD: char 9 : 39, idx 3 -> 7[0m
[2022-11-30T00:11:41.998Z] [0mD: char D : 44, idx 2 -> 5[0m
[2022-11-30T00:11:41.998Z] [0mD: char D : 44, idx 1 -> 3[0m
[2022-11-30T00:11:41.998Z] [0mD: char 5 : 35, idx 0 -> 1[0m
[2022-11-30T00:11:41.998Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2064-g7c8264277a93 ***[0m
[2022-11-30T00:11:41.998Z] [0mRunning TESTSUITE device_usb[0m
[2022-11-30T00:11:41.998Z] [0m===================================================================[0m
[2022-11-30T00:11:41.998Z] [0mD: lock usb_enable_lock mutex[0m
[2022-11-30T00:11:41.998Z] [0mD: set cb, ep: 0x81[0m
[2022-11-30T00:11:41.998Z] [0mD: EP enable: 0x00[0m
[2022-11-30T00:11:41.998Z] [0mD: EP enable: 0x80[0m
[2022-11-30T00:11:41.998Z] [0mD: unlock usb_enable_lock mutex[0m
[2022-11-30T00:11:41.998Z] [0mSTART - test_usb_dc_api[0m
[2022-11-30T00:11:41.998Z] [0mPASS - test_usb_dc_api in 0.001 seconds[0m
[2022-11-30T00:11:41.998Z] [0m===================================================================[0m
[2022-11-30T00:11:41.998Z] [0mSTART - test_usb_dc_api_invalid[0m
[2022-11-30T00:11:41.998Z] [0mD: ep_write: ep 0x20, len 1[0m
[2022-11-30T00:11:41.998Z] [0mD: ep_read: ep 0x20, maxlen 1[0m
[2022-11-30T00:11:41.998Z] [0mPASS - test_usb_dc_api_invalid in 0.013 seconds[0m
[2022-11-30T00:11:41.998Z] [0m===================================================================[0m
[2022-11-30T00:11:41.998Z] [0mSTART - test_usb_dc_api_read_write[0m
[2022-11-30T00:11:41.998Z] [0mD: ep_read: ep 0x20, maxlen 1[0m
[2022-11-30T00:11:41.998Z] [0mD: ep_write: ep 0x20, len 1[0m
[2022-11-30T00:11:41.998Z] [0mPASS - test_usb_dc_api_read_write in 0.013 seconds[0m
[2022-11-30T00:11:41.998Z] [0m===================================================================[0m
[2022-11-30T00:11:41.998Z] [0mSTART - test_usb_deconfig[0m
[2022-11-30T00:11:41.998Z] [0mPASS - test_usb_deconfig in 0.001 seconds[0m
[2022-11-30T00:11:41.998Z] [0m===================================================================[0m
[2022-11-30T00:11:41.998Z] [0mSTART - test_usb_disable[0m
[2022-11-30T00:11:41.998Z] [0mD: EP disable: 0x00[0m
[2022-11-30T00:11:41.998Z] [0mD: EP disable: 0x80[0m
[2022-11-30T00:11:41.998Z] [0m[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.1
 - Commit SHA or Version used: zephyr-v3.2.0-2064-g7c8264277a93

desowin, 2022-12-02:
I cannot reproduce this. Note that I had to use `/dev/ttyACM1` instead of `/dev/ttyACM0`, because the output appears only on the second serial port. 
```
[tmon@tmon-laptop zephyr]$ git log -1 --format=%H
7c8264277a9377cd41f8dfe3a777bb897015b562
[tmon@tmon-laptop zephyr]$ ./scripts/twister -T tests/subsys/usb/device -p nrf5340dk_nrf5340_cpuapp_ns --device-testing --device-serial /dev/ttyACM1 -v --inline-logs
ZEPHYR_BASE unset, using "/home/tmon/zephyrproject/zephyr"
Renaming output directory to /home/tmon/zephyrproject/zephyr/twister-out.4
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.2.0-2064-g7c8264277a93
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/tmon/zephyrproject/zephyr/twister-out/testplan.json

Device testing on:

| Platform                    | ID   | Serial device   |
|-----------------------------|------|-----------------|
| nrf5340dk_nrf5340_cpuapp_ns |      | /dev/ttyACM1    |

INFO    - JOBS: 12
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/1 nrf5340dk_nrf5340_cpuapp_ns zephyr/tests/subsys/usb/device/usb.device          PASSED (device 16.129s)

INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 27.31 seconds
INFO    - In total 5 test cases were executed, 0 skipped on 1 out of total 518 platforms (0.19%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board                       | ID   |   Counter |
|-----------------------------|------|-----------|
| nrf5340dk_nrf5340_cpuapp_ns |      |         1 |
INFO    - Saving reports...
INFO    - Writing JSON report /home/tmon/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

```
2022-12-02 08:30:38,353 - twister - INFO - Using Ninja..
2022-12-02 08:30:38,369 - twister - INFO - Zephyr version: zephyr-v3.2.0-2064-g7c8264277a93
2022-12-02 08:30:38,372 - twister - DEBUG - Running cmake script /home/tmon/zephyrproject/zephyr/cmake/verify-toolchain.cmake
2022-12-02 08:30:38,372 - twister - DEBUG - Calling cmake: /usr/bin/cmake -DFORMAT=json -P /home/tmon/zephyrproject/zephyr/cmake/verify-toolchain.cmake
2022-12-02 08:30:38,709 - twister - DEBUG - Finished running  /home/tmon/zephyrproject/zephyr/cmake/verify-toolchain.cmake
2022-12-02 08:30:38,709 - twister - INFO - Using 'zephyr' toolchain.
2022-12-02 08:30:38,783 - twister - DEBUG - Reading test case configuration files under /home/tmon/zephyrproject/zephyr/tests/subsys/usb/device...
2022-12-02 08:30:38,783 - twister - DEBUG - Found possible testsuite in /home/tmon/zephyrproject/zephyr/tests/subsys/usb/device
2022-12-02 08:30:38,785 - twister - DEBUG - Reading platform configuration files under /home/tmon/zephyrproject/zephyr/boards...
2022-12-02 08:30:39,128 - twister - DEBUG - Reading platform configuration files under /home/tmon/zephyrproject/zephyr/scripts/pylib/twister/boards...
2022-12-02 08:30:39,129 - twister - DEBUG - platform filter: ['nrf5340dk_nrf5340_cpuapp_ns']
2022-12-02 08:30:39,129 - twister - DEBUG -     arch_filter: None
2022-12-02 08:30:39,129 - twister - DEBUG -      tag_filter: None
2022-12-02 08:30:39,129 - twister - DEBUG -     exclude_tag: None
2022-12-02 08:30:39,129 - twister - INFO - Building initial testsuite list...
2022-12-02 08:30:39,129 - twister - INFO - Writing JSON report /home/tmon/zephyrproject/zephyr/twister-out/testplan.json
2022-12-02 08:30:39,135 - twister - INFO - JOBS: 12
2022-12-02 08:30:39,135 - twister - INFO - Adding tasks to the queue...
2022-12-02 08:30:39,135 - twister - DEBUG - adding nrf5340dk_nrf5340_cpuapp_ns/zephyr/tests/subsys/usb/device/usb.device
2022-12-02 08:30:39,136 - twister - INFO - Added initial list of jobs to queue
2022-12-02 08:30:39,136 - twister - DEBUG - Launch process 0
2022-12-02 08:30:39,136 - twister - DEBUG - Launch process 1
2022-12-02 08:30:39,137 - twister - DEBUG - Launch process 2
2022-12-02 08:30:39,138 - twister - DEBUG - Launch process 3
2022-12-02 08:30:39,139 - twister - DEBUG - Running cmake on /home/tmon/zephyrproject/zephyr/tests/subsys/usb/device for nrf5340dk_nrf5340_cpuapp_ns
2022-12-02 08:30:39,140 - twister - DEBUG - Calling cmake: /usr/bin/cmake -B/home/tmon/zephyrproject/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/zephyr/tests/subsys/usb/device/usb.device -DTC_RUNID=6823661a5af2e5762eb7e45936f2f632 -DEXTRA_CFLAGS=-Werror '-DEXTRA_AFLAGS=-Werror -Wa,--fatal-warnings' -DEXTRA_LDFLAGS=-Wl,--fatal-warnings -DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror -GNinja -S/home/tmon/zephyrproject/zephyr/tests/subsys/usb/device -DBOARD=nrf5340dk_nrf5340_cpuapp_ns
2022-12-02 08:30:39,140 - twister - DEBUG - Launch process 4
2022-12-02 08:30:39,141 - twister - DEBUG - Launch process 5
[... 207 lines cut ...]
2022-12-02 08:31:05,664 - twister - INFO - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
2022-12-02 08:31:05,664 - twister - DEBUG - --------------------------------
2022-12-02 08:31:05,664 - twister - DEBUG - Total test suites: 1
[... cut ...]
2022-12-02 08:31:05,665 - twister - INFO - Writing JSON report /home/tmon/zephyrproject/zephyr/twister-out/twister.json
2022-12-02 08:31:05,665 - twister - INFO - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister.xml...
2022-12-02 08:31:05,666 - twister - INFO - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister_report.xml...
2022-12-02 08:31:05,666 - twister - DEBUG - usb.device.usb_deconfig: No status
2022-12-02 08:31:05,666 - twister - DEBUG - usb.device.usb_deconfig: No status
2022-12-02 08:31:05,666 - twister - INFO - Run completed
```

desowin, 2022-12-02:
The very important detail is to have nRF USB not connected. If nRF USB is connected, everything works as expected.

desowin, 2022-12-02:
This is duplicate of #52703. That is, the actual root cause is the same, only the way the bug exhibits itself is different (hang on nRF530 DK vs "Illegal use of the EPSR" on nRF52840 DK)

closed 2022-12-02 by desowin