zephyrproject-rtos/zephyr #52673: mcux: flexcan: forever waiting semaphore in can_send()

MikhailSiomin, 2022-11-30:
If you call the `can_send()` function from several **threads** with **different priorities** with the function arguments **timeout = K_FOREVER** and **callback = NULL**, then after a certain period of time, for some reason, the thread will endlessly wait for the semaphore.

 - mimxrt1064_evk board
 - CAN baud rate 1Mbit/s
 - A thread with the highest priority hangs more often.

**To Reproduce**
`main.c`:
```c
#include <zephyr/drivers/can.h>
#include <zephyr/random/rand32.h>
#include <stdio.h>

#define THREAD_NUM 4
#define STATE_THREAD_STACK_SIZE 1024
K_KERNEL_STACK_ARRAY_DEFINE(state_stack, THREAD_NUM, STATE_THREAD_STACK_SIZE);
static struct k_thread state_thread_data[THREAD_NUM];

static int32_t threads_prior[THREAD_NUM] = {2, 4, 7, 2};

struct can_config
{
    uint8_t thr_n;
    uint8_t frame_count;
    uint32_t period;
    uint8_t prior;
};

static struct can_config can_cfg[THREAD_NUM];

void can_thread(void *arg1, void *arg2, void *arg3)
{
    const struct device *can_dev = (const struct device *)arg1;
    const struct can_config *cancfg = (const struct can_config *)arg2;

    int ret;
    struct can_frame frame = {
        .dlc = 8,
        .flags = CAN_FRAME_IDE,
        .id = 0xDEAD00 + cancfg->thr_n,
        .data = {0x00, cancfg->thr_n, cancfg->prior, 0xfa, 0x1e, 0x09, 0x37, 0xdb}
    };

    while(1)
    {
        for(uint8_t i = 0; i < cancfg->frame_count; i++)
        {
            frame.data[0] = i;
            ret = can_send(can_dev, &frame, K_FOREVER, NULL, NULL);
            if (ret != 0)
            {
                printf("CAN: Error: %d\n", ret);
                k_sleep(K_MSEC(1000));
            }
        }
        k_sleep(K_MSEC(cancfg->period));
    }
}

void main(void) 
{
    printf("\nFLEXCAN testing\n");
    printf(  "==================\n");

    const struct device *can_dev = DEVICE_DT_GET(DT_NODELABEL(flexcan2));

    if (!device_is_ready(can_dev)) {
        printf("CAN: Device %s not ready.\n", can_dev->name);
        return;
    }

    printf("device %s is ready\n", can_dev->name);

    int ret = can_start(can_dev);
    if (ret != 0) {
        printf("Error starting CAN controller [%d]\n", ret);
        return;
    }

    for (uint8_t i = 0; i < THREAD_NUM; i++)
    {
        uint32_t rand_num = sys_rand32_get();
        can_cfg[i].thr_n = i;
        can_cfg[i].frame_count = (rand_num & 0xFE) + 1;
        can_cfg[i].period = ((rand_num >> 8) & 0x7F) + 50;
        can_cfg[i].prior = threads_prior[i]; //((rand_num >> 15) & 0xF) + 1;

        printf("thread: %d, frame count: %3d, period: %3d, priority: %2d\n",
            can_cfg[i].thr_n, can_cfg[i].frame_count, can_cfg[i].period, can_cfg[i].prior);

        k_tid_t can_thr = k_thread_create(&state_thread_data[i],
                            state_stack[i],
                            K_THREAD_STACK_SIZEOF(state_stack[i]),
                            can_thread, can_dev, &can_cfg[i], NULL,
                            can_cfg[i].prior, 0,
                            K_NO_WAIT);

        if (!can_thr) {
            printf("ERROR spawning thread %d\n", 0);
            break;
        }
    }
}
```
`prj.conf`:
```
CONFIG_NO_OPTIMIZATIONS=y
CONFIG_DEBUG=y
CONFIG_DEBUG_THREAD_INFO=y

CONFIG_MAIN_STACK_SIZE=16384
CONFIG_ISR_STACK_SIZE=16384
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=16384

CONFIG_HEAP_MEM_POOL_SIZE=32768

CONFIG_CAN=y
CONFIG_CAN_MCUX_FLEXCAN=y
CONFIG_ENTROPY_GENERATOR=y
CONFIG_ENTROPY_DEVICE_RANDOM_GENERATOR=y
```

**Environment:**
 - OS: Windows 10 Pro (21H2)
 - Toolchain Zephyr SDK
 - Zephyr version 3.2.99 (commit 21a6bb3d2da332e2a6e024a6e7e38ace1f08175f)

henrikbrixandersen, 2022-12-05:
Calling `can_send()` without a timeout value of `K_FOREVER` will make the function call wait forever for an available TX mailbox. When not specifying a callback function, the `can_send()` will furthermore block until the CAN frame is transmitted on the CAN bus.

If no other CAN node ACKs the transmitted frame, it is constantly retried and the TX mailbox will be occupied forever (or until calling `can_stop()`, at least).

If I read your code and description correctly, I believe what you are describing is the documented behaviour of the CAN controller API and not a bug.

MikhailSiomin, 2022-12-06:
I did some research and found out that **threads** and the **interrupt handler** use `non-synchronized` and `non-atomic` `read-modify-write` operations on `global` resources (eg IMASK1, IMASK2 registers).

As a result, a **lower** priority thread may `overwrite` `global` data changes made by a **higher** priority thread. In the example above, the bits of the IMASK1-2 registers were `overwritten`, therefore interrupts was **not generated** and the semaphore was **not given** from the interrupt handler. Therefore, **higher** priority threads hung in endless waiting.

I suggest the following solution:
Place the `FLEXCAN_TransferSendNonBlocking()` function in the **critical section** and **disable interrupts** while it is running.

`zephyr\drivers\can\can_mcux_flexcan.c`:
```c
static int mcux_flexcan_send( ... ) {
    
    ...
    
    k_mutex_lock(&data->tx_mutex, K_FOREVER);
    irq_disable(config->irq_n);
    status = FLEXCAN_TransferSendNonBlocking(config->base, &data->handle,
                         &xfer);
    irq_enable(config->irq_n);
    k_mutex_unlock(&data->tx_mutex);

    ...

}
```
This solution has been `successfully` tested. the duration of the test was 3 days.
I can `commit` these changes.

PR #52897 (2023-01-03): mcux: flexcan: fixes the race condition
Since the FLEXCAN_TransferSendNonBlocking() function has read-modify-write operations on global resources, then if interrupts are enabled and/or if there is more than one thread with a different priority, threads with a lower priority can overwrite global resource changes made by higher priority threads.
Fixes #52673.
commit d4f65e40 (2023-01-03): mcux: flexcan: fixes the race condition

Since the FLEXCAN_TransferSendNonBlocking() function has
read-modify-write operations on global resources, then
if interrupts are enabled and/or if there is more than
one thread with a different priority, threads with
a lower priority can overwrite global resource changes
made by higher priority threads.
Fixes #52673.
closed 2023-01-03 by carlescufi