zephyrproject-rtos/zephyr #52690: coredump: stm32l5: I-cache error on coredump backend in flash

RomainPelletant, 2022-11-30:
**Describe the bug**
When enabling coredump feature with flash backend, on coredump, an error while writing on stm32l5 flash occurs.
` <err> os: Current thread: 0x200021e8 (app_tid)
[00:00:00.144,000] <err> flash_stm32l5_u5: I-cache error
[00:00:00.144,000] <err> coredump: Cannot start coredump!
[00:00:00.281,000] <err> os: Halting system
`

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? STM32L552Q

**To Reproduce**
Steps to reproduce the behavior:
1. Enable coredump in prj.conf
`CONFIG_DEBUG_COREDUMP=y
CONFIG_DEBUG_COREDUMP_BACKEND_FLASH_PARTITION=y
CONFIG_DEBUG_COREDUMP_BACKEND_LOGGING=n
CONFIG_DEBUG_COREDUMP_MEMORY_DUMP_MIN=y
CONFIG_DEBUG_COREDUMP_MEMORY_DUMP_LINKER_RAM=n`
2. In board dts, configure flash partition
```
&flash0 {
    partitions {
        compatible = "fixed-partitions";
        #address-cells = <1>;
        #size-cells = <1>;
    
        zephyr_code: partition@0 {
            label = "zephyr-code";
            reg = <0x00000000 0x0007d000>;
        };
        coredump_partition: partition@7d000 {
            label = "coredump-partition";
            reg = <0x0007d000 0x00003000>;
        };
    };
};
```
4. make & run
5. See error

**Expected behavior**
Coredump shall be written into flash

erwango, 2022-12-19:
@RomainPelletant, I've try to reproduce the error with the following:

Sample: `samples/drivers/flash_shell`
I added a `k_oops()` in `cmd_read()` of `drivers/flash/flash_shell.c`.
Target: `nucleo_l552ze_q` (with tuned flash partition)

Unfortunately, I didn't reproduced the error mentionned:
```
*** Booting Zephyr OS build zephyr-v3.2.0-2703-g954ffc38d300 ***
Found flash controller flash-controller@40022000.
Flash I/O commands can be run.
uart:~$ flash read 0 1 
[00:00:03.031,000] <dbg> flash_stm32: flash_stm32_write_protection: Disable write protection
[00:00:03.031,000] <dbg> flash_stm32l5_u5: stm32_icache_disable: I-cache Disable
[00:00:03.032,000] <dbg> flash_stm32l5_u5: icache_wait_for_invalidate_complete: I-cache Invalidation complee
[00:00:03.032,000] <dbg> flash_stm32l5_u5: stm32_icache_enable: I-cache Enable
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write_protection: Enable write protection
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write: Write offset: 256288, len: 8
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write_protection: Disable write protection
[00:00:03.032,000] <dbg> flash_stm32l5_u5: stm32_icache_disable: I-cache Disable
[00:00:03.032,000] <dbg> flash_stm32l5_u5: icache_wait_for_invalidate_complete: I-cache Invalidation complee
[00:00:03.032,000] <dbg> flash_stm32l5_u5: stm32_icache_enable: I-cache Enable
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write_protection: Enable write protection
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write: Write offset: 256296, len: 8
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write_protection: Disable write protection
[00:00:03.032,000] <dbg> flash_stm32l5_u5: stm32_icache_disable: I-cache Disable
[00:00:03.032,000] <dbg> flash_stm32l5_u5: icache_wait_for_invalidate_complete: I-cache Invalidation complee
[00:00:03.032,000] <dbg> flash_stm32l5_u5: stm32_icache_enable: I-cache Enable
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write_protection: Enable write protection
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write: Write offset: 256304, len: 8
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write_protection: Disable write protection
[00:00:03.032,000] <dbg> flash_stm32l5_u5: stm32_icache_disable: I-cache Disable
[00:00:03.032,000] <dbg> flash_stm32l5_u5: icache_wait_for_invalidate_complete: I-cache Invalidation complee
[00:00:03.032,000] <dbg> flash_stm32l5_u5: stm32_icache_enable: I-cache Enable
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write_protection: Enable write protection
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write: Write offset: 253952, len: 16
[00:00:03.032,000] <dbg> flash_stm32: flash_stm32_write_protection: Disable write protection
[00:00:03.032,000] <dbg> flash_stm32l5_u5: stm32_icache_disable: I-cache Disable
[00:00:03.033,000] <dbg> flash_stm32l5_u5: icache_wait_for_invalidate_complete: I-cache Invalidation complee
[00:00:03.033,000] <dbg> flash_stm32l5_u5: stm32_icache_enable: I-cache Enable
[00:00:03.033,000] <dbg> flash_stm32: flash_stm32_write_protection: Enable write protection
```

RomainPelletant, 2022-12-19:
@erwango Thanks you for investigating.
The error occurs only when using coredump feature directly writing into internal flash.
Maybe something disabled after coredump occured.
Do you want a sample program to reproduce it more easily and fastly?

RomainPelletant, 2022-12-19:
@erwango it is not a showstopper so let me investigating more deeply and I will let you know in few weeks :)
Thanks

erwango, 2022-12-19:
> The error occurs only when using coredump feature directly writing into internal flash.

Ok, This is what I tried to do by adding a `k_oops()` when running `flash_shell` sample, but maybe this isn't exactly similar to your use case.

> Do you want a sample program to reproduce it more easily and fastly?

Would be nice, txs

RomainPelletant, 2023-01-13:
Retested with the main branch : no more issues => close

RomainPelletant, 2023-10-20:
@erwango this time I got the same on u5
I found a workaround (in attachment) by increasing timeout and clearing err flag.
Only on coredump write. 
[007_workaround_coredump.patch](https://github.com/zephyrproject-rtos/zephyr/files/13055751/007_workaround_coredump.patch)

erwango, 2024-01-16:
> @erwango this time I got the same on u5 I found a workaround (in attachment) by increasing timeout and clearing err flag. Only on coredump write. [007_workaround_coredump.patch](https://github.com/zephyrproject-rtos/zephyr/files/13055751/007_workaround_coredump.patch)

I had a look to the proposed workaround and even if working, I'm not sure this should be integrated as basically you discard ICache errors and report status ok.
This being said, there might be something better than reporting -EIO on that kind of errors as ICache should be disabled and we're running the cache invalidation process. I'll investigate a bit deeper and get back to you.

closed 2024-06-01 by github-actions[bot]