zephyrproject-rtos/zephyr #52691: cannot read octospi flash when address exceeds 4mb

bdunlay, 2022-11-30:
**Describe the bug**
I am trying to use littlefs with the the 'store_partition' of the quad-spi flash peripheral using the octospi driver for the b_l4s5i_iot01a. The default size of the `store_partition` is 8MB. When littlefs tries to mount the filesystem, it reads the flash memory and results in an error:

```
[00:00:00.905,000] <err> flash_stm32_ospi: Error: address or size exceeds expected values: addr 0x3f2000, size 16
[00:00:00.905,000] <err> fs: failed to create directory (-22)
```

If I change the `store_partition` size to 4MB or less, this issue goes away.

 - What target platform are you using? stm32 b_l4s5i_iot01a
 - What have you tried to diagnose or workaround this issue? Changing the flash partition size to 4mb or less
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? No, octospi flash support for this device was just added recently: https://github.com/zephyrproject-rtos/zephyr/issues/52276
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. use littlefs sample with `store_partition` (the partition defined for the external flash memory)

**Expected behavior**
LFS mounts and lists the directory with no errors, all 8MB usable

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

Annoyance. 50% loss of usable space. Still have 4MB at least.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr sdk 0.15.1 with octospi commits cherry-picked from this PR:  https://github.com/zephyrproject-rtos/zephyr/pull/52374

GeorgeCGV, 2022-12-01:
@bdunlay hello, at first glance there shall be no issues for

```
flash_size = 0x800000
addr = 0x3F2000 
size = 16
```
https://github.com/zephyrproject-rtos/zephyr/blob/56284d7017dc593f18c3aaa128d2c4e596a8e4c8/drivers/flash/flash_stm32_ospi.c#L486

The board DTS provides flash size as `DT_SIZE_M(8)`, the result is `8388608` in **bytes**.  Unfortunately, the driver divides `flash_size` by `8` as well to convert expected `bits` to `bytes`.

https://github.com/zephyrproject-rtos/zephyr/blob/56284d7017dc593f18c3aaa128d2c4e596a8e4c8/dts/bindings/flash_controller/st%2Cstm32-ospi-nor.yaml#L32-L34

https://github.com/zephyrproject-rtos/zephyr/blob/56284d7017dc593f18c3aaa128d2c4e596a8e4c8/drivers/flash/flash_stm32_ospi.c#L2168

That results in `flash_size` to be `1048576` (`0x100000`) instead of desired `8388608` (`0x800000`). Therefore, `ospi_address_is_valid` fails. Unfortunately, that affects `QSPI` driver as well.

@bdunlay could you try to provide flash size as `0x4000000` and try again?

closed 2022-12-02 by carlescufi

bdunlay, 2022-12-02:
Thank you @GeorgeCGV for the detailed explanation and quick fix. I've pulled in your change and verified that now my littlefs code is working as expected with the fully available address space.

PR #52706 (2022-12-02): boards: arm: b_l4s5i_iot01a and stm32h7b3i_dk: fix OSPI flash size
The OSPI flash driver expects size in bits. This PR uses bits instead of bytes for the size.

Fixes #52691.