zephyrproject-rtos/zephyr #52695: Changes to network stack to facilitate iface-agnostic application code

glarsennordic, 2022-11-30:
# Introduction / Request For Comments
This proposal/issue proposes a set of changes to the Zephyr networking stack that would enable developers to write application code that is maximally iface/L2-agnostic. These changes could significantly reduce boilerplate overhead for Zephyr end-users writing connected applications **_even when only a single iface is used._**

I plan to present and discuss this proposal at the upcoming Zephyr Networking Summit on December 6th. It is my hope that this issue serves as a place to discuss and adjust the proposal both leading up to that event and following it. Feedback and comments from the Zephyr network maintainers would be deeply appreciated!

### Contributors:
- @maxd-nordic
- @wbober
- @rlubos
- @mgolu

### Problem description

Currently, applications that require multiple ifaces or L2-stacks must be aware of and manage each iface individually. Furthermore, Zephyr offers no standardized interface to configure or initiate a second association stage on ifaces that require it (for instance, Wi-Fi). Instead, applications must call iface-specific net_mgmt APIs to initiate and configure association.

Many applications only need direct knowledge of or control over whether L4 connectivity is established. Direct management of L2-stacks and ifaces is beyond the scope of such applications, and yet, is currently required by Zephyr's network APIs. Zephyr does currently provide [`conn_mgr`](https://github.com/nrfconnect/sdk-zephyr/tree/main/subsys/net/lib/conn_mgr), which alerts applications to changes in L4 connectivity across all ifaces simultaneously, but no similarly high-level mechanism exists to **_control_** connectivity.

All of this introduces a significant amount of connection-related overhead to applications that might otherwise be avoided if higher-level control APIs were provided. Even applications that require only a single iface might benefit from not needing to deal with the internal specifics of that iface's connection process.

### Proposed solution
- Expand the already-existing [`conn_mgr`](https://github.com/nrfconnect/sdk-zephyr/tree/main/subsys/net/lib/conn_mgr) module to serve as a simple multi-casting controller for all ifaces
- Introduce a number of new functions to the net_if API struct to facilitate generic association/connection on ifaces that need it (such as Wi-Fi). Namely:
  - `net_if_connect`
    - Including standardized timeout and persistence options
  - `net_if_disconnect`
  - `net_if_set_opt`
  - `net_if_get_opt`
- The addition of active association/connection as a necessary condition for entering the operational-up state on ifaces that require association/connection.
- The introduction of a new network interface flag to control whether `net_if_connect` is called automatically when entering the admin-up state: `NET_IF_NO_AUTO_CONNECT`

### Background
This issue/proposal is heavily based on some recent changes to the Zephyr networking stack, namely the introduction of the operational/admin state separation. See the following resources if you are not already familiar with this change.
- (net: if: Improve interface status reporting)[https://github.com/zephyrproject-rtos/zephyr/pull/51240]
- (net: if: Introduce carrier and dormant management on a network iface )[https://github.com/zephyrproject-rtos/zephyr/commit/efd03a86d77a2e23d9e48d15b3a84cbfeeaed51d#]
- (net: if: extend list of admin/operational states of network interface)[https://github.com/zephyrproject-rtos/zephyr/issues/28998]
- (Kernel subsystem documentation » Networking » Operational States)[https://www.kernel.org/doc/html/latest/networking/operstates.html]
- (RFC2863)[https://tools.ietf.org/html/rfc2863]

### Scope / Requirements
Applications require iface-agnostic tools to perform the following tasks:
  - L2 connectivity establishment (with timeout)
    - Allow for establishing persistent connectivity mostly without application involvement, and in a manner that is L2-stack-agnostic from the application's perspective.
    - Include the capacity to specify a timeout for connection attempts.
  - Control over L2 connectivity persistence
    - L2 stacks that support persistent(fault-tolerant and self-reconnecting) connectivity should offer the option to disable this persistence
  - L4 readiness monitoring by application
    - Allow for monitoring the status of IP connectivity as a whole.
  - L2 irrecoverable error notification
    - Applications should be informed if an unrecoverable error is preventing one or more L2 stacks from connecting.

Anything beyond these needs or their implementation is considered beyond the scope of this proposal.

# Proposed Changes (Detailed)

### Changes to [`conn_mgr`](https://github.com/nrfconnect/sdk-zephyr/tree/main/subsys/net/lib/conn_mgr):
#### Added net_mgmt requests:
  - `NET_REQUEST_ALL_IF_UP`: 
    - Calls `net_if_up()` on all available network interfaces
  - `NET_REQUEST_ALL_IF_DOWN`: 
    - Calls `net_if_down()` on all available network interfaces
  - `NET_REQUEST_ALL_IF_CONNECT`:
    - Calls `NET_REQUEST_ALL_IF_UP`, then calls (the herein proposed) `net_if_connect()` on all available network interfaces
  - `NET_REQUEST_ALL_IF_DISCONNECT`:
    - Calls (the herein proposed) `net_if_disconnect()` on all available network interfaces

### Changes to `net_if`:

We propose the addition of two new generic API functions to the `net_if` API: `net_if_connect()` and `net_if_disconnect()`, as well as the addition of a new network interface flag, `NET_IF_NO_AUTO_CONNECT`, and two new options customization API functions, `net_if_set_opt` and `net_if_get_opt()`

#### `int net_if_connect(struct net_if *iface)`:
[... 51 lines cut ...]

[image]

### `NET_IF_NO_AUTO_START` and `NET_IF_NO_AUTO_CONNECT` enabled:
In other words, neither auto-start nor auto-connect.

[image]

### `NET_IF_NO_AUTO_START` enabled and `NET_IF_NO_AUTO_CONNECT` disabled:
In other words, auto-connect without auto-start.

In this configuration the Application can either call `NET_REQUEST_ALL_IF_CONNECT`, or `NET_REQUEST_ALL_IF_UP`, and the result will be the same, but the sequence diagrams will be different.

The behavior if `NET_REQUEST_ALL_IF_CONNECT` is called is already demonstrated in the previous sequence diagrams, so we will only show the `NET_REQUEST_ALL_IF_UP` case.

[image]

# Alternatives / Existing Solutions

To our knowledge, there are not any existing solutions in Zephyr that solve this problem.

Alternative _potential_ solutions include:

1. Implementing agnostic connectivity as a disjoint library rather than as a direct feature of the Zephyr network stack. Without additions to the `net_if` API, such a library would require specific knowledge of each iface it supports, and therefore could not support all ifaces generically.
2. (downstream-only / for Nordic use only): Provide wrappers for network interfaces that map their events and controls into the Common Application Framework, and use this as an iface-agnostic control layer. Once again, generic control of ifaces would be impossible, and such a solution would be limited to only a small set of supported ifaces.

wbober, 2022-12-02:
LGTM

jukkar, 2022-12-05:
> #### `int net_if_connect(struct net_if *iface)`:
> 
>     * Need not be implemented on ifaces that do not have an association/connect phase

So I suppose for example for Ethernet, this function would return 0 (success) immediately if the cable is connected. What would be returned if cable is not connected, -ENETDOWN perhaps?
Or do you mean that this function could return -ENOTSUPP or similar for some L2 technologies?

> ### Changes to [`conn_mgr`](https://github.com/nrfconnect/sdk-zephyr/tree/main/subsys/net/lib/conn_mgr):
> #### Added net_mgmt requests:
>
>     * `NET_REQUEST_ALL_IF_UP`:
>
>       * Calls [...]

Hmm, just pondering here what will the API look like because there is the `iface` parameter in the `net_mgmt` API  but these `_ALL_` variant flags indicate that the operation is for **all** the interfaces. 

Anyway, looking forward to review the PR.

jukkar, 2022-12-05:
> Maybe it would make sense for these to not be NET_MGMT requests at all, and just direct API calls?

I was actually pondering the same thing. A separate API call could go through all the network interfaces and call net_mgmt() with suitable parameters. So instead of **ALL** variant of the net_mgmt request, you could have a request that is set to just one interface.

glarsennordic, 2022-12-15:
Oh... every single force push is creating a reference 🤦

glarsennordic, 2023-01-20:
As discussed at the aforementioned Zephyr Networking Forum (workgroup?) meeting, I've written an implementation of this proposal here:

https://github.com/zephyrproject-rtos/zephyr/pull/53118

This should hopefully make discussing this proposal much less abstract!

Note that this implementation differs in some ways from the PR! I will write some form of update soon with the details of what is different.

closed 2024-02-08 by glarsennordic