zephyrproject-rtos/zephyr #52703: tests: subsys: usb: device: usb.device USAGE FAULT exception

nordic-piks, 2022-12-01:
**Describe the bug**
The zephyr/tests/subsys/usb/device/usb.device fails

Observed for
- nrf5340dk_nrf5340_cpuapp
- nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/subsys/usb/device  -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-11-30T23:52:29.023Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:52:29.023Z] [0mWatchdog sample application[0m
[2022-11-30T23:52:29.023Z] [0mCallback in RESET_SOC disabled for this platform[0m
[2022-11-30T23:52:29.023Z] [0mFeeding watchdog 5 times[0m
[2022-11-30T23:52:29.023Z] [0mFeeding watchdog...[0m
[2022-11-30T23:52:29.023Z] [0mD: __usb_descriptor_start 0x2000012c[0m
[2022-11-30T23:52:29.023Z] [0mD: __usb_descriptor_end 0x2000019d[0m
[2022-11-30T23:52:29.023Z] [0mD: Configuration descriptor 0x2000013e[0m
[2022-11-30T23:52:29.023Z] [0mD: Interface descriptor 0x20000147[0m
[2022-11-30T23:52:29.024Z] [0mD: Endpoint descriptor 0x20000150[0m
[2022-11-30T23:52:29.024Z] [0mD: ep 0x81, mps 64, type 2[0m
[2022-11-30T23:52:29.024Z] [0mD: Fixing EP address 81 -> 81[0m
[2022-11-30T23:52:29.024Z] [0mD: endpoint 0x81[0m
[2022-11-30T23:52:29.024Z] [0mD: Now the wTotalLength is 25[0m
[2022-11-30T23:52:29.024Z] [0mD: idx_max 11, ascii_idx_max 5, buf 0x2000015d[0m
[2022-11-30T23:52:29.024Z] [0mD: char R : 52, idx 5 -> 11[0m
[2022-11-30T23:52:29.024Z] [0mD: char Y : 59, idx 4 -> 9[0m
[2022-11-30T23:52:29.024Z] [0mD: char H : 48, idx 3 -> 7[0m
[2022-11-30T23:52:29.024Z] [0mD: char P : 50, idx 2 -> 5[0m
[2022-11-30T23:52:29.024Z] [0mD: char E : 45, idx 1 -> 3[0m
[2022-11-30T23:52:29.024Z] [0mD: char Z : 5a, idx 0 -> 1[0m
[2022-11-30T23:52:29.024Z] [0mD: idx_max 13, ascii_idx_max 6, buf 0x2000016b[0m
[2022-11-30T23:52:29.024Z] [0mD: char V : 56, idx 6 -> 13[0m
[2022-11-30T23:52:29.024Z] [0mD: char E : 45, idx 5 -> 11[0m
[2022-11-30T23:52:29.024Z] [0mD: char D : 44, idx 4 -> 9[0m
[2022-11-30T23:52:29.024Z] [0mD: char - : 2d, idx 3 -> 7[0m
[2022-11-30T23:52:29.024Z] [0mD: char B : 42, idx 2 -> 5[0m
[2022-11-30T23:52:29.024Z] [0mD: char S : 53, idx 1 -> 3[0m
[2022-11-30T23:52:29.024Z] [0mD: char U : 55, idx 0 -> 1[0m
[2022-11-30T23:52:29.024Z] [0mD: Serial Number[0m
[2022-11-30T23:52:29.024Z] [0mD: 47 5a c4 c0 1d 7e a0 57 |GZ...~.W[0m
[2022-11-30T23:52:29.024Z] [0mD: idx_max 31, ascii_idx_max 15, buf 0x2000017b[0m
[2022-11-30T23:52:29.024Z] [0mD: char 7 : 37, idx 15 -> 31[0m
[2022-11-30T23:52:29.024Z] [0mD: char 5 : 35, idx 14 -> 29[0m
[2022-11-30T23:52:29.024Z] [0mD: char 0 : 30, idx 13 -> 27[0m
[2022-11-30T23:52:29.024Z] [0mD: char A : 41, idx 12 -> 25[0m
[2022-11-30T23:52:29.024Z] [0mD: char E : 45, idx 11 -> 23[0m
[2022-11-30T23:52:29.024Z] [0mD: char 7 : 37, idx 10 -> 21[0m
[2022-11-30T23:52:29.024Z] [0mD: char D : 44, idx 9 -> 19[0m
[2022-11-30T23:52:29.024Z] [0mD: char 1 : 31, idx 8 -> 17[0m
[2022-11-30T23:52:29.024Z] [0mD: char 0 : 30, idx 7 -> 15[0m
[2022-11-30T23:52:29.024Z] [0mD: char C : 43, idx 6 -> 13[0m
[2022-11-30T23:52:29.024Z] [0mD: char 4 : 34, idx 5 -> 11[0m
[2022-11-30T23:52:29.024Z] [0mD: char C : 43, idx 4 -> 9[0m
[2022-11-30T23:52:29.024Z] [0mD: char A : 41, idx 3 -> 7[0m
[2022-11-30T23:52:29.024Z] [0mD: char 5 : 35, idx 2 -> 5[0m
[2022-11-30T23:52:29.024Z] [0mD: char 7 : 37, idx 1 -> 3[0m
[2022-11-30T23:52:29.024Z] [0mD: char 4 : 34, idx 0 -> 1[0m
[2022-11-30T23:52:29.024Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:52:29.024Z] [0mRunning TESTSUITE device_usb[0m
[2022-11-30T23:52:29.024Z] [0m===================================================================[0m
[2022-11-30T23:52:29.024Z] [0mD: lock usb_enable_lock mutex[0m
[2022-11-30T23:52:29.024Z] [0mD: set cb, ep: 0x81[0m
[2022-11-30T23:52:29.024Z] [0mD: EP enable: 0x00[0m
[2022-11-30T23:52:29.024Z] [0mD: EP enable: 0x80[0m
[2022-11-30T23:52:29.024Z] [0mD: unlock usb_enable_lock mutex[0m
[2022-11-30T23:52:29.024Z] [0mSTART - test_usb_dc_api[0m
[2022-11-30T23:52:29.024Z] [0mPASS - test_usb_dc_api in 0.001 seconds[0m
[2022-11-30T23:52:29.024Z] [0m===================================================================[0m
[2022-11-30T23:52:29.024Z] [0mSTART - test_usb_dc_api_invalid[0m
[2022-11-30T23:52:29.024Z] [0mD: ep_write: ep 0x20, len 1[0m
[2022-11-30T23:52:29.024Z] [0mD: ep_read: ep 0x20, maxlen 1[0m
[2022-11-30T23:52:29.024Z] [0mPASS - test_usb_dc_api_invalid in 0.006 seconds[0m
[2022-11-30T23:52:29.024Z] [0m===================================================================[0m
[2022-11-30T23:52:29.024Z] [0mSTART - test_usb_dc_api_read_write[0m
[2022-11-30T23:52:29.024Z] [0mD: ep_read: ep 0x20, maxlen 1[0m
[2022-11-30T23:52:29.024Z] [0mD: ep_write: ep 0x20, len 1[0m
[2022-11-30T23:52:29.024Z] [0mPASS - test_usb_dc_api_read_write in 0.006 seconds[0m
[2022-11-30T23:52:29.024Z] [0m===================================================================[0m
[2022-11-30T23:52:29.024Z] [0mSTART - test_usb_deconfig[0m
[2022-11-30T23:52:29.024Z] [0mPASS - test_usb_deconfig in 0.001 seconds[0m
[2022-11-30T23:52:29.024Z] [0m===================================================================[0m
[2022-11-30T23:52:29.024Z] [0mSTART - test_usb_disable[0m
[2022-11-30T23:52:29.024Z] [0mD: EP disable: 0x00[0m
[2022-11-30T23:52:29.024Z] [0mD: EP disable: 0x80[0m
[2022-11-30T23:52:29.024Z] [0mE: ***** USAGE FAULT *****[0m
[2022-11-30T23:52:29.024Z] [0mE:   Illegal use of the EPSR[0m
[2022-11-30T23:52:29.024Z] [0mE: r0/a1:  0x20001c1c  r1/a2:  0x00000000  r2/a3:  0x00000006[0m
[2022-11-30T23:52:29.024Z] [0mE: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00008629[0m
[2022-11-30T23:52:29.024Z] [0mE:  xpsr:  0x00000000[0m
[2022-11-30T23:52:29.024Z] [0mE: Faulting instruction address (r15/pc): 0x00000000[0m
[2022-11-30T23:52:29.024Z] [0mE: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[0m
[2022-11-30T23:52:29.024Z] [0mE: Current thread: 0x200001c8 (test_usb_disable)[0m
[2022-11-30T23:52:29.024Z] [0mE: Halting system[0m
[2022-11-30T23:52:29.024Z] [0m[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.1
 - Commit SHA or Version used: zephyr-v3.2.0-2123-g14358aee1cda

desowin, 2022-12-02:
I cannot reproduce this issue.

```
[tmon@tmon-laptop zephyr]$ git log -1 --format=%H
14358aee1cda16b50b5e931101d653a744da9c17
[tmon@tmon-laptop zephyr]$ ./scripts/twister -T tests/subsys/usb/device -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
ZEPHYR_BASE unset, using "/home/tmon/zephyrproject/zephyr"
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.2.0-2123-g14358aee1cda
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/tmon/zephyrproject/zephyr/twister-out/testplan.json

Device testing on:

| Platform            | ID   | Serial device   |
|---------------------|------|-----------------|
| nrf52840dk_nrf52840 |      | /dev/ttyACM0    |

INFO    - JOBS: 12
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/1 nrf52840dk_nrf52840       zephyr/tests/subsys/usb/device/usb.device          PASSED (device 6.249s)

INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 12.30 seconds
INFO    - In total 5 test cases were executed, 0 skipped on 1 out of total 518 platforms (0.19%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board               | ID   |   Counter |
|---------------------|------|-----------|
| nrf52840dk_nrf52840 |      |         1 |
INFO    - Saving reports...
INFO    - Writing JSON report /home/tmon/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

desowin, 2022-12-02:
```
2022-12-02 07:43:19,872 - twister - INFO - Using Ninja..
2022-12-02 07:43:19,888 - twister - INFO - Zephyr version: zephyr-v3.2.0-2123-g14358aee1cda
2022-12-02 07:43:19,890 - twister - DEBUG - Running cmake script /home/tmon/zephyrproject/zephyr/cmake/verify-toolchain.cmake
2022-12-02 07:43:19,890 - twister - DEBUG - Calling cmake: /usr/bin/cmake -DFORMAT=json -P /home/tmon/zephyrproject/zephyr/cmake/verify-toolchain.cmake
2022-12-02 07:43:20,228 - twister - DEBUG - Finished running  /home/tmon/zephyrproject/zephyr/cmake/verify-toolchain.cmake
2022-12-02 07:43:20,228 - twister - INFO - Using 'zephyr' toolchain.
2022-12-02 07:43:20,305 - twister - DEBUG - Reading test case configuration files under /home/tmon/zephyrproject/zephyr/tests/subsys/usb/device...
2022-12-02 07:43:20,305 - twister - DEBUG - Found possible testsuite in /home/tmon/zephyrproject/zephyr/tests/subsys/usb/device
2022-12-02 07:43:20,307 - twister - DEBUG - Reading platform configuration files under /home/tmon/zephyrproject/zephyr/boards...
2022-12-02 07:43:20,655 - twister - DEBUG - Reading platform configuration files under /home/tmon/zephyrproject/zephyr/scripts/pylib/twister/boards...
2022-12-02 07:43:20,656 - twister - DEBUG - platform filter: ['nrf52840dk_nrf52840']
2022-12-02 07:43:20,656 - twister - DEBUG -     arch_filter: None
2022-12-02 07:43:20,656 - twister - DEBUG -      tag_filter: None
2022-12-02 07:43:20,656 - twister - DEBUG -     exclude_tag: None
2022-12-02 07:43:20,656 - twister - INFO - Building initial testsuite list...
2022-12-02 07:43:20,656 - twister - INFO - Writing JSON report /home/tmon/zephyrproject/zephyr/twister-out/testplan.json
2022-12-02 07:43:20,662 - twister - INFO - JOBS: 12
2022-12-02 07:43:20,663 - twister - INFO - Adding tasks to the queue...
2022-12-02 07:43:20,663 - twister - DEBUG - adding nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device
2022-12-02 07:43:20,663 - twister - INFO - Added initial list of jobs to queue
2022-12-02 07:43:20,664 - twister - DEBUG - Launch process 0
2022-12-02 07:43:20,664 - twister - DEBUG - Launch process 1
2022-12-02 07:43:20,666 - twister - DEBUG - Launch process 2
2022-12-02 07:43:20,667 - twister - DEBUG - Launch process 3
2022-12-02 07:43:20,668 - twister - DEBUG - Running cmake on /home/tmon/zephyrproject/zephyr/tests/subsys/usb/device for nrf52840dk_nrf52840
2022-12-02 07:43:20,669 - twister - DEBUG - Calling cmake: /usr/bin/cmake -B/home/tmon/zephyrproject/zephyr/twister-out/nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device -DTC_RUNID=f5c4157fdd7b0e4860e327c20929d38e -DEXTRA_CFLAGS=-Werror '-DEXTRA_AFLAGS=-Werror -Wa,--fatal-warnings' -DEXTRA_LDFLAGS=-Wl,--fatal-warnings -DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror -GNinja -S/home/tmon/zephyrproject/zephyr/tests/subsys/usb/device -DBOARD=nrf52840dk_nrf52840
2022-12-02 07:43:20,668 - twister - DEBUG - Launch process 4
2022-12-02 07:43:20,670 - twister - DEBUG - Launch process 5
2022-12-02 07:43:20,672 - twister - DEBUG - Launch process 6
2022-12-02 07:43:20,673 - twister - DEBUG - Launch process 7
2022-12-02 07:43:20,674 - twister - DEBUG - Launch process 8
2022-12-02 07:43:20,675 - twister - DEBUG - Launch process 9
2022-12-02 07:43:20,676 - twister - DEBUG - Launch process 10
2022-12-02 07:43:20,678 - twister - DEBUG - Launch process 11
2022-12-02 07:43:22,548 - twister - DEBUG - Finished building /home/tmon/zephyrproject/zephyr/tests/subsys/usb/device for nrf52840dk_nrf52840
2022-12-02 07:43:22,550 - twister - DEBUG - build test: nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device
2022-12-02 07:43:22,550 - twister - DEBUG - Building /home/tmon/zephyrproject/zephyr/tests/subsys/usb/device for nrf52840dk_nrf52840
2022-12-02 07:43:24,779 - twister - DEBUG - Determine test cases for test instance: nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device
2022-12-02 07:43:24,779 - twister - DEBUG - zephyr.symbols found: /home/tmon/zephyrproject/zephyr/twister-out/nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device/zephyr/zephyr.symbols
2022-12-02 07:43:24,779 - twister - DEBUG - Determine test cases for test suite: usb.device
2022-12-02 07:43:24,780 - twister - DEBUG - Test instance nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device already has 5 cases.
2022-12-02 07:43:24,781 - twister - DEBUG - run test: nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device
2022-12-02 07:43:24,781 - twister - DEBUG - Using serial device /dev/ttyACM0 @ 115200 baud
2022-12-02 07:43:25,922 - twister - DEBUG - Flash command: ['ninja', '-C', '/home/tmon/zephyrproject/zephyr/twister-out/nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device', 'flash']
2022-12-02 07:43:31,551 - twister - DEBUG - ninja: Entering directory `/home/tmon/zephyrproject/zephyr/twister-out/nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device'
[0/1] Flashing nrf52840dk_nrf52840
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner nrfjprog
Using board 1050229242
-- runners.nrfjprog: Flashing file: /home/tmon/zephyrproject/zephyr/twister-out/nrf52840dk_nrf52840/zephyr/tests/subsys/usb/device/usb.device/zephyr/zephyr.hex

[ ######               ]   0.000s | Erase file - Check image                                                           

[ ####                 ]   0.000s | Check image validity - Initialize device info                                      

[... cut ...]
2022-12-02 07:43:32,176 - twister - INFO - Writing JSON report /home/tmon/zephyrproject/zephyr/twister-out/twister.json
2022-12-02 07:43:32,177 - twister - INFO - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister.xml...
2022-12-02 07:43:32,178 - twister - INFO - Writing xunit report /home/tmon/zephyrproject/zephyr/twister-out/twister_report.xml...
2022-12-02 07:43:32,178 - twister - DEBUG - usb.device.usb_deconfig: No status
2022-12-02 07:43:32,178 - twister - DEBUG - usb.device.usb_deconfig: No status
2022-12-02 07:43:32,178 - twister - INFO - Run completed
```

desowin, 2022-12-02:
The issue is essentially calling a NULL pointer (`m_event_handler`) in `usbd_ep_abort()` in `modules/hal/nordic/nrfx/drivers/src/nrfx_usbd.c`. The global variable `m_event_handler` is set to NULL in `nrfx_usbd_uninit()` called by `usb_dc_detach()` called by `usb_disable()`.

closed 2022-12-05 by carlescufi