zephyrproject-rtos/zephyr #52705: RNDIS fails to enumerate on Raspberry Pi Pico

msasdk, 2022-12-01:
**Describe the bug**
Dumb HTTP Server example with CONFIG_USB_DEVICE_NETWORK_RNDIS=y fails to enumerate.
Same example with ECM works fine. Also, both RNDIS and ECM on STM32 targets. 

**To Reproduce**
Steps to reproduce the behavior:
`west build -p -b rpi_pico -- -DOVERLAY_CONFIG="overlay-netusb.conf"`

**Logs and console output**
RNDIS
```
[  +0,324812] usb 2-1: new full-speed USB device number 35 using xhci_hcd
[  +0,150217] usb 2-1: New USB device found, idVendor=2fe3, idProduct=0100, bcdDevice= 3.02
[  +0,000015] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000005] usb 2-1: Product: USB-DEV
[  +0,000005] usb 2-1: Manufacturer: ZEPHYR
[  +0,000003] usb 2-1: SerialNumber: E66038B7137A7B33
[  +5,253850] rndis_host 2-1:1.0: RNDIS_MSG_QUERY(0x00010202) failed, -110
[  +5,119961] rndis_host 2-1:1.0: RNDIS_MSG_QUERY(0x01010101) failed, -110
[  +0,000027] rndis_host 2-1:1.0: rndis get ethaddr, -110
[  +5,120014] rndis_host: probe of 2-1:1.0 failed with error -110
[  +5,119995] rndis_wlan 2-1:1.0: RNDIS init failed, -110
[  +0,000201] rndis_wlan: probe of 2-1:1.0 failed with error -110
```

ECM
```
[  +0,313240] usb 2-1: new full-speed USB device number 37 using xhci_hcd
[  +0,149581] usb 2-1: New USB device found, idVendor=2fe3, idProduct=0100, bcdDevice= 3.02
[  +0,000015] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000005] usb 2-1: Product: USB-DEV
[  +0,000004] usb 2-1: Manufacturer: ZEPHYR
[  +0,000004] usb 2-1: SerialNumber: E66038B7137A7B33
[  +0,005005] cdc_ether 2-1:1.0 eth0: register 'cdc_ether' at usb-0000:00:14.0-1, CDC Ethernet Device, 00:00:5e:00:53:01
[  +0,022384] cdc_ether 2-1:1.0 enx00005e005301: renamed from eth0
```

**Environment (please complete the following information):**
 - OS: Build OS Linux. Fails to enumerate on both Linux and Windows
 - Zephyr main

msasdk, 2022-12-07:
@yonsch Yes sorry if this was not clear.

I modified socket_dumb_http.c by including
```
#include <zephyr/usb/usb_device.h>
#include <zephyr/net/net_config.h>
```
and adding following on top of main()
```
 #if defined(CONFIG_USB_DEVICE_STACK)
    int ret_usb;
    ret_usb = usb_enable(NULL);
    if (ret_usb != 0) {
        printk("usb enable error %d\n", ret_usb);
    }
    (void)net_config_init_app(NULL, "Initializing network");
    #endif /* CONFIG_USB_DEVICE_STACK */
```

Further, toggle between ECM and RNDIS with following configs in overlay-netusb.conf
```
CONFIG_USB_DEVICE_NETWORK_ECM=n
CONFIG_USB_DEVICE_NETWORK_RNDIS=y
```

yonsch, 2022-12-08:
Sharing my findings so far:
By analyzing the USB captures sent by @msasdk , I figured that the pico responds to the 0x10202 command with ENOENT (no such file or directory). After lots of debugging I found out that the USB thread of the driver does not do anything after calling the endpoint callback:

https://github.com/zephyrproject-rtos/zephyr/blob/997f5edf1a43252ffa644f471dddbfe7f9d99e47/drivers/usb/device/usb_dc_rpi_pico.c#L720-L742

On the STM32 driver, I found this snippet:

https://github.com/zephyrproject-rtos/zephyr/blob/997f5edf1a43252ffa644f471dddbfe7f9d99e47/drivers/usb/device/usb_dc_stm32.c#L1133-L1142

So I implemented something similar:
```
if (USB_EP_DIR_IS_OUT(msg.ep) && setup_wlength > 0) {
	usb_dc_ep_start_read(msg.ep, setup_wlength);
	setup_wlength = 0;
}
```
Since I don't have access to `setup` like ST, I passed the wLength through a global, that I set on https://github.com/zephyrproject-rtos/zephyr/blob/997f5edf1a43252ffa644f471dddbfe7f9d99e47/subsys/usb/device/usb_device.c#L328 
This was just to see if it works, but I'll need a better solution later.

That did work for the 0x10202 packet, so I suspect I'm close. But the next packet (0x01010101) failed the same way.

Now I'm stuck. @jfischer-no @petejohanson maybe you have any ideas? I'm not that much of a USB expert

jfischer-no, 2022-12-14:
@yonsch I can reproduce it, it seems that in specific situations the controller cannot receive packets for the control OUT endpoint. Maybe DATA0/DATA1 PID issue, just a guess. I need to look at it closely.

purdeaandrei, 2023-01-28:
I have got this working.
You will need:
 - Revert PR #53973  (or set CONFIG_ARM_MPU=n) Without this current main doesn't boot on rp2040, because it fails when trying to execute flash access code from sram.
 - Merge PR #54192 - @jfischer-no was correct about there being a data0/data1 pid issue.
 - Merge PR #54030 - Without this, if you only apply the above changes, it will be extremely unreliable.

Note that with all of the above, with default logging, it is extremely stable, I've seen no failures.
When I started enabling a lot of usb-related console logging, in immediate mode, I started seeing  USB RX timeout errors, I feel like that may only have been caused somehow by logging slowing things down, and I haven't tried debugging it yet but I thought I'd mention this, just in case, if you see any usb errors let me know.

jfischer-no, 2023-01-30:
>[282861.509252] usb 1-2: New USB device found, idVendor=584e, idProduct=5342, bcdDevice= 0.01
...
[282861.509263] usb 1-2: Manufacturer: NuttX

@TimJTi Why are you posting it here?

TimJTi, 2023-01-30:
> > [282861.509252] usb 1-2: New USB device found, idVendor=584e, idProduct=5342, bcdDevice= 0.01
> > ...
> > [282861.509263] usb 1-2: Manufacturer: NuttX
> 
> @TimJTi Why are you posting it here?

Coz I'm an idiot! Tracking down a similar problem, was reading this and had the tab open, amongst others.

Shall I delete it?

jfischer-no, 2023-01-30:
>Coz I'm an idiot! Tracking down a similar problem, was reading this and had the tab open, amongst others.

>Shall I delete it?

Nah, no problem. You are almost on the [right track](https://docs.zephyrproject.org/latest/develop/getting_started/index.html), Zephyr OS offers a lot of advantages and features.

stephanosio, 2023-02-10:
Re-opening since https://github.com/zephyrproject-rtos/zephyr/pull/54030 is likely also needed to fix this issue.