zephyrproject-rtos/zephyr #52707: samples: subsys: task_wdt: sample.subsys.task_wdt loops endlessly

nordic-piks, 2022-12-01:
**Describe the bug**
The samples/subsys/task_wdt/sample.subsys.task_wdt loops endlessly

Observed for
- nrf52dk_nrf52832
- nrf52840dk_nrf52840
- nrf5340dk_nrf5340_cpunet
- nrf5340dk_nrf5340_cpuapp
- nrf9160dk_nrf9160
- nrf9160dk_nrf9160_ns

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/subsys/usb/device  -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-11-30T23:55:55.608Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.608Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mControl thread started.[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.608Z] [0m[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.1
 - Commit SHA or Version used: zephyr-v3.2.0-2123-g14358aee1cda

henrikbrixandersen, 2022-12-05:
Why is this a bug?

nordic-piks, 2022-12-07:
> Why is this a bug?

The test does not ends in expected time - there is timeout in twister after 65 sec.
There are no expected messages on console so twister waits till timeout occurs.

henrikbrixandersen, 2022-12-07:
> > Why is this a bug?
> 
> 
> 
> The test does not ends in expected time - there is timeout in twister after 65 sec.
> 
> There are no expected messages on console so twister waits till timeout occurs.

Thank you. Information like the above is crucial to understanding the issue, you are seeing.

FRASTM, 2022-12-07:
Also seen on many stm32 platforms :
b_u585i_iot02a, disco_l475_iot1, nucleo_f091rc, nucleo_f103rb, nucleo_f207zg, nucleo_f429zi, nucleo_g071rb, nucleo_g474re, nucleo_h743zi, nucleo_l073rz, nucleo_l152re, nucleo_wb55rg, nucleo_wl55jc, stm32f3_disco,   stm32l562e_dk  for window watchdog .

PR #52949 (2022-12-10): samples: Use immediate logging mode in watchdog related samples
Starting with commit b9b43a0eb772a464bba13833d11e3a31fbf4e09e, printk() messages are handled by the logging subsystem. This can make troubles in watchdog related samples if the deferred logging mode is used (and currently it is by default), because those samples are ended by a reset and some messages may not get a chance to be outputted. Since the same problem concerns also the ordinary logging messages that may be produced during execution of the samples, this commit fixes it by switching to the immediate logging mode, not by jus [... cut]
closed 2022-12-10 by carlescufi