zephyrproject-rtos/zephyr #52709: samples: subsys: nvs: sample.nvs.basic loops endlessly

nordic-piks, 2022-12-01:
**Describe the bug**
The samples/subsys/nvs/sample.nvs.basic loops endlessly

Observed for
- nrf52dk_nrf52832

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52dk_nrf52832 connected
2. go to your zephyr dir
3. call ./scripts/twister -T samples/subsys/nvs  -p nrf52dk_nrf52832 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-11-30T23:55:55.609Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.609Z] [0mTask watchdog sample application.[0m
[2022-11-30T23:55:55.609Z] [0mMain thread still alive...[0m
[2022-11-30T23:55:55.609Z] [0m*** Booting Zephyr OS build zephyr-v3.2.0-2123-g14358aee1cda ***[0m
[2022-11-30T23:55:55.609Z] [0m[00:00:00.400,299] <dbg> fs_nvs: nvs_recover_last_ate: Recovering last ate from sector 0[0m
[2022-11-30T23:55:55.609Z] [0m[00:00:00.406,799] <dbg> fs_nvs: nvs_add_gc_done_ate: Adding gc done ate at ff0[0m
[2022-11-30T23:55:55.609Z] [0m[00:00:00.406,951] <inf> fs_nvs: 3 Sectors of 4096 bytes[0m
[2022-11-30T23:55:55.609Z] [0m[00:00:00.406,982] <inf> fs_nv61 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0m61 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0m61 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0m61 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0mReboot counter history: ...3...2...1...0[0m
[2022-11-30T23:55:55.609Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.609Z] [0mRebooting in ...5...4...3...261 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0mReboot counter history: ...4...3...2...1...0[0m
[2022-11-30T23:55:55.609Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.609Z] [0mRebooting in ...5...4...361 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0mReboot counter history: ...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.609Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.609Z] [0mRebooting in ...5...461 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0mReboot counter history: ...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.609Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.609Z] [0mRebooting in ...561 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0mReboot counter history: ...7...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.609Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.609Z] [0mRebooting in ...561 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0mReboot counter history: ...8...7...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.609Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.609Z] [0mRebooting in ...561 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0mReboot counter history: ...9...8...7...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.609Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.609Z] [0mRebooting in ...561 62 --- 116 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0mReboot counter history: ...10...9...8...7...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.609Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.609Z] [0mRebooting in ...561 62 --- 115 messages dropped ---[0m
[2022-11-30T23:55:55.609Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.609Z] [0mReboot counter history: ...11...10...9...8...7...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.610Z] [0mOldest reboot counter: 0[0m
[... 297 lines cut ...]
[2022-11-30T23:55:55.611Z] [0m...28--- 47 messages dropped ---[0m
[2022-11-30T23:55:55.611Z] [0m...27...26...25...24...23...22...21...20...19...18...17...16...15...14...13...12...11...10...9...8...7...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.611Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.611Z] [0mRebooting in ...561 62 --- 115 messages dropped ---[0m
[2022-11-30T23:55:55.611Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.611Z] [0m...28--- 48 messages dropped ---[0m
[2022-11-30T23:55:55.611Z] [0m...27...26...25...24...23...22...21...20...19...18...17...16...15...14...13...12...11...10...9...8...7...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.611Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.611Z] [0mRebooting in ...561 62 --- 115 messages dropped ---[0m
[2022-11-30T23:55:55.611Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.611Z] [0m...28--- 49 messages dropped ---[0m
[2022-11-30T23:55:55.611Z] [0m...27...26...25...24...23...22...21...20...19...18...17...16...15...14...13...12...11...10...9...8...7...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.611Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.611Z] [0mRebooting in ...561 62 --- 115 messages dropped ---[0m
[2022-11-30T23:55:55.611Z] [0m63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f[0m
[2022-11-30T23:55:55.611Z] [0m...28--- 50 messages dropped ---[0m
[2022-11-30T23:55:55.611Z] [0m...27...26...25...24...23...22...21...20...19...18...17...16...15...14...13...12...11...10...9...8...7...6...5...4...3...2...1...0[0m
[2022-11-30T23:55:55.611Z] [0mOldest reboot counter: 0[0m
[2022-11-30T23:55:55.611Z] [0m[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.1
 - Commit SHA or Version used: zephyr-v3.2.0-2123-g14358aee1cda

henrikbrixandersen, 2022-12-05:
Why is this a bug?

Laczen, 2022-12-06:
@nordic-piks, the test does not loop endlessly. Be a bit more patient. The test should stop after 400 reboots.

laurenmurphyx64, 2022-12-06:
@Laczen So can we close this? It seems like from the log the oldest reboot count isn't increasing.

nordic-piks, 2022-12-07:
I would like to reopen this issue.
The test fails due to timeout waiting for expected output on console: https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/nvs/sample.yaml#L12
At my execution, test failed at timeout after 85 sec - I believe this is default value.
If this test requires more time, this should be noted in sample.yaml - custom timeout provided.
If it is "long", that should be also marked by adding tag "slow".
Please see https://docs.zephyrproject.org/latest/develop/test/twister.html for details.
@laurenmurphyx64 Please reopen this issue.

Laczen, 2022-12-07:
@nordic-piks change the title to "samples: subsys: nvs: sample.nvs.basic does not complete within twister default timeout".

nordic-piks, 2022-12-07:
> @nordic-piks change the title to "samples: subsys: nvs: sample.nvs.basic does not complete within twister default timeout".

Done, thanks.

Laczen, 2022-12-07:
@nordic-piks, there are two possible ways to treat this "bug":
1. Adapt the yaml file to better fit the test (as you proposed),
2. Reduce the `MAX_REBOOT` in the test (https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/nvs/src/main.c#L59),

BTW this "bug" could be appearing as a result of code changes not related to the sample at all. If as a result of code changes the boot proces is slowed down this will result in a timeout problem.

nordic-piks, 2022-12-08:
> @nordic-piks, there are two possible ways to treat this "bug":
>
> 1. Adapt the yaml file to better fit the test (as you proposed),
> 2. Reduce the `MAX_REBOOT` in the test [...]

For my perspective, step 1 is needed to improve documentation (and execution). The step 2 depends what is expected to be checked in this sample.

FRASTM, 2022-12-12:
Does the https://github.com/zephyrproject-rtos/zephyr/pull/52742  help in your case ? It fixes the testcase for any the stm32 board.

nordic-piks, 2022-12-12:
> Does the #52742 help in your case ? It fixes the testcase for any the stm32 board.

Indeed it helps!

Laczen, 2022-12-12:
As I mentioned, this problem is not related to the sample at all. Some change at a different subsystem (moving printk into log_printk) resulted in a failure of the test.

closed 2022-12-13 by laurenmurphyx64