zephyrproject-rtos/zephyr #52712: clang-format formatted code cannot pass CI: max line length

broglep-work, 2022-12-01:
**Describe the bug**
Code formatted with clang-format does not always respect the max line length of 100 while checkpatch insists on the line length limit. 

**Expected behavior**
clang-formatted code should pass CI without having to manually adjust code. 

**Impact**
It is an annoyance and distracts while solving the actual real problems at hand. 
For auto generated code, where you usually don't want to manually format / adjust code, you end up in a dilemma that you can't resolve. 

**To Reproduce**
Use [lwm2m_senml_cbor_encode.c](https://github.com/zephyrproject-rtos/zephyr/blob/f17e3b80e40484bc12b487c4a733f7e4542d2554/subsys/net/lib/lwm2m/generated/lwm2m_senml_cbor_encode.c)

```
$ clang-format -i subsys/net/lib/lwm2m/generated/lwm2m_senml_cbor_encode.c | wc -L subsys/net/lib/lwm2m/generated/lwm2m_senml_cbor_encode.c
117 subsys/net/lib/lwm2m/generated/lwm2m_senml_cbor_encode.c
```

**Additional context**
My attempt to make checkpatch happy with clang-formatted code by tweaking the clang-format failed after quite some time spent. I tied around with clang-format configs like `PenaltyExcessCharacter`, `PenaltyIndentedWhitespace`, `PenaltyReturnTypeOnItsOwnLine`, `PenaltyBreakAssignment`, `PenaltyBreakBeforeFirstCallParameter`, `PenaltyBreakComment`, `PenaltyBreakFirstLessLess`, `PenaltyBreakString`, `PenaltyBreakTemplateDeclaration` and was using clang-format v16

Imho the hard limit on 100 lines should be removed for code and instead clang-format should be used as the tool defining the code style. There are very good reasons to deviate from 100 lines where insisting on 100 line length is hurting readability of the code (which funny enough, actually counteracts the initial motivation of the max line length limit, which I assume to be to help with readability).

clang-format has way better understanding of the code and will thus always be superior to a "dumb" checkpatch max line length check and as a result will make much better decision and contribute to better readable code.

In addition to that, clang-format provides ways to let it respect human judgment and allows to turn off formatting, while checkpatch lacks that flexibility.

stephanosio, 2022-12-01:
As already mentioned in https://github.com/zephyrproject-rtos/zephyr/issues/51973#issuecomment-1303955670, this is not a bug.

> There are very good reasons to deviate from 100 lines where insisting on 100 line length is hurting readability of the code

I do not think so. It is really all about where you break the line -- you can always make any code readable as long as you do not have identifiers with unreasonable length (something to avoid) and break the line where it makes sense. The hard limit of 100 has nothing to do with that.

stephanosio, 2022-12-01:
cc @gmarull

gmarull, 2022-12-01:
> As already mentioned in [#51973 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/51973#issuecomment-1303955670), this is not a bug.
> 
> > There are very good reasons to deviate from 100 lines where insisting on 100 line length is hurting readability of the code
> 
> I do not think so. It is really all about where you break the line -- you can always make any code readable as long as you do not have identifiers with unreasonable length (something to avoid) and break the line where it makes sense. The hard limit of 100 has nothing to do with that.

+1, exceeding 100 is not desirable (in fact, I think 100 should only be used if readability is improved, but in general stick to 80).

broglep-work, 2022-12-01:
> I do not think so. It is really all about where you break the line -- you can always make any code readable as long as you do not have identifiers with unreasonable length (something to avoid) and break the line where it makes sense. The hard limit of 100 has nothing to do with that.

I don't want to get into a debate about this. There are many different opinions on that matter, and if I remember correctly, there is even debate in academic literature about these topics (could have changed in the past years). Code readability is a difficult topic as it involves humans, and thus judging what is more readable vs something other is inherent difficult. 

Where is the line between an unreasonable length identifier vs. a non-descriptive short identifier? What is easier for the human brain to process visually, than mentally: A single line of 120 chars vs the same thing scattered on 2 or 3 lines?
Those are interesting question, and answers might depend very much on the context, but as I said, I don't want to get into that discussion. The only conclusion I can draw is that you can't make any absolute statements on those matters.

I still think this is kind of a bug / problem though. We have conflicting / contracting style guides / rules in use. That should be resolved. The max line length (and all the other rules) I don't really care about. While I understand that clang-format is considered experimental, we should still make an effort to bring clang-format in-line with the other code guidelines.

gmarull, 2022-12-01:
> > I do not think so. It is really all about where you break the line -- you can always make any code readable as long as you do not have identifiers with unreasonable length (something to avoid) and break the line where it makes sense. The hard limit of 100 has nothing to do with that.
> 
> I don't want to get into a debate about this. There are many different opinions on that matter, and if I remember correctly, there is even debate in academic literature about these topics (could have changed in the past years). Code readability is a difficult topic as it involves humans, and thus judging what is more readable vs something other is inherent difficult.
> 
> Where is the line between an unreasonable length identifier vs. a non-descriptive short identifier? What is easier for the human brain to process visually, and then mentally: A single line of 120 chars vs the same thing scattered on 2 or 3 lines? Those are interesting question, and answers might depend very much on the context, but as I said, I don't want to get into that discussion. The only conclusion I can draw is that you can't make any absolute statements on those matters.
> 
> I still think this is kind of a bug / problem though. We have conflicting / contracting style guides / rules in use. That should be resolved. The max line length (and all the other rules) I don't really care about. While I understand that clang-format is considered experimental, we should still make an effort to bring clang-format in-line with the other code guidelines.

I think the problem with these files is that they are autogenerated. They could likely be 1) generated at build time, 2) excluded from checks, or 3) be made compliant.

This must hurt any programmer eyes, no matter what academia says:

https://github.com/zephyrproject-rtos/zephyr/blob/f17e3b80e40484bc12b487c4a733f7e4542d2554/subsys/net/lib/lwm2m/generated/lwm2m_senml_cbor_encode.c#L89-L137

broglep-work, 2022-12-01:
> +1, exceeding 100 is not desirable (in fact, I think 100 should only be used if readability is improved, but in general stick to 80).

Sound very reasonable to me. Maybe we should encode that rule (given there is general consent) into clang-format by reducing max column to 80? 
Maybe that would even  resolve this the actual problem here where it would allow clang-format to occasionally exceed 80 by 10-20 chars while keeping checkpatch still happy

broglep-work, 2022-12-01:
> I think the problem with these files is that they are autogenerated. They could likely be 1) generated at build time, 2) excluded from checks, or 3) be made compliant.

I agree, but let's not get sidetracked on that discussions (it should be part to that PR, and your option 1) I have already suggested and 2) I've implemented for the time being)

This issue here is about inconsistency in the rule sets, in this particular instance of max line length

gmarull, 2022-12-01:
> I agree, but let's not get sidetracked on that discussions (it should be part to that PR, and your option 1) I have already suggested and 2) I've implemented for the time being)
>
> This issue here is [...]

I'd personally limit clang-format to 80, and if possible, allow it to exceed the rule under certain circumstances. Maybe the `Penalty*` rules allow this behavior?

broglep-work, 2022-12-01:
> This must hurt any programmer eyes, no matter what academia says

Fully agree. And I would even got as far as to say the all in only one line instead of ~ 35 with max 100 chars  would even hurt less, thus preferable.  It would signal the brain: Too much stuff, you will never really understand it, just move on, it does not really matter what code is written here. 
Now with the nicely formatted code, with all the indentation and pretty respecting of max line length, you need to scroll through multiple lines of code to find where where this big chunk of hard to understand code ends. 

As I said, it is not easy.

broglep-work, 2022-12-01:
> Maybe the `Penalty*` rules allow this behavior?
I only have limited understanding of the clang-format and the penalty system. I wasn't able to tweak penalties so that it was always below ColumLimit. I tried to increase some penalties to really high values while decreasing other penalties to zero to make it respect ColumLimit at all costs, but didn't succeed. 

I think the other way around, lowering the limit it could work and we could maybe even keep the default penalties (which I assume should be set to sensible defaults)

chrta, 2022-12-01:
Maybe setting the ColumLimit to 0 is an option worth testing. From the doc:

> A column limit of 0 means that there is no column limit. In this case, clang-format will respect the input’s line breaking decisions within statements unless they contradict other rules.

jfischer-no, 2022-12-01:
> **Describe the bug**
> Code formatted with clang-format does not always respect the max line length of 100 while checkpatch insists on the line length limit.

clang-format does not define project coding style, it is up to you to post-fix the code formatted by clang-format, #48785.

> I think the other way around, lowering the limit it could work. And we could maybe even keep the default penalties (which I assume should be set to sensible defaults)

Yes, `ColumnLimit: 80` formats code more closer to established coding style, but is still up to the developer to check the style.

closed 2023-04-20 by stephanosio