zephyrproject-rtos/zephyr #52718: Simplify handling of fragments in `net_buf`

carlescufi, 2022-12-01:
Today:

https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/net/buf.h#L915-L921

jfischer-no, 2022-12-01:
```
                            .....                           
                          ......'..                         
                         .,.......''                        
                         .,........''                       
                         .,'........,.                      
                          ''........',                      
                          .,........';.                     
                          .,........':.                     
                           '........':.                     
                           '.......';;                      
                          .'.......';'                      
                         .'........';.                      
                        .''........',                       
                        ''.........',.                      
                       ''..........','                      
                     .''............',,'..                  
                   .,,...............,,,,,,'...........     
                  .,'..............',,,''............'''    
      .'''.......''...............',''''...............''   
     .''.......''................';''''''..............';   
    .............................,:,''''''...........'';;   
    ............................';lc;;,,,,'''''''',,;::;.   
   ............................',;lddollccc:::::cclodd:.    
   ...........................'',;cloc::::;;;;;;:::;;,,,.   
   ...........................'',,:c;,''''.............',.  
   ..........................''',,::,,,,,''...........',:.  
   ..........................''',,:c;,,,,,''......'''',;:.  
   ...........................'',,;ll::::;;,,,,,,,,,,;;:.   
   ...........................'',,,:oollllllllllllcclc,.    
   ...........................''',,:c;;,;,,,''''''''',,.    
   .............'..............''',:;,,,,,'..........','    
   .......''''',,,''''''''.....''',:;,,,,,''........'',,    
    .'''''',,,,;:cc;,,,''''''''''',cc:;;;;,''''''''',;:.    
    .'',,;;;;::ccodoc;,,,,,,'''''',:odolcc:;;;;;;;:::;.     
     .,;:ccloodddddoccc:;,,,,,,,,,;:lollllllllllloo;.       
       ':oool:;'..   .,lcc:;;;;;;;;:c;,,,,''''''',;;.       
         .             .;lollccccc:cc:;,,,''.''''',;.       
                          .';clodddddl::;;,,,,,,;:c;        
                                ....'.';:::cccccc;'         
                                         .',,,,..   
```

carlescufi, 2022-12-01:
@jfischer-no thanks, I guess that means this would be welcome for USB
@jhedberg @jori-nordic any objections for Bluetooth?
@rlubos @jukkar would you OK this for networking?

jukkar, 2022-12-01:
This proposal LGTM.

jhedberg, 2022-12-01:
> @jhedberg @jori-nordic any objections for Bluetooth?

I'm ok with it

jori-nordic, 2022-12-02:
okay too

closed 2022-12-06 by jhedberg