zephyrproject-rtos/zephyr #52732: Develop an Input subsystem for Zephyr

o7-machinehum, 2022-12-01:
## Introduction
Develop an Input subsystem for Zephyr, similar but stripped down from what you may find in Linux. I developed a driver in #52649 and it was mentioned that we should develop an input subsystem. I am willing to work on this, but I need some support from an architect level. The first and probably most common input device type will be a touch controller, but I imagine one could use a joystick, mouse, keyboard etc...

### Problem description
An input subsystem would yield the following advantages.

- Uniform handling of the functionally similar devices
- An easy interface for dispatching input events to user applications
- Swappability of different devices

### Proposed change
A generic API for input devices. 

## Detailed RFC
I've already created a [touch.h](https://github.com/zephyrproject-rtos/zephyr/pull/52649/commits/4bebffa3bdcaa573949245d3f59607ea2793b8ad#diff-6dbe52815a4f015b84721cb4ca6cfa8bb4ef4d2f246fad1d2109a8687e2ab0ea). This will get renamed to input.h and be used for generic input devices.

Here are the existing syscalls. I imagine there will be some more.
``` C
__syscall bool single_tap_get(const struct device *dev);
__syscall bool two_finger_tap_get(const struct device *dev);
__syscall int16_t x_pos_get_abs(const struct device *dev);
__syscall int16_t y_pos_get_abs(const struct device *dev);
__syscall int16_t x_pos_get_rel(const struct device *dev);
__syscall int16_t y_pos_get_rel(const struct device *dev);
__syscall int num_fingers_get(const struct device *dev);
```

### Proposed change (Detailed)
These are additions; no changed to the existing codebase.

### Dependencies
Unsure

### Concerns and Unresolved Questions
Need some feedback and guidance if possible.

gmarull, 2022-12-02:
FYI https://github.com/zephyrproject-rtos/zephyr/pull/48342

o7-machinehum, 2022-12-07:
> FYI #48342

Looks to be specifically for a Keyboard, so to get my iqs5xx driver in, should I wait for this to land, modify this, and then modify my PR for the iqs5xx device?

nordicjm, 2023-03-24:
@fabiobaltieri seems like this can be closed too

closed 2023-03-24 by fabiobaltieri