zephyrproject-rtos/zephyr #52754: tests/drivers/bbram: Refactor to use a common prj.conf

erwango, 2022-12-02:
**Is your enhancement proposal related to a problem? Please describe.**
Implementation of `tests/drivers/bbram` doesn't use a common prj.conf but requires each platform to declare it's own \<vendor\>.conf
file that are all copy paste of each others

**Describe the solution you'd like**
Refactor test to use a single prj.conf and vendors overlays when required

markCwatson, 2022-12-24:
Hi @erwango you can assign this issue to me. I will take care of it. It is only my second contribution, so please expect to give some feedback. Thanks.

markCwatson, 2022-12-28:
@erwango I have an [open PR](https://github.com/zephyrproject-rtos/zephyr/pull/53344) that is good to go and ready for approval.

commit 6f9559c8 (2022-12-29): tests: drivers: bbram: refactoring

Refactored to use a single prj.conf file.

Fixes #52754
closed 2022-12-29 by carlescufi