zephyrproject-rtos/zephyr #52764: boards: esp32c3_devkitm: unable to read memory-mapped flash memory

cfriedt, 2022-12-02:
**Describe the bug**
It's fairly clear from the TRM [1] that external flash should be mapped to address `0x3c000000`. However, doing a simple experiment shows that the data is not actually readable.

[1] https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf

**Please also mention any information which could help others to understand
the problem you're facing**:
 - **What target platform are you using?**: `esp32c3_devkitm`
 - **What have you tried to diagnose or workaround this issue?**: I actually would prefer not to hack a workaround. This should be a bug that gets fixed.
 - **Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?**: no, it looks like it's always been broken
 - ...

**To Reproduce**
Steps to reproduce the behavior:

Flash a program to the device
```
west build -p auto -b esp32c3_devkitm -t flash samples/hello_world -- \
  -DCONFIG_FLASH=y \
  -DCONFIG_SHELL=y \
  -DCONFIG_FLASH_PAGE_LAYOUT=y \
  -DCONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN=n \
  -DCONFIG_CRC=y \
  -DCONFIG_CRC_SHELL=y \
  -DCONFIG_HEAP_MEM_POOL_SIZE=16384
```

Open a terminal with this (your exact tty device might vary):
```
minicom -D /dev/tty.usbserial-1130 -b 115200
```

Run the following commands:
```
flash erase flash-controller@60002000 0x00210000 0x1000
flash load flash-controller@60002000 0x00210000 0x20
```

On the host, run
```
# generate 0x01 to 0x20 in binary
for i in `seq 1 32`; do printf "%02x " $i; done | xxd -r -p - > /tmp/output.bin
crc32 /tmp/output.bin
# 87e6ec25
# write 32-bytes to the image-scratch partition
dd if=/tmp/output.bin of=/dev/tty.usbserial-1130 bs=32 count=1
```

Then on the device,
```
uart:~$ flash read flash-controller@60002000 0x00210000 0x20
00210000: 01 02 03 04 05 06 07 08  09 0a 0b 0c 0d 0e 0f 10 |........ ........|
00210010: 11 12 13 14 15 16 17 18  19 1a 1b 1c 1d 1e 1f 20 |........ ....... |
```

I was under the assumption that the `image-scratch` partition would have been mapped in at `0x3c000000 + 0x00210000`, but then this just reads 0 ((I would expect 1).
 ```
devmem 0x3c210000 8
Using data width 8
Read value 0x0
```

Note: if the `devmem dump` command is available (#52747), then the entire region can be dumped with
```
uart:~$ devmem dump -a 0x3c210000 -s 32 -w 32
3C210000: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........|
3C210010: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........|
```

The fact that it's returning zero indicates to me that the area of memory has not been mapped. How does one ensure that external flash is mapped on this device?

Ultimately, I would expect the same CRC of `87e6ec25`, but there is a mismatch (since `devmem` fails to read back the same data, that is understandable).
```
crc 0x3c210000 32
0x190a55ad
```

**Expected behavior**
I would imagine that data written to memory-mapped flash could be read-back via memory reads.

**Impact**
It's a showstopper for the `icev_wireless`, which I hope to demonstrate at RISC-V Summit.

**Logs and console output**
See above

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Any
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.15.2
 - Commit SHA or Version used: dc26a617d792f0a41b96c1965f1a967372caf8cb

**Additional context**
To give some more context, I need to be able to load a 104090-byte FPGA image into flash and then program the FPGA over SPI (the former already works, and the only thing preventing the latter from working is this issue).

cfriedt, 2022-12-02:
It looks like this is a similar issue
https://esp32.com/viewtopic.php?t=29654#p102824

cfriedt, 2022-12-02:
cc @fabiobaltieri

cfriedt, 2022-12-03:
For reference, I tried injecting this small snippet into ` above API on the `icev_wireless` (esp32-c3 module), 
```
diff --git a/drivers/flash/flash_esp32.c b/drivers/flash/flash_esp32.c
index d849283414..10c2d6c793 100644
--- a/drivers/flash/flash_esp32.c
+++ b/drivers/flash/flash_esp32.c
@@ -148,6 +148,21 @@ static int flash_esp32_init(const struct device *dev)
 
        k_sem_init(&dev_data->sem, 1, 1);
 
+       /* map flash into memory */
+       size_t src_addr = 0x210000;
+       size_t size = 0x00040000;
+       spi_flash_mmap_memory_t memory = SPI_FLASH_MMAP_DATA;
+       const void *ptr = (void *)-1;
+       spi_flash_mmap_handle_t handle = -1;
+
+       printk("mapping %zu bytes of flash from offset %zx\n", size, src_addr);
+       if (spi_flash_mmap(src_addr, size, memory, &ptr, &handle) != ESP_OK) {
+               printk("spi_flash_mmap(%zx, %zu, %d, .., .. ) failed\n", src_addr, size, memory);
+       } else {
+               printk("mapped %zu bytes of flash from offset %zx to address %p as handle %x\n",
+                      size, src_addr, ptr, handle);
+       }
+
        return 0;
 }
```

and it prints the following:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c632
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x15e4
load:0x403ce000,len:0x920
load:0x403d0000,len:0x2c4c
SHA-256 comparison failed:
Calculated: 3628ce653aec1f7ce3c1cb1bea5e3a5f6833e2e95d422a7df4491f732fba756b
Expected: 6055787ba2837e2550eafa95d4080d38cc2d09558a073f1a209d69f762cb0c63
Attempting to boot anyway...
entry 0x403ce000
I (43) boot: ESP-IDF 8c7ee781ea 2nd stage bootloader
I (43) boot: compile time 08:28:28
I (43) boot: chip revision: 3
I (45) boot.esp32c3: SPI Speed      : 40MHz
I (50) boot.esp32c3: SPI Mode       : DIO
I (54) boot.esp32c3: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00100000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28) 
I (110) esp_image: segment 1: paddr=00010044 vaddr=3fc8d980 size=000e8h (   232) load
I (119) esp_image: segment 2: paddr=00010134 vaddr=3fc8da68 size=0020ch (   524) load
I (127) esp_image: segment 3: paddr=00010348 vaddr=40380000 size=05fech ( 24556) load
I (142) esp_image: segment 4: paddr=0001633c vaddr=00000000 size=09cfch ( 40188) 
I (153) esp_image: segment 5: paddr=00020040 vaddr=3c000040 size=023c4h (  9156) map
I (155) esp_image: segment 6: paddr=0002240c vaddr=00000000 size=0dc0ch ( 56332) 
I (173) esp_image: segment 7: paddr=00030020 vaddr=42010020 size=0a4b8h ( 42168) map
I (184) boot: Loaded app from partition at offset 0x10000
I (184) boot: Disabling RNG early entropy source...
mapping 262144 bytes of flash from offset 210000
mapped 262144 bytes of flash from offset 210000 to address 0x42010000 as handle 1
*** Booting Zephyr OS build zephyr-v3.2.0-2246-g041e85303c7f ***
Hello World! icev_wireless

uart:~$ 
```

But then I tried loading / dumping flash using the steps above (i've tried dumping both `0x3c210000` and `0x42010000`) and it seems to not work.

```
devmem dump -a 0x3c210000 -s 32 -w 8
3C210000: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........|
3C210010: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........|
```

```
devmem dump -a 0x42010000 -s 32 -w 8
42010000: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........|
42010010: 00 00 00 00 00 00 00 00  20 00 01 42 b8 a4 00 00 |........  ..B....|
```

cfriedt, 2022-12-03:
Would probably be worth trying the same thing with `esp_partition_mmap()`.

cfriedt, 2022-12-03:
@sylvioalves - Oh, was just reading a bit more and `0x42010000` corresponds to the instruction virtual memory region. The equivalent data virtual memory region is `0x3c000000`, but for some reason it is being mapped into instruction memory rather than data memory even though `SPI_FLASH_MMAP_DATA` is specified.

cfriedt, 2022-12-03:
@beriberikix - do you think this is something that your entourage would have possibly solved already?

beriberikix, 2022-12-03:
Hmm...definitely have issues in the past with addressing memory but largely not an issue. Unfortunately we mostly deal with ESP32-S3 and a bit of the S2 these days. @szczys has the most experience with Espressif on Zephyr - any ideas?

cfriedt, 2023-04-22:
Reported on Discord as well
https://discord.com/channels/720317445772017664/733036686195294269/1098899656802566164

cfriedt, 2023-05-14:
@sylvioalves - any updates on this? I would be happy to do some testing.

cfriedt, 2023-05-26:
@sylvioalves - would be good to see if you have made any progress. The ICE-V Wireless is on its 3rd GrploupGets campaign, so it's fairly popular.

tyalie, 2023-10-22:
Hi. I'm also running into issue here but with the esp32s3. Are there any updates on this?

PR #64231 (2023-10-25): soc: esp32: fix memory-mapped reading address
ESP32 flash_mmap() function requires `_rodata_reserved_start` address
to be at the beginning of RODATA. This allows adding memory-mapped flash
areas.

This also adds a sample code describing a way to use it.

Fixes #52764
commit 3f5ea785 (2023-10-25): linker: esp32: fix linker to enable proper MMU usage

ESP32 flash_mmap() function requires `_rodata_reserved_start` address
to be at the beginning of RODATA. This allows adding memory-mapped flash
areas.

Fixes #52764
closed 2023-10-25 by carlescufi