zephyrproject-rtos/zephyr #52776: ite: eSPI driver: espi_it8xxx2_send_vwire() is not setting valid flag along with respective virtual wire when invoked from app code.

NagarajuGundimi, 2022-12-05:
[image]
**Describe the bug**
ite: eSPI driver: [image]() is not setting valid flag along with respective virtual wire when invoked from app code.

When any virtual wire is set/clear from app code using espihub_send_vw() API it is invoking espi_it8xxx2_send_vwire(). But espi_it8xxx2_send_vwire() is only setting respective virtual wire and NOT corresponding valid flag. Even though espi_vw_set_valid() function is available there is no API to invoke it from app code.

**Expected behavior**
espi_it8xxx2_send_vwire() should also set respective valid flag along with virtual wire.

**Impact**
Not able to boot to OS(Windows) in MAFS mode.

albertofloyd, 2022-12-07:
@Dino-Li @GTLin08 could you take a look into this?

Dino-Li, 2022-12-07:
Loop BJ @itebj

itebj, 2022-12-07:
Hi sirs, 

I was just looking into this and I think that it's about the current MAFS under experiments with the Intel and I think that this item has been already resolved after the discussions yesterday with @NagarajuGundimi. We've also applied the changes to the current MAFS code.

@NagarajuGundimi 
According to your suggestions, I've modified and tested the new App code in espi flash and now it can directly boot after the MAFS mirroring. Now it works well after the direct jumping to the App code. It can boot all the way to the OS.

Thanks again for great patience.

PR #53053 (2022-12-19): ITE: drivers/espi: fix the VW valid flag issue in MAFS mode
Add CONFIG_ESPI_SET_VW_VALID_WITH_LEVEL option to fix the eSPI driver VW Valid Bits Issue in MAFS mode

fixes #52776
commit 108d5163 (2022-12-19): ITE: drivers/espi: fix the VW valid flag issue in MAFS mode

Add CONFIG_ESPI_SET_VW_VALID_WITH_LEVEL to fix the VW Valid Bits Issue

fixes #52776
closed 2022-12-19 by fabiobaltieri