zephyrproject-rtos/zephyr #52784: NRF_DRIVE_S0D1 option is not always set in the nordic,nrf-twi and nordic,nrf-twim nodes, when using shield?

pideu-sj, 2022-12-05:
In https://github.com/zephyrproject-rtos/zephyr/pull/47145 and in https://github.com/zephyrproject-rtos/zephyr/issues/49956 was described, that the drive mode S0D1 must be set for i2c interface.

But when I build e.g. ``west build -p always -b pan1780_evb samples/sensor/amg88xx -DSHIELD=<Shield Designation>`` the project can be build successfully, but after flashing I get the following output on the uart:

````
[00:00:00.749,572] <err> i2c_nrfx_twi: Error on I2C line occurred for message 0
[00:00:00.749,633] <err> AMG88XX: Failed to read Power mode
[00:00:00.749,633] <err> AMG88XX: Failed to initialize device!
*** Booting Zephyr OS build zephyr-v3.2.0-2264-g510dca57da82 ***
sensor: device not ready.
````

After adding NRF_DRIVE_S0D1  drive mode to the `pan1780_evb-pinctrl.dtsi`, like:
````
	i2c0_default: i2c0_default {
		group1 {
			psels = <NRF_PSEL(TWIM_SDA, 0, 26)>,
				<NRF_PSEL(TWIM_SCL, 0, 27)>;
				bias-pull-up;
			        nordic,drive-mode = < NRF_DRIVE_S0D1 >;
		};
	};
````
The interface is working an delivers the expected data from sensor.

I'am using the following hardware and configurations:
- Toolchain: Zephyr SDK
- Platform: pan1780_evb
- Shield: new custom Grid Eye Evaluation Shield (similar to the currently existing amg88xx_eval_kit, just with interrupt pin on D2 of the Arduino Header)
- on main branch: 510dca57da82faf8a546af403c7a2cebbf42289b

Is there maybe a special handling needed, if a shield is involved into the build process?

anangl, 2022-12-05:
> Is there maybe a special handling needed, if a shield is involved into the build process?

No, it does not matter if a shield is used or not. After #49956 was fixed, S0S1 is overridden with S0D1 in [pinctrl_nrf.c](https://github.com/zephyrproject-rtos/zephyr/blob/89c7745c0148e54999f31cf9c8ae034cd18fd2f1/drivers/pinctrl/pinctrl_nrf.c#L166-L174), so you should see no difference if you remove `nordic,drive-mode = < NRF_DRIVE_S0D1 >;` in the above snippet.

But since you also added `bias-pull-up;`, perhaps the problem is that there are no external pull-ups on that custom shield?

pideu-sj, 2022-12-06:
You are right. The problem was not the drive mode. The problem was, that i really need to add the pull up's for SDA and SCL. And it looks like I've always tested that together with the drive mode and not separately. 

There are pull up's on the shield as well, so I can see the communication working either by adding them on the shield or by adding ``bias-pull-up;`` like mentioned above to the ``pan1780_evb-pinctrl.dtsi``. But that is different now, in the past I did some testing with the nRF Connect SDK as well and there was no needed to add them, which confused me a little bit. So thanks for clarification.

I will just add the description of how to add the pull up's on the board or set the ``bias-pull-up;`` in the ``*.dtsi`` file to the index.rst documentation of the shield, when adding it with a Pull Request.

closed 2022-12-06 by pideu-sj