zephyrproject-rtos/zephyr #52792: ATWINC1500 : (wifi_winc1500_nm_bsp.c : nm_bsp_reset) The reset function is not logical and more

hichemMenta, 2022-12-05:
Hello,

I use the nRF5340 DK and the nRF Connect SDK. 
The module used is ATWINC1500-XPRO.

The bug comes from the nm_bsp_reset function in the 
drivers/wifi/winc1500/wifi_winc1500_nm_bsp.c.

```
void nm_bsp_reset(void)
{
	gpio_pin_set_dt(&winc1500_config.chip_en_gpio, 0);
	gpio_pin_set_dt(&winc1500_config.reset_gpio, 0);
	nm_bsp_sleep(100);

	gpio_pin_set_dt(&winc1500_config.chip_en_gpio, 1);
	nm_bsp_sleep(10);

	gpio_pin_set_dt(&winc1500_config.reset_gpio, 1);
	nm_bsp_sleep(10);
}
```

At the end of the reset function (nm_bsp_reset), the reset pin stays in the active state, low physical level. 
This is not the expected behavior. 

The problem is that the function `gpio_pin_set_dt`  is not used correctly.
It is a regression. This part of the code has changed 5 months ago. 
[image]
The function is used in the physical sense and not in the logical sense which is not correct because the reset pin is active in the low state.

I am convinced that the WINC1500 driver has not been fully tested, as I cannot get the module to work. 
The chip_id check fails.

Thanks,
Hichem

closed 2023-05-17 by kartben

GeorgePriestner-Eaton, 2023-10-13:
Is this still an issue?
@hichemMenta Did you set the reset-gpio to GPIO_ACTIVE_LOW in the device tree?

[image]

GeorgePriestner-Eaton, 2023-10-16:
@hichemMenta How did you resolve? Codebase still seems the same.