zephyrproject-rtos/zephyr #52803: Kconfig: STM32F4 UF2 family ID

nguyendown, 2022-12-05:
**Describe the bug**
STM32F411CE does not have family ID pre-filled to produce UF2 file.

The ID `0x57755a57` is for the STM32F4 series, not just STM32F401.
https://github.com/microsoft/uf2/commit/9b03a7d08058302b979a3a7568a52789a290fe25

https://github.com/zephyrproject-rtos/zephyr/blob/510dca57da82faf8a546af403c7a2cebbf42289b/Kconfig.zephyr#L557

**To Reproduce**
```
$ west build -p -b blackpill_f411ce samples/basic/blinky -- -DCONFIG_BUILD_OUTPUT_UF2=y
...
usage: uf2conv.py [-h] [-b BASE] [-o FILE] [-d DEVICE_PATH] [-l] [-c] [-D]
                  [-f FAMILY] [-C]
                  [INPUT]
uf2conv.py: error: argument -f/--family: expected one argument
ninja: build stopped: subcommand failed.
```

**Expected behavior**
```
$ west build -p -b blackpill_f401ce samples/basic/blinky -- -DCONFIG_BUILD_OUTPUT_UF2=y
...
Converting to uf2, output size: 30208, start address: 0x0
Wrote 30208 bytes to zephyr.uf2
```

**Impact**
Probably STM32F4 series.

**Environment:**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.1
 - Commit: 510dca57da82faf8a546af403c7a2cebbf42289b

erwango, 2022-12-07:
Not sure if it could apply to the whole F4 series as there are already IDs attributed to other F4 SoC as well:
```
	default "0x5ee21072" if SOC_STM32F103XE
	default "0x57755a57" if SOC_STM32F401XC || SOC_STM32F401XE
	default "0x6d0922fa" if SOC_STM32F407XE
	default "0x8fb060fe" if SOC_STM32F407XG
```
So probably some specific rule has to be done for these SoCs.

gautierg-st, 2022-12-21:
We have indeed 3 IDs for the STM32F4xx series. According to the [utils/uf2families.json](https://github.com/microsoft/uf2/blob/master/utils/uf2families.json) file we have the following:
```
    {
        "id": "0x57755a57",
        "short_name": "STM32F4",
        "description": "ST STM32F4xx"
    },
    {
        "id": "0x6d0922fa",
        "short_name": "STM32F407",
        "description": "ST STM32F407"
    },
    {
        "id": "0x8fb060fe",
        "short_name": "STM32F407VG",
        "description": "ST STM32F407VG"
    },
```
I don't know why we have specific IDs for F407 and F407VG. 
@erwango Should we just ignore them and use the generic ID for all STM32F4xx?

erwango, 2022-12-21:
> @erwango Should we just ignore them and use the generic ID for all STM32F4xx?

Let's keep the differentiation, unless we're 100% sure there won't be any impact.

PR #53264 (2022-12-22): Kconfig.zephyr: Update UF2 ID for STM32F4 series
Update UF2 family ID for the entire STM32F4 series, except STM32F407xx which have their own dedicated IDs.

Fixes #52803
closed 2022-12-22 by carlescufi