zephyrproject-rtos/zephyr #52806: RISC-V mode supervisor execution

josecm, 2022-12-05:
I'd like to run Zephyr on top of the [bao hypervisor](https://github.com/bao-project/bao-hypervisor) on qemu riscv64. I hit a wall as soon as I started. Zephyr only seems to support machine/user mode execution. However, as the hypervisor depends on the hypervisor extensions for running, it only supports supervisor/user mode guests, i.e. running in VS/VU modes. The same is likely to be a problem for hypervisors which will soon have upstream support for RISC-V such as Xen or Jailhouse.

I'd like to know if there are any on-ongoing efforts or, at least, plans to support supervisor mode execution for zephyr on RISC-V. If not, would the maintainers be open to such an effort?

npitre, 2022-12-13:
There is no such plan that I'm aware of.
Of course you are welcome to contribute such ability.

The Linux kernel implements an abstraction that appears to take care
of most differences e.g.:

```
#ifdef CONFIG_RISCV_M_MODE
#define CSR_STATUS mstatus
#else
#define CSR_STATUS sstatus
#endif
```

and `CSR_STATUS` is then used so the same code can be compiled for 
either modes, etc. A similar approach could be used here.

josecm, 2022-12-13:
@npitre thanks for answering! And yes, that kind of abstraction would be the first task. There is a quite straightforward mapping between s- and m-mode CSRs. But we'll probably get into all kind of other troubles. I'll try to find the time in the next few months to draft a first approach and bootstrap this.