zephyrproject-rtos/zephyr #52812: Various problems with pipes (Not unblocking, Data Access Violation, unblocking wrong thread...)

m-wichmann, 2022-12-06:
**Describe the bug**
I'm trying to use a pipe object to send data from a UART IRQ (async) to a thread. Since f86027f this should be possible, but there is some weird behaviour. It could be multiple bugs, but I suspect that these are related, so I'm combining them here. Here is a short description of the bugs (details below):
1. `k_pipe_get(..., K_FOREVER)` does not unlock if data is written after the read call is blocking. I worked around this issue by polling `k_pipe_get` with a small timeout. Code to reproduce this error is added below.
2. Sporadic Data Access Violation because of NULL-Pointer in `dest_list` in `z_impl_k_pipe_put()`, if `k_pipe_put()` is called from IRQ. It seems that the list is ok after being filled in `pipe_waiter_list_populate()`, but by the time `pipe_write()` is called the list isn't valid anymore. This can be caught by checking `prev` and `next` in `sys_dlist_remove()`. Code to reproduce this error is added below.
3. Unused pipes that wait for data sometimes unlock with "return value==0" and "bytes_read==buffer_size" although no data was received. I was not able to produce a minimal working example for this error or narrow this down. But this happens with completly unrelated pipes, so I suspect, that the `waiter_list` from the previous point leads to this problem and unblocks a wrong thread.

The code to reproduce these issues can be seen below. I could only really narrow down the second error. Here the stack trace and log output for the second error.

```
k_sys_fatal_error_handler(unsigned int reason, const z_arch_esf_t * esf) (c:\ncs\v2.2.0-rc1\zephyr\kernel\fatal.c:41)
z_fatal_error(unsigned int reason, const z_arch_esf_t * esf) (c:\ncs\v2.2.0-rc1\zephyr\kernel\fatal.c:131)
z_arm_fatal_error(unsigned int reason, const z_arch_esf_t * esf) (c:\ncs\v2.2.0-rc1\zephyr\arch\arm\core\aarch32\fatal.c:63)
z_arm_fault(uint32_t msp, uint32_t psp, uint32_t exc_return, _callee_saved_t * callee_regs) (c:\ncs\v2.2.0-rc1\zephyr\arch\arm\core\aarch32\cortex_m\fault.c:1097)
z_arm_usage_fault() (c:\ncs\v2.2.0-rc1\zephyr\arch\arm\core\aarch32\cortex_m\fault_s.S:102)
<signal handler called> (Unbekannte Quelle:0)
sys_dlist_remove(sys_dnode_t * node) (c:\ncs\v2.2.0-rc1\zephyr\include\zephyr\sys\dlist.h:501)
sys_dlist_get(sys_dlist_t * list) (c:\ncs\v2.2.0-rc1\zephyr\include\zephyr\sys\dlist.h:522)
pipe_write(struct k_pipe * pipe, sys_dlist_t * src_list, sys_dlist_t * dest_list, _Bool * reschedule) (c:\ncs\v2.2.0-rc1\zephyr\kernel\pipes.c:307)
z_impl_k_pipe_put(struct k_pipe * pipe, void * data, size_t bytes_to_write, size_t * bytes_written, size_t min_xfer, k_timeout_t timeout) (c:\ncs\v2.2.0-rc1\zephyr\kernel\pipes.c:420)
k_pipe_put(size_t min_xfer, size_t * bytes_written, size_t bytes_to_write, void * data, struct k_pipe * pipe) (c:\ncs\projects\eblem\build_eblem\zephyr\include\generated\syscalls\kernel.h:1284)
test_write(const char * str) (c:\ncs\projects\eblem\src\main.c:213)
test_write_irq() (c:\ncs\projects\eblem\src\main.c:237)
uart_cb(const struct device * dev, struct uart_event * evt, void * user_data) (c:\ncs\projects\eblem\src\uart\ngp_uart.c:27)
user_callback(const struct device * dev, struct uart_event * evt) (c:\ncs\v2.2.0-rc1\zephyr\drivers\serial\uart_nrfx_uarte.c:849)
notify_uart_rx_rdy(const struct device * dev, size_t len) (c:\ncs\v2.2.0-rc1\zephyr\drivers\serial\uart_nrfx_uarte.c:863)
rx_timeout(struct k_timer * timer) (c:\ncs\v2.2.0-rc1\zephyr\drivers\serial\uart_nrfx_uarte.c:1102)
z_timer_expiration_handler(struct _timeout * t) (c:\ncs\v2.2.0-rc1\zephyr\kernel\timer.c:63)
sys_clock_announce(int32_t ticks) (c:\ncs\v2.2.0-rc1\zephyr\kernel\timeout.c:270)
sys_clock_timeout_handler(int32_t chan, uint64_t expire_time, void * user_data) (c:\ncs\v2.2.0-rc1\zephyr\drivers\timer\nrf_rtc_timer.c:422)
process_channel(int32_t chan) (c:\ncs\v2.2.0-rc1\zephyr\drivers\timer\nrf_rtc_timer.c:494)
rtc_nrf_isr(const void * arg) (c:\ncs\v2.2.0-rc1\zephyr\drivers\timer\nrf_rtc_timer.c:518)
_isr_wrapper() (c:\ncs\v2.2.0-rc1\zephyr\arch\arm\core\aarch32\isr_wrapper.S:259)
<signal handler called> (Unbekannte Quelle:0)
arch_swap(unsigned int key) (c:\ncs\v2.2.0-rc1\zephyr\arch\arm\core\aarch32\swap.c:53)
z_swap_irqlock(unsigned int key) (c:\ncs\v2.2.0-rc1\zephyr\kernel\include\kswap.h:185)
z_swap(struct k_spinlock * lock) (c:\ncs\v2.2.0-rc1\zephyr\kernel\include\kswap.h:196)
z_pend_curr(struct k_spinlock * lock, k_spinlock_key_t key, _wait_q_t * wait_q, k_timeout_t timeout) (c:\ncs\v2.2.0-rc1\zephyr\kernel\sched.c:859)
z_sched_wait(struct k_spinlock * lock, k_spinlock_key_t key, _wait_q_t * wait_q, k_timeout_t timeout, void ** data) (c:\ncs\v2.2.0-rc1\zephyr\kernel\sched.c:1912)
pipe_get_internal(k_spinlock_key_t key, struct k_pipe * pipe, void * data, size_t bytes_to_read, size_t * bytes_read, size_t min_xfer, k_timeout_t timeout) (c:\ncs\v2.2.0-rc1\zephyr\kernel\pipes.c:635)
z_impl_k_pipe_get(struct k_pipe * pipe, void * data, size_t bytes_to_read, size_t * bytes_read, size_t min_xfer, k_timeout_t timeout) (c:\ncs\v2.2.0-rc1\zephyr\kernel\pipes.c:668)
k_pipe_get(size_t min_xfer, size_t * bytes_read, size_t bytes_to_read, void * data, struct k_pipe * pipe) (c:\ncs\projects\eblem\build_eblem\zephyr\include\generated\syscalls\kernel.h:1309)
test_read(char * data, size_t len) (c:\ncs\projects\eblem\src\main.c:226)
test_thread_entry(void * x, void * y, void * z) (c:\ncs\projects\eblem\src\main.c:245)
z_thread_entry(k_thread_entry_t entry, void * p1, void * p2, void * p3) (c:\ncs\v2.2.0-rc1\zephyr\lib\os\thread_entry.c:36)
z_thread_entry(k_thread_entry_t entry, void * p1, void * p2, void * p3) (c:\ncs\v2.2.0-rc1\zephyr\lib\os\thread_entry.c:36)
[Unknown code] (Unbekannte Quelle:0)
```

```
*** Booting Zephyr OS build v3.2.99-ncs1-rc1 ***
[00:00:05.693,939] <inf> main: test_write: test_write("Hello World")
[00:00:05.694,305] <inf> main: test_read:   Read OK (11 bytes, "Hello World")
[00:00:11.715,454] <inf> main: test_write: test_write("Hello World")
[00:00:11.715,606] <inf> main: test_read:   Read OK (11 bytes, "Hello World")
[00:00:11.715,667] <inf> main: test_write: test_write("Hello World")
[00:00:11.715,850] <inf> main: test_write: test_write("Hello World")
[00:00:11.715,911] <inf> main: test_read:   Read OK (11 bytes, "Hello World")
[00:00:11.716,094] <inf> main: test_read:   Read OK (11 bytes, "Hello World")
....
[00:00:56.939,178] <err> os: mem_manage_fault: ***** MPU FAULT *****
[00:00:56.939,208] <err> os: mem_manage_fault:   Data Access Violation
[00:00:56.939,208] <err> os: mem_manage_fault:   MMFAR Address: 0x0
[00:00:56.939,239] <err> os: esf_dump: r0/a1:  0x20000bf8  r1/a2:  0x20010208  r2/a3:  0x00000000
[00:00:56.939,239] <err> os: esf_dump: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0003be4f
[00:00:56.939,270] <err> os: esf_dump:  xpsr:  0x21000021
[00:00:56.939,300] <err> os: esf_dump: Faulting instruction address (r15/pc): 0x0004673a
[00:00:56.939,331] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:56.939,331] <err> os: z_fatal_error: Fault during interrupt handling
[00:00:56.939,392] <err> os: z_fatal_error: Current thread: 0x20001f98 (test_thread_tid)
[00:01:01.753,631] <err> os: k_sys_fatal_error_handler: Halting system
[... 151 lines cut ...]
static void test_thread_entry(void * x, void * y, void * z) {
   char data[100];
   while (1) {
      memset(data, 0, sizeof(data));
      test_read(data, sizeof(data));
   }
}
```

**Expected behavior**
I expect the pipe objects to work according to the documentation and not affect unrelated kernel objects.

**Impact**
With these bugs there is basically no way to use pipes to asynchronously wait for data or use them from IRQs. I could use other kernel objects (e.g. FIFO), but I like the copying of the pipe, to make memory management easier.

**Logs and console output**
Logs are included in the top sections.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: Nordic nRF Connect SDK v2.2.0-rc1 (Based on Zephyr cd16a83; Release candidate used because it includes f86027f)
 - MCU: nRF52840

**Additional context**
None

peter-mitsis, 2022-12-06:
I'm looking into this right away.

peter-mitsis, 2022-12-06:
Update on issue 1/3.

Pipes currently have some weird (legacy) quirks. Blame me. This one has to do with interactions between the minimum of bytes to receive, the requested number of bytes to receive and the timeout.  For reference, the documentation for k_pipe_get() currently reads as ... 

> Data is synchronously received from a pipe by a thread. If the specified minimum number of bytes can not be immediately satisfied, then the operation will either fail immediately or attempt to receive as many bytes as possible and then pend in the hope that the receive can be completed later."

There is some ambiguity as to how much data the routine should wait for while pended. The current implementation attempts to wait for all the requested data--not the minimum data; if the timeout finishes before all the requested data has been received then, the return value is set based on how much of the data was received as compared to the minimum requested. This is not the most consistent as sometimes we measure the completeness and success as it compares to the minimum requested and sometimes with total requested. In the case of the first issue (read first, then write) it is waiting forever for all the data and not the minimum data.

I suspect it would ultimately be better to be uniform in our approach and always base both completion and success on the minimum number of bytes transferred. As an aside, this would mean that the simplest way to interpret the bytes_to_read and bytes_to_write parameters of k_pipe_get() and k_pipe_put() would be as maximum values. Such an interpretation would dovetail nicely with the min_size parameter.

peter-mitsis, 2022-12-14:
Update:

1/3: I almost have an implementation ready to resolve this specific issue.
2/3: This one is a little more perplexing since there seems to be a spin-lock around the bulk of the contents. However, it is my suspicion that there is an interaction with something that is not protected by pipe's spinlock.  At this point, it is not known what it is, but the first place I plan on investigating is the timer removing something from the pipe's wait queue. And as I write this, I think that I have figured out the key issue--pipes, when they are generating the lists from the wait queues, do not have a lock on those same lists. Thus, they are vulnerable to having those lists corrupted by another processor servicing a timer interrupt that may remove one of the pended threads from the wait queue (and thus the list).
3/3: I think this is directly related to the revelation in 2/3 above.

Although it was not stated in the issue, based on the previous analysis, I think that this problem that you are describing exists on SMP systems and not on those with 1 core/cpu. If it were UP (or SMP with only 1 core), the spinlock would be equivalent to an irq lock, which in turn would prevent ISRs from being serviced (thereby preventing the threads from being removed from the wait queue while a pipe operation was in progress).

I will have to ponder it a bit more to figure out how to resolve this properly.

m-wichmann, 2022-12-15:
Thanks for working on this problem. I'm new to Zephyr so I'm not much of a help, but the explanation seems plausible. But please note that the problem was originally seen on an nRF52840, which is a single core CPU. SMP is also not activated. So it seems that it affects non SMP systems as well.

havoc-dlt, 2023-01-02:
I'm seeing the exact same problem on the exact same (single core) processor. I too use k_pipe_put to write (one byte at a time) from an ISR, which then wakes up a potentially suspended thread that services the pipe data.
So far I can't add any more insight than the quite detailed report on this issue other than "this affects more than one person".
Before finding this bug report I had already figured out that the problem stems from commit f86027f, as on all releases I used prior to that I had no issue.

peter-mitsis, 2023-01-04:
I am back from holidays and am continuing to work on this.

m-wichmann, 2023-01-10:
Thanks for the PR. I just tried the changes (#53634) but the error is still there as described above (error two). Same code to reproduce, same stack trace on error (`prev` is null in `sys_dlist_remove`).

I tested this by using Nordic nRF Connect SDK v2.2.0 (Based on Zephyr cd16a83) and applying the changes manually (because I couldn't work out how to use upstream Zephyr directly) on a nRF52840 single core MCU.

peter-mitsis, 2023-01-10:
Thanks for the valuable feedback. I had previously tried my testing with a different ARM board (frdm_k64f) using the Zephyr SDK and using irq_offload() to execute test_write_irq() routine and was unable to replicate the issue that way. There are a number of obvious differences between your environment and my own (board, compiler and irq source).

Let's see what else I can come up with to help resolve this bug.

m-wichmann, 2023-01-11:
I didn't know about `irq_offload()` and just tested it. Couldn't reproduce the error either, but I did see a difference that could be relevant (or not...): the IRQ priorities* are different. `irq_offload()` uses the SVC IRQ that is set to priority 1 (`SCB.SHPR` value 0x20; set in `arch/arm/include/aarch32/cortex_m/exc.h`), while the UARTE IRQ that I use is priority 3 (`NVIC.IPR` value 0x60). In `arch_irq_lock()` `BASEPRI` is set to priority 2 (value 0x40), so the UARTE IRQ should be blocked, while the SVC IRQ should still be able to run. I would expect that blocking should be safer in this case, but I'm not sure how this could influence spinlocks. Maybe if the BASEPRI gets set while UART IRQ is already running. I tried raising the priority of the UART IRQ, but it seems to deadlock after a while. It's probably not designed to be used that way.

------

*IRQ Priorities on ARM are kind of weird, so here some notes so there is no confusion (on my side):
- Low values are high priorities
- Only top three bits of priority registers are used on nRF52840. So priority 3 corresponds to register value 0x60

peter-mitsis, 2023-01-11:
@m-wichmann - One thing that I observed in the provided stack trace was that it appears that the exception appears to be on the path of handling a k_timer handler--`rx_timeout()`. That particular k_timer is instantiated by `uart_nrfx_uarte.c`. This could indicate that there may be an alternate method of re-creating the problem on other platforms, or at least other ARM boards by using a periodic timer to call `test_write_irq()`.

Would you be so kind as to attach a copy of your `.config` as well? For reference, this can be found at ...
    `build/zephyr/.config`

m-wichmann, 2023-01-11:
Here my currently used config: [.config](https://github.com/zephyrproject-rtos/zephyr/files/10393284/config.txt)

m-wichmann, 2023-01-16:
I managed to strip away most of my code and got a minimal example that shows the error (error 2). It's more or less like the original code, but the write function is called from a timer. For some reason the problem only shows when I call `uart_rx_enable` on init. There is no UART used in this example except for that call. The problem does not occur every time, but most of the time about 30 to 60 seconds after running this example there is a null pointer in `sys_dlist_remove` that leads to a crash later on.

```C
#include <stdint.h>
#include <stddef.h>
#include <string.h>
#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/drivers/uart.h>
#include <zephyr/logging/log.h>

LOG_MODULE_REGISTER(main);

K_PIPE_DEFINE(test_pipe, 256, 4);

static void test_write(const char * str) {
   LOG_INF("test_write(\"%s\")", str);

   size_t bytes_written;
   int err = k_pipe_put(&test_pipe, (void *) str, strlen(str), &bytes_written, 1, K_NO_WAIT);
   if (err || (strlen(str) != bytes_written)) {
      LOG_WRN("  test_write failed (%d, %d)", err, (int) bytes_written);
   }
}

static void test_read(char * data, size_t len) {
   int err;
   size_t bytes_read;

   do {
      bytes_read = 0;
      err = k_pipe_get(&test_pipe, data, len, &bytes_read, 1, K_MSEC(1));
      if ((err != 0) && (err != -EAGAIN)) {
         LOG_WRN("  test_read failed (%d)", (int) err);
      }
   } while (err != 0);
   LOG_INF("  Read OK (%d bytes, \"%s\")", (int) bytes_read, data);
}

static void test_thread_entry(void * x, void * y, void * z) {
   char data[100];
   while (1) {
      memset(data, 0, sizeof(data));
      test_read(data, sizeof(data));
   }
}
K_THREAD_DEFINE(test_thread_tid, 1024, test_thread_entry, NULL, NULL, NULL, 5, 0, 0);

void test_timer_handler(struct k_timer *dummy) {
   test_write("Hello World");
}
K_TIMER_DEFINE(test_timer, test_timer_handler, NULL);

void main(void) {
   k_timer_start(&test_timer, K_MSEC(300), K_MSEC(300));

   static const struct device * uart_dev = DEVICE_DT_GET(DT_NODELABEL(uart1));
   static uint8_t rx_buffer[50];
   int err = uart_rx_enable(uart_dev, rx_buffer, sizeof(rx_buffer), 50);
   __ASSERT_NO_MSG(err == 0);

   while (1) {
      k_sleep(K_MSEC(500));
   }
}
```

peter-mitsis, 2023-01-23:
I am able to reproduce it in a few seconds on the NRF52840 DK now without the need for uart_rx_enable(). (See below). 

However, if CONFIG_LOG=n, then I can't reproduce the problem.

Still digging.
```
/*
 * Copyright (c) 2016 Intel Corporation
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr/kernel.h>
#include <zephyr/irq_offload.h>

#include <string.h>
#include <stdio.h>

static void test_thread_entry(void * x, void * y, void * z);
K_THREAD_DEFINE(test_thread_tid, 1024, test_thread_entry, NULL, NULL, NULL,
                5, 0, 0);
K_PIPE_DEFINE(test_pipe, 256, 4);

static void test_write(const char * str)
{
        printk("test_write(\"%s\")\n", str);

        size_t bytes_written;
        int err = k_pipe_put(&test_pipe, (void *) str, strlen(str),
                             &bytes_written, 1, K_NO_WAIT);

#if 1
        if (err || (strlen(str) != bytes_written)) {
                printk("     ---  test_write failed (%d, %d)\n",
                        err, (int) bytes_written);
        }
#endif
}

static void test_read(char * data, size_t len)
{
        int err;
        size_t bytes_read;

        do {
               err = k_pipe_get(&test_pipe, data, len, &bytes_read,
                                 1, K_MSEC(10));
#if 1
                if ((err != 0) && (err != -EAGAIN)) {
                        printk("  test_read failed (%d)", (int) err);
#endif
                }
        } while (err != 0);

        printk("  Read OK (%d bytes, \"%s\")\n", (int) bytes_read, data);
}

static int my_counter = 0;

/* Function has to be called periodically from IRQ */
void test_write_irq(struct k_timer *unused)
{
        char myStr[50];

        sprintf(myStr, "Hello World %02d", my_counter);
        test_write(myStr);

        my_counter++;
        my_counter %= 100;
}

static void test_thread_entry(void * x, void * y, void * z)
{
        char data[100];

        printk("Test priority: %d\n", k_thread_priority_get(_current));

        while (1) {
                memset(data, 0, sizeof(data));
                test_read(data, sizeof(data));
        }
}

struct k_timer my_timer;

void main(void)
{
        k_timer_init(&my_timer, test_write_irq, NULL);
        k_timer_start(&my_timer, K_SECONDS(1), K_USEC(9950));

        printk("Main priority: %d\n", k_thread_priority_get(_current));
}

prj.conf
CONFIG_MP_MAX_NUM_CPUS=1
CONFIG_PIPES=y
CONFIG_PRINTK=y
CONFIG_TICKLESS_KERNEL=y
CONFIG_NULL_POINTER_EXCEPTION_DETECTION_MPU=y
CONFIG_EXTRA_EXCEPTION_INFO=y
CONFIG_LOG=y

```

peter-mitsis, 2023-01-23:
Although it was a low probability cause of error, I wanted to verify that the interrupt locking is being respected in the pipe code. That is, I wanted to verify that we did not have some weird esoteric case where a timer ISR was interrupting a pipe operation. With some not so fancy flag checking, I was able to confirm to a high enough level of satisfaction that that is not the case. This strongly suggests that there is an algorithmic flaw somewhere in the pipe code. Still digging.

arbrauns, 2023-01-26:
I'm also encountering issue 3 (a pipe fed from an ISR causes an unrelated pipe to wake up with a supposedly full buffer, even though nothing was actually written), #53634 does **not** fix it (at head commit 02c0f49). I have worked around it for now my replacing the ISR pipe with a `k_msgq`.

peter-mitsis, 2023-01-26:
I may have found the source of the instability.  Both the `z_impl_k_pipe_put()` and `pipe_get_internal()` routines obtain a pointer to the current thread's pipe descriptor `pipe_desc`, which is a structure embedded within the `k_thread` structure. For example, ...

`dest_desc = &_current->pipe_desc;`

This is fine when those routines get invoked at thread-level, but it is not safe to do so from the context of an ISR because this will refer to the thread that was interrupted.

Fixing ....

peter-mitsis, 2023-01-26:
I have updated #53634 to fix that Sporadic Data Access Violation error. Locally, on my own setup I am no longer able to reproduce the problem as described above after applying that commit.

m-wichmann, 2023-01-27:
Nice find! I tested the new changes and the problem seems to be fixed. I can't reproduce the errors anymore either. Not with the code from the issue description and neither with my original code. And this would also explain why unrelated pipes from other threads were also affected. Thanks for fixing this!

peter-mitsis, 2023-01-27:
Thank you for reporting it! I've also done a bit of a mini-retrospective to determine where I failed in my own personal development process that led to the bug being introduced. I think I have an idea, so hopefully I will be less likely to make the same kind of error again in the future. :)

stephanosio, 2023-02-09:
I have marked this issue as a release blocker for now since it is an quite serious kernel bug.

closed 2023-02-10 by stephanosio

JamesWhitlock, 2023-12-12:
It appears the first issue reported never had a fix committed before the issue was closed. We are still seeing this behaviour on NRF-Connect SDK 2.5.0 - Zephyr a768a05e6205e415564226543cee67559d15b736

mikk-leini-krakul, 2025-11-05:
Reporting similar issues with Cortex M7 and Zephyr v4.2 when using pipes within UART interrupt handler (dynamic interrupts enabled). When calling k_pipe_write I have three times got usage fault on this line:
https://github.com/zephyrproject-rtos/zephyr/blob/57ab471c3acd4d826d1dbbed13c3bfc2048e7792/kernel/pipe.c#L184
"len" value is 1, "written" is 1.
Zephyr error handler prints out "Illegal use of the EPSR".

And once I got bus fault here on this callback chain:
https://github.com/zephyrproject-rtos/zephyr/blob/57ab471c3acd4d826d1dbbed13c3bfc2048e7792/kernel/pipe.c#L215
https://github.com/zephyrproject-rtos/zephyr/blob/57ab471c3acd4d826d1dbbed13c3bfc2048e7792/include/zephyr/spinlock.h#L305
https://github.com/zephyrproject-rtos/zephyr/blob/57ab471c3acd4d826d1dbbed13c3bfc2048e7792/kernel/spinlock_validate.c#L25
Error message is "Imprecise data bus error".

PR #53634 (2023-02-10): kernel: Fix pipe ISR instability issues (#52812)
Fixes #52812 (well, most of it)

This series of commits is an attempt to fix the pipe instability issues raised in issue #52812 (the second and third issues). The issues that I have been able identify related to those boil down to improperly protecting the wait queue from within the pipe routines.

When the working list of pended threads (to read from or write to) is generated, sched_spinlock must be taken to protect the wait queue. Otherwise, an in-opportune timeout can lead to problems.

As there is a gap in time between when that list is gene [... cut]