zephyrproject-rtos/zephyr #52813: stm32h7: dsi: ltdc: clock: PLL3: clock not set up correctly or side effect

dariis, 2022-12-06:
# Bug or Not
<https://lists.zephyrproject.org/g/devel/topic/95406281#8795>
Thanks to @erwango for the fast response.

**Describe the bug**
Using the DSI host on stm32h747i-disco with a custom display needs additional initialisation of the PLL3.
I'm not sure if I'm hunting a ghost, because I still have other issues with the display (init sequence).

## Bad Case
**To Reproduce**
Vanilla `display_stm32_ltdc.c` Zephyr v3.2.0.
`stm32h747i_disco_m7.overlay`:
```dts
&pll3 {
	status = "okay";
	clocks = <&clk_hse>;
	div-m = <5>;
	mul-n = <132>;
	div-p = <2>;
	div-q = <2>;
	div-r = <24>; /* 27.5 MHz */
};
```
showing color bars with `HAL_DSI_PatternGeneratorStart(&hdsi, 0, 1)`

**Expected behavior**
Show color bars.

**Logs and console output**
Colors do not match.
[image]

## Better Case
**To Reproduce**
patched  `display_stm32_ltdc.c`
```diff
diff --git a/drivers/display/display_stm32_ltdc.c b/drivers/display/display_stm32_ltdc.c
index 840c276e28..04d387cf37 100644
--- a/drivers/display/display_stm32_ltdc.c
+++ b/drivers/display/display_stm32_ltdc.c
@@ -280,7 +280,18 @@ static int stm32_ltdc_init(const struct device *dev)
                LOG_ERR("clock control device not ready");
                return -ENODEV;
        }
-
+#if 1
+       RCC_PeriphCLKInitTypeDef  PeriphClkInitStruct;
+
+       PeriphClkInitStruct.PeriphClockSelection   = RCC_PERIPHCLK_LTDC;
+       PeriphClkInitStruct.PLL3.PLL3M      = DT_PROP(DT_NODELABEL(pll3), div_m);
+       PeriphClkInitStruct.PLL3.PLL3N      = DT_PROP(DT_NODELABEL(pll3), mul_n);
+       PeriphClkInitStruct.PLL3.PLL3P      = DT_PROP_OR(DT_NODELABEL(pll3), div_p, 1);
+       PeriphClkInitStruct.PLL3.PLL3Q      = DT_PROP_OR(DT_NODELABEL(pll3), div_q, 1);
+       PeriphClkInitStruct.PLL3.PLL3R      = DT_PROP_OR(DT_NODELABEL(pll3), div_r, 1);
+       PeriphClkInitStruct.PLL3.PLL3FRACN  = 0U;
+       HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct);
+#endif
        /* Turn on LTDC peripheral clock */
        err = clock_control_on(DEVICE_DT_GET(STM32_CLOCK_CONTROL_NODE),
                                (clock_control_subsys_t) &config->pclken);
```
same overlay and showing color bars as [above](#bad-case).

**Expected behavior**
Show color bars.

**Logs and console output**
stm32h7 DSI host experts will see that the R and B colors are flipped, but the colors match.
[image]

## Comments
**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 15.1
 - v3.2.0

**Additional context**
**RCC** register diff from gdb ` x/1024xw 0x58024400`:
```diff
< 0x58024420:     0x00000040      0x00000000      0x00520052      0x01ff0808
---
> 0x58024420:     0x00000040      0x00000000      0x00520052      0x01ff0bd8
```
Set  `RCC_PeriphCLKInitTypeDef  PeriphClkInitStruct = {0};` to zero gives the following diff:
```diff
< 0x58024420:     0x00000040      0x00000000      0x00520052      0x01ff0808
---
> 0x58024420:     0x00000040      0x00000000      0x00520052      0x01ff0108
```

rcc 0x2C: 0x01ff0808 = [Bad Case](#bad-case)

rcc 0x2C: 0x01ff0108, 0x01ff0bd8 = [Better Case](#better-case)

Could this be the problem?
Or is the problem more likely at the display init sequence?
Or at the LTDC and DSI init order?

---

I'm lost...
I appreciate every comment, critics and hints.

erwango, 2022-12-07:
Just a hint that will help you reading registers in gdb: https://github.com/1udo6arre/svd-tools.
It could also be used to dump the full register in good/bad cases and then compare dump files.

erwango, 2022-12-07:
@dariis Since this is not an actual zephyr bug (DSI not supported) I'll move this to discussion, but github remains a good place to discuss.
Then I'm not quit eclear on your set up DSI and LTDC are 2 differents IPs, so I'm a bit surprised to see that you're able to achieve somewhat functional DSI using LTDC driver. Am I missing something ?