zephyrproject-rtos/zephyr #52815: ESP32 not support bt_disable()

sam131208, 2022-12-06:
**Describe the bug**
  while call the bt_disable(), it will check the hci drviver close handler, and return the value -ENOTSUP.

  In many cases, our Bluetooth products will be far enough from the mobile phone, and at this distance forward or backward, Bluetooth is frequently between connection and disconnection. At some point, connectivity becomes impossible and Bluetooth needs to be reset to become reachable again.

Thanks!

hermabe, 2022-12-07:
There seems to be two things here:
 - A feature request for the ESP32 HCI driver to support `bt_hci_driver::close`
 - An issue where the stack is unable to handle disconnections properly. You should not have to disable and re-enable the bluetooth stack after a disconnection. If you are able to reproduce this or provide more details, please create a separate issue for that.

sam131208, 2022-12-08:
@hermabe

     You are right. we need support the bt_hci_driver::close.
     When the connection is lost, it cannot be reconnected. This situation is relatively easy to reproduce in ESP-IDF. However, it performs better in Zephyr and is difficult to reproduce. Therefore, resetting Bluetooth is a faster method.
      When this happens, it does not affect the connection with other phones. There are two ways to solve the problem of not being able to reconnect: 
      1. Reset the ESP32 and reinitialize the Bluetooth. 
      2. Delete the pairing on your phone to reconnect.

hermabe, 2022-12-12:
Ok, I'll change this issue to a enhancement and then you can create a separate issue for the bug.

sylvioalves, 2024-02-12:
Fixed with #66693.

closed 2024-02-12 by sylvioalves