zephyrproject-rtos/zephyr #52818: samples: subsys: usb: shell: sample.usbd.shell  fails - no output from console

nordic-piks, 2022-12-06:
**Describe the bug**
The samples/subsys/usb/shell/sample.usbd.shell fails - no output from console

Observed for
- nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T samples/subsys/usb/shell -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-12-06T00:20:04.774Z] [0mERROR   -[0m
[2022-12-06T00:20:04.774Z] [0m[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.1
 - Commit SHA or Version used: zephyr-v3.2.0-2251-g95d8943c69c

jfischer-no, 2022-12-06:
I can not reproduce it, but see also #52820

gopiotr, 2022-12-07:
@desowin yes, this is related with how Twister read output by `ser.readline()`. This causes, that it cannot read `uart:~$` because this expression is not ended with new line signs (`\r\n`).

You had nice idea with "poking" shell by sending this "help" to enforce sending more data what could allows Twister to read whole lines (with `uart:~$` inside). But you placed this `ser.write(b'help\n')` in wrong place. To make it possible to work in this way, you need:

1. Add to yaml file `console: harness` and expected regex as `uart:~` (without `$` sign):
```yaml
  sample.usbd.shell:
    platform_allow: nrf52840dk_nrf52840 frdm_k64f
    depends_on: usb_device
    tags: usb
    harness: console
    harness_config:
      type: one_line
      regex:
        - "uart:~"
```
2. Add following code:
```python
time.sleep(5)
ser.write(b'help\n')
```
immediately after line 611 in `handlers.py` file:
https://github.com/zephyrproject-rtos/zephyr/blob/dfc97f3f382f63b283931a02acd2bdbf785dfe64/scripts/pylib/twister/twisterlib/handlers.py#L610-L611
to get, something like:
```python
        ...
        if post_flash_script:
            self.run_custom_script(post_flash_script, 30)

        time.sleep(5)
        ser.write(b'help\n')

        if not flash_error:
            t.join(self.timeout)
        ...
```

And then you should observe that Twister parse this output properly and is able to pass this test/sample.

I know that above solution is not applicable now in Twister 1:1, but it can show/inspire somebody how challenge with testing `shell` module could be solved: by adding in Twister possibility to sending such "extra" command like `help`.

desowin, 2022-12-07:
Putting the serial port write hack after flashing in main twister thread, as opposed to serial monitor thread as I did, requires the sleep to be long enough for the software to boot up (instead of long enough for both flashing and boot up). I didn't configure expected regex and the `No Console Output(Timeout)` essentially is due to lack of (expected by default) ZTest test suite output (and not due to no output at all).

gopiotr, 2022-12-08:
To summarize:
1. Test `samples/subsys/usb/shell/sample.usbd.shell` fails because no specific harness is set, so Twister tries to use default harness "ZTest" (https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/harness.py#L254), which is used to parse output from tests basing on ZTest framework (which is not used in this sample). For example this harness expects that platform/board will print at the end `PROJECT EXECUTION SUCCESSFUL` sentence and other typical ZTest statements.
2. Thus, other type of harness should be used for samples. The most popular harness "console" is used for detecting specific regex in printed output. As @desowin observed - the problem with using this harness in this sample (`sample.usbd.shell`) is that, it do not print whole line (ended with `\r\n` signs) and so, it is impossible for Twister to parse it properly by this harness.
3. The conclusion is - at this moment Twister cannot be used to verify properly if this sample works properly after flashing it on platform. So, for Twister it can be build-only. This can be achieved directly by adding to `sample.yaml` register `build_only: true`, or indirectly, by adding register `harness: keyboard`. This second solution will cause Twister detect, that there is no implementation of such a harness and stop execution this sample on platform/board.
4. If you really care about have possibility to test such a samples (`shell`) by Twister, please create separate issue for this with description what the challenge is and proposition of how it could be solved or at least how it should work.

desowin, 2023-01-16:
The sample no longer fails after a01fbd07316f2547668a03fcd977e823ad4a0b6e (PR #52820). It does not fail, but does not execute either because there is no `keyboard` harness. This is really Twister limitation, not USB stack. Issue #52889 is supposed to track Twister ability to test shell.

closed 2023-01-16 by desowin