zephyrproject-rtos/zephyr #52825: Overflow in settime posix function

AndreyDodonov-EH, 2022-12-06:
**Describe the bug**
Overflow in settime posix function

This is a follow-up of https://github.com/zephyrproject-rtos/zephyr/issues/41111
and https://github.com/zephyrproject-rtos/zephyr/pull/41814/

In this PR gettime function was changed: https://github.com/zephyrproject-rtos/zephyr/pull/41814/
Settime uses same conversion and has same effect.
So API should be symmetrical.

AndreyDodonov-EH, 2022-12-06:
Obviously it will reduce precision, but it was already done anyway by changing gettime function.

nordic-krch, 2022-12-07:
> Obviously it will reduce precision, but it was already done anyway by changing gettime function.

It does not reduce the precision. It's just that calculation is performed on seconds and only remainder is in nanoseconds. Contrary to the approach where all is calculated in nanoseconds and that can result in overflow.

AndreyDodonov-EH, 2022-12-08:
> It does not reduce the precision. It's just that calculation is performed on seconds and only remainder is in nanoseconds. Contrary to the approach where all is calculated in nanoseconds and that [...]

But doesn't in this case remainder have milliseconds precision? (even though saved in ns member of the struct)
I'm talking about using ms in settime.

cfriedt, 2023-04-07:
@AndreyDodonov-EH - can you verify that this is also fixed by #52977

AndreyDodonov-EH, 2023-04-07:
> @AndreyDodonov-EH - can you verify that this is also fixed by #52977

Will check on Tuesday, thanks for updating!

AndreyDodonov-EH, 2023-04-12:
@NilsRuf-EH FYI

AndreyDodonov-EH, 2023-04-12:
@cfriedt 
Thanks for pinging!
We had a look at your PR, and it fixes the symmetry problem which I mention in this issue.

As for the settime overflow issue, it is addressed in https://github.com/zephyrproject-rtos/zephyr/commit/50c7c7b1e4027b08a594f108df5d62bce1fc229f

However, it depends there on an arbitrary rule-of-thumb constant `CONFIG_SYS_CLOCK_MAX_TIMEOUT_DAYS`, which is by default 365, which means device is expected to have maximum uptime of roughly a year.

cfriedt, 2023-04-12:
> However, it depends there on an arbitrary rule-of-thumb constant `CONFIG_SYS_CLOCK_MAX_TIMEOUT_DAYS`, which is by default 365, which means device is expected to have maximum uptime of roughly a year.

@AndreyDodonov-EH - I agree. I'm not a huge fan of the concept of `CONFIG_SYS _CLOCK_MAX_TIMEOUT_DAYS` either.

I feel like the better thing to do would be to specify either
* the number of bits in the counter (preferable)
* the counter maximum (less preferable)

cfriedt, 2023-06-01:
@AndreyDodonov-EH - I'm going to close this because it seems as though the issues have been resolved. 

If there are other related enhancement requests, please open up a new issue.

closed 2023-06-01 by cfriedt