zephyrproject-rtos/zephyr #52830: Annoying Slirp Message console output from qemu_x86 board target

mkettn, 2022-12-06:
**Describe the bug**
The easiest way to use Networking with QEMU is in user mode with slirp. here QEMU provides the (tcp|udp)/ip stack. Zephyr works just like a charme, however, the qemu version zephyr sdk ships litters the console output with `qemu-system-i386: Slirp: Failed to send packet, ret: -1` messages. As far as I could trace this output back, it does not originate from Zephyr RTOS but from QEMU itself and is false alarm. QEMU was able to send the packet. I can see the response from the server.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/net/dhcpv4_client/
2. mkdir boards
3. save file qemu_x86.conf here (see below)
4. mkdir build; cd build
5. cmake -GNinja -DBOARD=qemu\_x86
6. ninja run
7. See console output

**Expected behavior**
QEMU shall remain silent! there is no error. the ip packet gets transmitted by the qemu user mode just fine.

**Impact**
each ip packet results in one `qemu-system-i386: Slirp: Failed to send packet, ret: -1` line on the console output. debugging output of udp applications is annoying, while debugging ip packet heavy tcp traffic from multiple connections is tedious.

**Logs and console output**

```
$ ninja run                        
[1/183] Preparing syscall dependency handling

[2/183] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (/home/mark/zephyrproject/zephyr), build: zephyr-v3.2.0-1869-g9ebb9abab7b0
[168/183] Linking C executable zephyr/zephyr_pre0.elf

[172/183] Linking C executable zephyr/zephyr_pre1.elf

[182/183] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
             RAM:      184352 B         3 MB      5.86%
        IDT_LIST:          0 GB         2 KB      0.00%
[182/183] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)

iPXE (http://ipxe.org) 00:02.0 CA00 PCI2.10 PnP PMM+00392120+002F2120 CA00
                                                                               

Booting from ROM..
*** Booting Zephyr OS build zephyr-v3.2.0-1869-g9ebb9abab7b0 ***

[00:00:00.000,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
uart:~$ qemu-system-i386: Slirp: Failed to send packet, ret: -1
qemu-system-i386: Slirp: Failed to send packet, ret: -1
[00:00:07.010,000] <inf> net_dhcpv4: Received: 10.0.2.15
[00:00:07.010,000] <inf> net_dhcpv4_client_sample: Your address: 10.0.2.15
[00:00:07.010,000] <inf> net_dhcpv4_client_sample: Lease time: 86400 seconds
[00:00:07.010,000] <inf> net_dhcpv4_client_sample: Subnet: 255.255.255.0
[00:00:07.010,000] <inf> net_dhcpv4_client_sample: Router: 10.0.2.2
uart:~$ qemu-system-i386: Slirp: Failed to send packet, ret: -1
```

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain Zephyr SDK,
 - Version: 9ebb9abab7b0d662e0d5379367fb7f052959c545

**Additional context**
qemu_x86.conf:
```
CONFIG_FPU=y
CONFIG_PCIE=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_QEMU_ETHERNET=y
CONFIG_NET_QEMU_USER=y
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="10.0.2.2"
```
This config advises `ninja run` to use the QEMU-internal virtual network instead of the more complicated tap0 network (with ifconfig, dnsmasq and/or dhcpd) which has to be set up manually and is more error prone than the virtual QEMU-internal network.

rlubos, 2022-12-08:
I'm afraid I'm not able to help here, looks like QEMU  misconfiguration, we need some QEMU experts here.

@bwasim I see that you've originally introduced support for Qemu User Networking, perhaps you have some idea?

dcpleung, 2022-12-13:
Really need someone from networking to look into this.

Maybe it's the e1000 driver (either in Zephyr or QEMU)? @tbursztyka, any insights?

kartben, 2023-05-17:
Closing with the correct reason as this was improperly marked as "completed".

closed 2023-05-17 by kartben