zephyrproject-rtos/zephyr #52833: HARD FAULT on restarting sensor (lll.c:473)

caco3, 2022-12-06:
**Describe the bug**
We have a BLE application based on a nrf52.
With Zephyr `3.2.0` it runs stable, we didn't see any crashes for a long time.
Now we tried the latest commit `da23050` and we sporadically see crashes when we try to restart the sensor (either using `sys_reboot(SYS_REBOOT_COLD)` or `NVIC_SystemReset()`).

```
[00:00:06.817,962] \x1b[1;31m<err> peripherals: restart: \n\n*** RESTART ***\n
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:473\r\n
\tlll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 2044\r\n
[00:00:06.844,299] \x1b[1;31m<err> os: hard_fault: ***** HARD FAULT *****
[00:00:06.851,074] \x1b[1;31m<err> os: hard_fault:   Fault escalation (see below)
[00:00:06.858,551] \x1b[1;31m<err> os: hard_fault: ARCH_EXCEPT with reason 3\n

[00:00:06.865,722] \x1b[1;31m<err> os: esf_dump: r0/a1:  0x00000003  r1/a2:  0x20016698  r2/a3:  0x20016698
[00:00:06.875,579] \x1b[1;31m<err> os: esf_dump: r3/a4:  0x200169c8 r12/ip:  0x00000000 r14/lr:  0x00002c13
[00:00:06.885,406] \x1b[1;31m<err> os: esf_dump:  xpsr:  0x61000028
[00:00:06.891,693] \x1b[1;31m<err> os: esf_dump: Faulting instruction address (r15/pc): 0x00002c1e
[00:00:06.900,726] \x1b[1;31m<err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:06.909,912] \x1b[1;31m<err> os: z_fatal_error: Fault during interrupt handling\n

[00:00:06.917,877] \x1b[1;31m<err> os: z_fatal_error: Current thread: 0x20016698 (unknown)
[00:00:06.926,208] \x1b[1;31m<err> os: k_sys_fatal_error_handler: Halting system
```

Relevant code:
```c
oid restart(void) {   
    err = bt_conn_disconnect(bt_connection, BT_HCI_ERR_REMOTE_USER_TERM_CONN);

    log_panic(); // Disable log buffer, all messages get flushed immediately

    LOG_ERR("\n\n*** RESTART ***\n");

    sys_reboot(SYS_REBOOT_COLD);
    NVIC_SystemReset(); // Alternative call in case above does not work
    k_msleep(100);
}
```

Prj.conf:
```
## Optimization
CONFIG_SPEED_OPTIMIZATIONS=y
CONFIG_BT_CTLR_OPTIMIZE_FOR_SPEED=y
CONFIG_COMPILER_OPT="-Os"

## Assert Check
CONFIG_ASSERT=y

## Bluetooth
CONFIG_BT=y
CONFIG_BT_CTLR_LLCP_LOCAL_PROC_CTX_BUF_NUM=6
CONFIG_BT_AUTO_PHY_UPDATE=y
CONFIG_BT_AUTO_DATA_LEN_UPDATE=y

## BLE
CONFIG_HEAP_MEM_POOL_SIZE=2048
CONFIG_BT_BUF_ACL_RX_SIZE=254
CONFIG_BT_BUF_EVT_RX_SIZE=255
CONFIG_BT_BUF_CMD_TX_SIZE=255

CONFIG_BT_GATT_CLIENT=y
CONFIG_BT_ATT_PREPARE_COUNT=2
CONFIG_BT_CONN_TX_MAX=255
CONFIG_BT_L2CAP_TX_BUF_COUNT=10
CONFIG_BT_L2CAP_TX_MTU=247
CONFIG_BT_BUF_ACL_RX_SIZE=251
CONFIG_BT_L2CAP_DYNAMIC_CHANNEL=y
CONFIG_BT_CTLR_PHY_2M=y
CONFIG_BT_BUF_ACL_TX_COUNT=10
CONFIG_BT_BUF_ACL_TX_SIZE=251

CONFIG_BT_USER_PHY_UPDATE=y
CONFIG_BT_USER_DATA_LEN_UPDATE=y

CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
CONFIG_BT_CTLR_XTAL_THRESHOLD=500000

CONFIG_BT_PHY_UPDATE=y
CONFIG_BT_DATA_LEN_UPDATE=y

CONFIG_ZERO_LATENCY_IRQS=y
CONFIG_BT_CTLR_ZLI=y
CONFIG_BT_CTLR_ULL_HIGH_PRIO=2
CONFIG_BT_CTLR_ADVANCED_FEATURES=y

# Mcumgr SMP Service
CONFIG_MCUMGR_SMP_BT=y
CONFIG_MCUMGR_SMP_BT_AUTHEN=n

# MCUMgr Functionality
CONFIG_MCUMGR=y
CONFIG_MCUMGR_CMD_IMG_MGMT=y
CONFIG_MCUMGR_CMD_OS_MGMT=y
```

**To Reproduce**
-

**Expected behavior**
no assert

**Impact**
showstopper

**Logs and console output**
-
**Environment (please complete the following information):**
 - Target: nrf52840
 - OS: Linux (Ubuntu)
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA or Version used: v3.2.99 (da23050)

**Additional context**
-

cvinayak, 2022-12-07:
The encountered assertion check was added recently to catch skipped BLE radio events that could happen due to any externally introduced latencies.

Does this only happen on restart? Please check if waiting for `disconnected()` helps. Or alternately, you can use `bt_disable()` interface before restart.

cvinayak, 2022-12-08:
Thank you for verifying the solution.

closed 2022-12-08 by cvinayak

cvinayak, 2023-02-23:
As you use `CONFIG_BT_CTLR_ZLI=y`, this may be interesting: https://github.com/zephyrproject-rtos/zephyr/pull/55114

caco3, 2023-06-12:
@cvinayak Thanks for the notification.
I retested now with [Zephyr 3.4-RC2](https://github.com/zephyrproject-rtos/zephyr/releases/tag/v3.4.0-rc2) where I expect https://github.com/zephyrproject-rtos/zephyr/pull/55114 to be contained.

How ever I am still able to provoke this error sporadically:
```
ASSERTION FAIL [0] @ ZEPHYR_BASE/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:473
        lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[00:27:40.987,335] <err> os: ***** HARD FAULT *****
[00:27:40.987,335] <err> os:   Fault escalation (see below)
[00:27:40.987,365] <err> os: ARCH_EXCEPT with reason 3

[00:27:40.987,396] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000000
[00:27:40.987,426] <err> os: r3/a4:  0x00000000 r12/ip:  0x20023a34 r14/lr:  0x0000361d
[00:27:40.987,457] <err> os:  xpsr:  0x41000011
[00:27:40.987,487] <err> os: Faulting instruction address (r15/pc): 0x00003628
[00:27:40.987,518] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:27:40.987,548] <err> os: Fault during interrupt handling

[00:27:40.987,609] <err> os: Current thread: 0x200168f8 (unknown)
[00:27:41.119,689] <err> os: Halting system
```

The way I am able provoke it sporadically (after 30 minutes or more) is to have my measurement system active (meaning I am sending notifications at a high sampling rate of 333 notifications/second) and adittionally running GATT read/writes as fast as possible.

cvinayak, 2023-06-12:
The assertion you see could very well be due to any latencies introduced from any user/application ISRs in ZLI priority.

Is this assertion be reproducible without any user ISRs in ZLI (or highest priority level)?

cvinayak, 2023-06-13:
I hope you have the default peripheral interrupt level value greater than 2? i.e. use lower priorities for other SoC peripherals used in your application by having the below

`#define NRF_DEFAULT_IRQ_PRIORITY 5`

inside your application folder in `dts/arm/nordic/override.dtsi`  file.

In spite these, you still get the assertion, then the value of `EVENT_OVERHEAD_START_US` has to be increased from `275`  to `306` or higher as reported by the assertion when it happens.

caco3, 2023-06-13:
Yes, I have set `#define NRF_DEFAULT_IRQ_PRIORITY 5` as needed.

So should I simply increase the define in  https://github.com/zephyrproject-rtos/zephyr/blob/v3.4.0-rc3/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_vendor.h#L39?

What is the disadventage on setting it to high?

SInc eit is very difficult to provoke this issue, it also wil lbe hard to check that the set value is sufficient.

I also saw your suggestion in [52491 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/52491#issuecomment-1482223082) to simply disable the ASSERT. Would this be a safer approach?

cvinayak, 2023-06-14:
> What is the disadventage on setting it to high?

Higher value reduces radio utilization. I.e. each radio event has this possible CPU use overhead margin between next radio use.

> SInc eit is very difficult to provoke this issue, it also wil lbe hard to check that the set value is sufficient.
>
> I also saw your suggestion in [52491 [...]

Since you mention that it is difficult to reproduce, yes, it is safe to disable the assertion check.

caco3, 2023-06-15:
> Since you mention that it is difficult to reproduce, yes, it is safe to disable the assertion check.

Thanks for the confirmation.

I also now have a test run where I see that changing it to a LOG message does not harm our applications functionality (No BLE disconnects or reduced throughput, no crashes).

May I suggest to provide a Kconfig to disable this assert or to adjust the value?

For me I always see the same pattern:
```

[00:54:17.256,286] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:07:18.482,543] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:07:19.997,589] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:07:25.157,745] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:38.122,833] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:40.537,902] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:42.187,957] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:42.892,974] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:43.102,996] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:44.198,028] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:44.963,043] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:45.923,095] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:46.703,094] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:46.973,114] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:47.198,120] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:48.458,160] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:13:49.028,167] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:20:11.728,576] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:20:12.778,594] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:20:13.738,616] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:20:13.813,629] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:20:13.918,640] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:20:15.088,684] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:20:15.613,677] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:20:23.668,945] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:26:42.634,216] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:26:43.234,222] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:26:43.474,243] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:26:43.729,248] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:26:44.149,261] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:26:45.034,301] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:26:45.409,301] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[01:26:46.654,327] <err> lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
```

cvinayak, 2023-06-15:
> May I suggest to provide a Kconfig to disable this assert or to adjust the value?

I have some work in progress in a local repo, I will see what I can do:
1. I want to move the assertion to individual state/role c source file instead of the current lll.c
2. Have a Kconfig to disable assertion and permit radio event skipping

But this will take some time before I have a clean PR to upstream.

cvinayak, 2023-06-28:
@caco3 hope you can check for any regressions by this PR: https://github.com/zephyrproject-rtos/zephyr/pull/59801