zephyrproject-rtos/zephyr #52857: Adafruit WINC1500 Wifi Shield doesn't work on nRF528XX

aectaan, 2022-12-07:
Three setups: `nucleo_wb55`, `nrf52dk_nrf52832`, `nrf52840dk_nrf52840`. 
`west build -p -b nrf52840dk_nrf52840 -s ../zephyr/samples/net/wifi/ -- -DSHIELD=adafruit_winc1500 && west flash` to reproduce

Looks like corrupted SPI communication, captures from Saleae attached.

**Expected behavior**
Same as with ST

**Impact**
Breaking

Linux 22.04, Zephyr 3.2.0
[ATWINC1500_dumps.zip](https://github.com/zephyrproject-rtos/zephyr/files/10177114/ATWINC1500_dumps.zip)

diocorreia, 2022-12-07:
Have you updated the WINC1500 with the correct firmware (v3.35.1)? Unfortunately, WINC1500 FW changes a lot between releases and you must have corresponding versions between your driver and firmware. You can find the correct firmware in Microchip's website (https://www.microchip.com/en-us/tools-resources/archives/avr-sam-mcus) . Look for the ASF v3.35.1, the example is in /common/components/wifi/winc1500/firmware_update_project/firmware

aectaan, 2022-12-08:
Thank you, I'll try it today. But it doesn't describe why behavior differs between two platforms. BTW Zephyr's winc1500 driver version is quite old (19.5.2) and the latest microchip version is 19.7.7. Right now it looks like SPI problem in nordic
```
(APP)(INFO)Chip ID 1502b1
(APP)(INFO)DriverVerInfo: 0x13521352
(APP)(INFO)Firmware ver   : 0.0.0 Svnrev 0
(APP)(INFO)Firmware Build  Time 
(APP)(INFO)Firmware Min driver ver : 0.0.0
(APP)(INFO)Driver ver: 19.5.2
(APP)(INFO)Driver built at Dec  7 2022	17:22:16
*** Booting Zephyr OS build zephyr-v3.2.0  ***
```

aectaan, 2022-12-08:
Looks like nrf sends additional bit to MOSI after last 0xCA byte.
NRF:
[image]
ST:
[image]

aectaan, 2022-12-09:
Thank you for your help!
Unfortunatelly UART update failed. And I don't have any SAM boards nearby to check recommended way

diocorreia, 2022-12-09:
It is possible to update WINC1500 without a SAM board. For that, you need to call microchip's driver API directly (you can also temporarily modify zephyr's driver) . Call nm_bsp_reset() and prevent any other driver "subsystem" to initilize. I recommend you to put an infinite busy wait, after nm_bsp_reset() is called. After this you just need to run microchip's upgrading script:

.\firmware\winc_programmer_uarte.exe -p \\.\COMx -d winc1500 -i firmware\m2m_image_3A0.bin -if prog -e -w -r -pfw programmer_firmware\release3A0\programmer_release_text.bin

aectaan, 2022-12-12:
Unfortunatelly could not open UART. Tried with good CP2102, connected with short wires

aectaan, 2022-12-12:
Successfully updated, but same errors on spi communications is present

aectaan, 2022-12-12:
Another guess: D/C M1448 on module. Does it mean 48 week of 2014? May my issue be caused by kind of old HW revision?

aectaan, 2022-12-13:
[image]
nRF
[image]
ST

It seems like clock wasn't stopped between first and second MISO bytes on nRF. May it cause problem?

aectaan, 2022-12-13:
I tried:
* different SPI channels
* different pins
* bitbang
What's wrong with setup? It isn't my first SPI device with Nordic or Zephyr, but right now it looks frustrating

anangl, 2022-12-14:
@aectaan Can you try using the H0H1 drive mode for SPI pins, i.e. using an overlay with the following content:
```
&arduino_spi {
	group1 {
		nordic,drive-mode = <NRF_DRIVE_H0H1>;
	};
};
```
or adding that property directly in [nrf52840dk_nrf52840-pinctrl.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840-pinctrl.dtsi#L174), if this could help?
Although the test with SPI bitbang seems to be indicating that this is not the case, but I don't have any other ideas at the moment. And I don't have this shield to try it myself.

aectaan, 2022-12-14:
@anangl It helped! But why? Because without this option MOSI packets were the same!

anangl, 2022-12-14:
> @anangl It helped!

Great!

> Because without this option MOSI packets were the same!

As seen by the logic analyzer. But not necessarily for the WINC1500 chip.

closed 2022-12-15 by aectaan