zephyrproject-rtos/zephyr #52868: ESP32 Wifi driver returns EIO (-5) if connecting without a sleep sometime before calling

dean-at-evos, 2022-12-07:
Using ESP32-C3, although I suspect this could occur with any ESP32 as they all appear to use the same driver, zephyrproject/zephyr/drivers/wifi/esp32/src/esp_wifi_drv.c.

Calling this function from main at start up fails the connect, unless you have slept (or yielded) to allow the driver's handler thread to run;
```c
#define WIFI DT_NODELABEL(wifi)
const struct device *const wifi_dev = DEVICE_DT_GET(WIFI);

#define MY_SSID "your_ssid"
#define MY_PSK "your_psk"

int iStartWiFi() {
	int ret = 0;

	if(!device_is_ready(wifi_dev)) { 
		LOG_ERR("WiFI device not ready");
		return -ENODEV;
	}

	//Set up call backs for net events
	net_mgmt_init_event_callback(&con_discon, handler_cb, NET_EVENT_WIFI_CONNECT_RESULT | NET_EVENT_WIFI_DISCONNECT_RESULT);
	net_mgmt_init_event_callback(&ipV4_cb, handler_cb, NET_EVENT_IPV4_DHCP_START | NET_EVENT_IPV4_DHCP_BOUND);
	net_mgmt_add_event_callback(&con_discon);
	net_mgmt_add_event_callback(&ipV4_cb);

	//Fill out the wifi config struct
	struct wifi_connect_req_params wifi_config;
	//Esp driver only seems to use these params...
	wifi_config.security = WIFI_SECURITY_TYPE_PSK;
	wifi_config.ssid = MY_SSID;
	wifi_config.ssid_length = MIN(MY_SSID, WIFI_SSID_MAX_LEN);
	wifi_config.psk = MY_PSK;
	wifi_config.psk_length = MIN(MY_PSK, WIFI_PSK_MAX_LEN);
	
	//Get the api for the wifi dev
	const struct net_wifi_mgmt_offload *wifi_api = wifi_dev->api;

	LOG_DBG("Trying to connect to %s:%s", wifi_config.ssid, wifi_config.psk);

	//Try and connect....
        k_msleep(1) //Will possibly fail with EIO if no sleep here
	ret = wifi_api->connect(wifi_dev, &wifi_config);
	
	if (ret)
	{
		event_cb = NULL;
		LOG_ERR("WiFi Connect failed %d", ret);
	}

	return (ret);
}
```
It seems to fail as when the call to connect reaches here;
```c
static int esp32_wifi_connect(const struct device *dev,
			    struct wifi_connect_req_params *params)
////////////////////->
	if (data->state != ESP32_STA_STARTED) {
		LOG_ERR("Wi-Fi not in station mode");
		wifi_mgmt_raise_connect_result_event(esp32_wifi_iface, -1);
		return -EIO;
	}
```
Unless the main thread has slept or yielded, this section hasn't run;
```c
static void esp_wifi_event_task(void)
////////////////////->
		case ESP32_WIFI_EVENT_STA_START:
			esp32_data.state = ESP32_STA_STARTED;
			net_eth_carrier_on(esp32_wifi_iface);
			break;
```
Hope this makes sense to you gurus out there.

sylvioalves, 2023-01-23:
@dean-at-evos, would you try using proper `net_mgmt` API for this? `net_mgmt(NET_REQUEST_WIFI_CONNECT..`?

dean-at-evos, 2023-01-23:
I will come back to that code and try that. I couldn't find a good example of the 'right' way to setup the connection, so I just hacked away :) I'll update this issue when I get back to it. Thanks for your time @sylvioalves

dean-at-evos, 2023-03-18:
@sylvioalves Have got back to this project now and tried the ```net_mgmt``` api. Still shows the same behaviour when trying to connect from within the main loop if the system workqueue hasn't had a chance to run.
The work around for me in the end was to add something like...
```c
int sleep_cnt = 0;
while (!net_if_is_up(iface) && (sleep_cnt < 10)) {
        k_sleep(K_MSEC(100));
	sleep_cnt++;
	LOG_DBG("Wait net iface %d", sleep_cnt);
}

if (!net_if_is_up(iface)) {
	LOG_ERR("WiFi iface not up");
	return -ENOTCONN;
}
```
before connecting. Not a big deal but I guess a time sucker trying to work out why I could connect using the sample code but not from my app.....
If you suggest a good place to document this, or an approach to fix it I'll submit a PR. Otherwise, I'll close the issue.

nashif, 2023-05-30:
@sylvioalves  Can you please take another look at this?

closed 2023-08-13 by github-actions[bot]