zephyrproject-rtos/zephyr #52872: Logging to USB CDC ACM limited to very low rate

xyzzy42, 2022-12-08:
**Describe the bug**
Logging, using the Zephyr logging system and the log_backend_uart, is limited to about 2200 bytes/sec when the uart is a USB CDC ACM UART.  This is quite slow.  USB should be capable for much higher speed.

Logging to the shell_uart_backend, with the shell also on a USB CDC ACM uart, isn't limited to this low speed.

It's specific to the interaction between the USB CDC ACM uart and the logging uart backend.

**To Reproduce**
Steps to reproduce the behavior:
Log more than 2200 bytes/sec continuously to the backend log_backend_uart with a USB CDC ACM uart.  Eventually the logging system's buffer will fill and messages will start getting dropped.

**Expected behavior**
The logging system should be able to achieve a much higher data rate before dropping messages.

**Impact**
Logging over USB CDC ACM is very limited.  This is especially bad for devices which do not have another logging interface.

**Logs and console output**
USB captures from log_backend_uart and shell_uart_backend.
[usb_captures.zip](https://github.com/zephyrproject-rtos/zephyr/files/10181593/usb_captures.zip)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: GNU
 - Commit SHA or Version used: b05b8ad63acf30e08a570ce553831a06db27a2e0
 - Nordic NRF52840

**Additional context**
The USB URBs will contain at exactly one byte of data with log_backend_uart.  With shell_uart_backend, the URBs contain more data.

The issue appears to be a compounding limitations in cdc_acm and log_backend_uart.
* log_backend_uart will only write one byte a time with uart_poll_out() unless the async UART API is enabled.
* cdc_acm does not support the async UART API, only the polling and interrupt-driven UART APIs.
* log_backend_uart does not support the interrupt-driven UART API.
* cdc_acm does not appear to allow merging multiple bytes written with uart_poll_out() into a single URB.

jfischer-no, 2022-12-08:
This is not a bug. Here the same answer as on discord: "Everything that falls back to uart_poll_out is very slow due the nature of the UART API, for USB device stack it means, write one character -> trigger transfer, regardless of new or old stack."

xyzzy42, 2022-12-08:
It's not a bug that the logging system is limited to 2.2 kB/sec over 12 Mbit USB?  That makes it nearly useless for anything but occasional warning messages.

jfischer-no, 2022-12-08:
> It's not a bug that the logging system is limited to 2.2 kB/sec over 12 Mbit USB? That makes it nearly useless for anything but occasional warning messages.

You claim that it is a bug in USB support.
You also noticed that it behaves as expected by UART polling API.
I told you that it behaves as expected by UART polling API and is not a bug.
Further discussion with your bug description above is pointless.
You are welcome to open an enhancement issue for logging subsystem.

jfischer-no, 2022-12-14:
The description of the issue does not fit an enhancement at all and it remains closed.

>Further discussion with your bug description above is pointless.
You are welcome to open an enhancement issue for logging subsystem.

closed 2022-12-14 by jfischer-no

nordicjm, 2022-12-14:
> The description of the issue does not fit an enhancement at all and it remains closed.

@jfischer-no this literally has a fix that was submitted here: https://github.com/zephyrproject-rtos/zephyr/pull/53003

Hence why I re-opened it because it fixes this person's issue

xyzzy42, 2022-12-14:
The interrupt-driven interface of the CDC ACM driver stops the transfer when no host is connected, while the polling interface will drop data without a host.  I couldn't find any justification for this difference in behavior in the UART APIs.

It gets in the way when trying to use interrupt-driven mode to increase speed.

jfischer-no, 2022-12-14:
> The interrupt-driven interface of the CDC ACM driver stops the transfer when no host is connected, while the polling interface will drop data without a host. I couldn't find any justification for this difference in behavior in the UART APIs.

Yes, poll_out does not block, this is the result of constant complaints from users. Otherwise device would mostly not start (depending on the context from which first call comes).

> It gets in the way when trying to use interrupt-driven mode to increase speed.

Again, please open a specific enhancement issue to track this.

PR #53003 (2022-12-14): usb: device: cdc_acm: send more than 1 byte in poll out
Schedule CDC ACM tx work to happen 1 ms after first unprocessed data write. This gives enough leeway for the task writing to the queue to write more than 1 byte before USB workqueue preempts and schedules IN transfer. Sending more than 1 byte at a time increases data throughput because transaction overhead remains the same regardless of packet size.

Prior to this change, virtually every IN transaction carried only one character when CDC ACM was used as a console.

Related to #52981
Fixes #52872