zephyrproject-rtos/zephyr #52878: Bluetooth: Unable use native_posix with shell demo

LingaoM, 2022-12-08:
**Describe the bug**
Hi, i'm use latest main branch,

`west build -b native_posix zephyr/tests/bluetooth/shell -p auto`

`sudo ./build/zephyr/zephyr.exe --bt-dev=hci0`

No boot information at all.
[image]

nordicjm, 2022-12-08:
Did you connect a serial utility to `/dev/pts/18` as the output indicated?

LingaoM, 2022-12-08:
yeah，no response at all

nordicjm, 2022-12-08:
Works fine here.
```
uart connected to pseudotty: /dev/pts/1
```
and
```
uart:~$ 
  bt       clear    device   devmem   flash    gatt     help     history
  hrs      iso      kernel   l2cap    log      resize   shell    stats
uart:~$ kernel 
  cycles     stacks     threads    uptime     version    sleep      log-level
uart:~$ kernel 
  cycles     stacks     threads    uptime     version    sleep      log-level
uart:~$ kernel version 
Zephyr version 3.2.99
uart:~$ 
```

LingaoM, 2022-12-08:
please type: bt init

LingaoM, 2022-12-08:
Confirm my problem. Sorry to bother you.

closed 2022-12-08 by LingaoM