zephyrproject-rtos/zephyr #52903: [RFC] Uart backend for ipc service

Riphiphip, 2022-12-08:
## Introduction
I propose a UART-based backend for the [IPC service](https://docs.zephyrproject.org/2.7.0/samples/subsys/ipc/ipc_service/README.html) API to provide low-boilerplate inter-MCU communication. In particular this would allow the use of the [Event Manager Proxy Library](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/libraries/others/event_manager_proxy.html) from nRF Connect SDK to more easily create event-based multi-MCU applications on devices supported by nCS.

A quick-and-dirty proof of concept can be found here [https://github.com/Riphiphip/distributed-events-development-app](https://github.com/Riphiphip/distributed-events-development-app).

### Problem description

Creating solutions that requires multiple MCUs to work together is quite common. Setting up that communication often requires a lot of boilerplate code that revolves around keeping track of busses, accumulating data, juggling buffers etc. This proposal aims to slightly simplify that process by allowing developers to leverage the already existing IPC Service API to send data between processes located on different MCUs.

My particular choice of proposal comes from my experience with developing applications for the nRF9160 DK using nRF Connect SDK. In particular I have worked on several applications that uses the [Application Event Manager](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/libraries/others/app_event_manager.html) which is a framework for developing event-based applications. A common pattern I fell into while working on these apps was that I created a module that listened to events, decoded their data, transferred the data over uart using some custom data structure, read and decoded the data on the other MCU, and submitted a new event there. This workflow was implemented anew for every new event that needed to be shared between the MCUs. After some digging, it turned out that this problem had already been solved for multi-core MCUs through the [Event Manager Proxy](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/libraries/others/event_manager_proxy.html) library, which serializes, transmits, and re-emits any arbitrary event. This library used the IPC service for transmission, so I figured that the easiest and most robust way to get this library to work was to implement an inter-MCU IPC service backend.

### Proposed change
Implement an IPC service backend that uses uart to transfer data.

## Detailed RFC
### Proposed change (Detailed)
Implementing an IPC Service backend comes down to implementing a device driver. This driver will take control of a uart device where it will accumulate incoming data and transmit outgoing data. When all data has been received it will be transmitted in full to the IPC Service endpoint. To properly handle this, a mechanism for knowing how much data is going to be received is required. My previously mentioned POC solved this by splitting the message into fixed-size packets which were transferred in sequence. This works fine, but has a quite large data and transmission time overhead when sending small amounts of data. A better choice might be to send the data as a solid block prefixed by a header that contains information about how much data is being sent. It could also be possible to use the UART timeout mechanism to handle this.

Sending data should be pretty straight forward. Generating data in a specific format is usually easier than interpreting it. One thing that might be worth noting is that the IPC Service API seems to expect a blocking send. This can either be solved by using a blocking uart tx or some other synchronization mechanism on top of the ASYNC API.

### Dependencies
Adding this backend should not interfere with any other systems if not explicitly enabled by the application developer. When in use it will inevitably take control over a uart device which is then rendered unavailable for the rest of the application. The required rx and tx buffers will also consume a decent amount of memory.

### Concerns and Unresolved Questions

#### Is this useful to zephyr?
This proposal sprung up from a problem that appeared outside of the "pure" zephyr ecosystem and as such might not be relevant for zephyr itself. On the other hand it provides an abstraction that extends the functionality of a 3D party library without interfering with systems that don't wish to use it.

#### Does this already exist?
From what I've been able to tell there are no inter-MCU IPC Service backends, but I have misinterpreted the zephyr documentation before. 

#### Is this a good fit?
So far, the only IPC Service backends I've found are made for intra-IC communication. There might be a good reason for this that I have yet to think about.

## Alternatives
Since my main motivation for this change is to make the Event Manager Proxy more user-friendly, it might make more sense for this change to be implemented in the nRF Connect SDK, but since it doesn't require any other nCS specific components to be used, I figured that it would be a good idea to see if it could fit in Zephyr.

nordicjm, 2022-12-12:
CC @carlocaione @arnopo

carlocaione, 2022-12-12:
I'm not against this but I wonder if someone already tried to solve this because we want to avoid trying again to re-invent the wheel. @arnopo am I crazy or I see something like this implemented on top of OpenAMP?

arnopo, 2022-12-12:
I made a demo in ELCE 2018 based on an IPC over UART backend between Linux and Zephyr
On zephyr side it was directly implemented in a sample:

> STM32MP1 demo that demonstrates RPMsg over a UART serial link. In this example 3 RPMsg channels are created:
> 
> "rpmsg-button": creation of a /dev/input device on Linux side to get event
> "rpmsg-vl5310x": creation of an IIO device on Linux to read the rpmsg-vl5310x distance information
> "rpmsg-tty-channel" creation of a /dev/ttyRPMSG0 to stop the firmware on the L457 IOT board ( release of the rpmsg channels).
> Zephyr branch: https://github.com/arnopo/zephyr_stm32mp1/tree/RPMSG_o_UART_demo Linux branch: https://github.com/arnopo/linux/tree/ELCE_demos
> 
> Readme: https://github.com/arnopo/zephyr_stm32mp1/tree/RPMSG_o_UART_demo/samples/subsys/ipc/openamp_o_uart
> 

It might help you...

carlocaione, 2022-12-12:
> I made a demo in ELCE 2018 based on an IPC over UART backend between Linux and Zephyr On zephyr side it was directly implemented in a sample:

Ok, I'm not crazy :)

arnopo, 2022-12-12:
> I'm not against this but I wonder if someone already tried to solve this because we want to avoid trying again to re-invent the wheel. @arnopo am I crazy or I see something like this implemented on top of OpenAMP?
No UART backend implemented in openAMP, only a demo

Riphiphip, 2023-01-10:
> I made a demo in ELCE 2018 based on an IPC over UART backend between Linux and Zephyr On zephyr side it was directly implemented in a sample:
>...
>>Readme: https://github.com/arnopo/zephyr_stm32mp1/tree/RPMSG_o_UART_demo/samples/subsys/ipc/openamp_o_uart

In the readme for that sample you mentioned that the clean way to use RPMsg over uart would be to implement a uart backend as a part of the OpenAMP library. I am not too familiar with either RPMsg or OpenAMP, but from what I can tell that backend would need to interface directly with the zephyr-specific uart drivers. If that is not an issue, or if I have misunderstood something, making an RPMsg backend instead of an IPC service backend should still solve my initial problem and perhaps be a little more general.

Sorry for taking so long to respond. I just got back from vacation.

arnopo, 2023-01-12:
> In the readme for that demo you mentioned that the clean way to use RPMsg over uart would be to implement a uart backend as a part of the OpenAMP library. 

The aim of the OpenAMP library is to implement the rpmsg protocol whatever the platform and the OS. 
The user should always call the open-amp rpmsg user API. then depending on backend (vitio or UART) the message are transferred  to the remote processor.
To keep the rpmsg user API, you need  that a part of the rpmsg UART backend is implemented in the open-amp library and registered in place of the rpmsg-virtio.

If you implement your rpmsg UART backend in zephyr, you will fix the protocol for Zephyr to Zephyr IPC. And if protocol is extended you will also have to maintain it.
If you implement it in openAMP other OSes can also implements it and this would be more generic as could be reuse for other OSes.

That said, it depends on what you want to implement: a design dedicated to  your project or an  upstreamable version... Here I consider an upstreamable design.

>I am not too familiar with either RPMsg or OpenAMP, but from what I can tell such a backend would need to interface directly with the zephyr-specific uart drivers. Since OpenAMP is included as a module, it doesn't quite make sense to me that it should host code that is so tightly bound to Zephyr. If that is not an issue, or if I have misunderstood something, making an RPMsg backend instead of an IPC service backend should still solve my initial problem and perhaps be a little more general.

In term of split,  you could have the rpmsg over UART core implemented in the open-amp library. The zephyr would provide some ops on init for the UART primitives such as uart_read() uart_write().
Something similar to what we have implemented in Linux for the demo:
https://github.com/arnopo/linux/commit/c244a0b0360d0e520a01e9a9e981f1363dbdaf11

Riphiphip, 2023-01-17:
Do you have any suggestions for which abstractions could be useful as a base for the OpenAMP side of the backend? From what I can tell, libmetal has the concept of a bus, but it does not seem to have generic send/receive capabilities. It could also be possible to use an I/O region to model UART, but that feels like a bit of a hack.

maxd-nordic, 2024-02-12:
I think this would be very useful! The problem of inter-chip IPC crops up again and again here.
What can we do to make this happen?