zephyrproject-rtos/zephyr #52907: Bluetooth: Add functionality for bt_disable for off-chip HCI drivers

sem1-nordic, 2022-12-08:
**Is your feature request related to a problem? Please describe.**
At the moment bt_disable cannot be used with rpmsg, spi, uart etc. 

**Describe the solution you'd like**
Currently bt_hci_driver.open just sets up communication for off-chip HCI drivers, this would also have to also set up the controller on this command. A bt_hci_driver.close functionality would have to be added that tears down the controller.

pawelosyp-hid, 2023-03-14:
In my project I have observed the same issue. Is there any update?

swojo, 2023-03-17:
We're facing the same issue as well.

jori-nordic, 2023-03-20:
I was doing something like this in https://github.com/zephyrproject-rtos/zephyr/pull/55648 .
The building blocks are here, it just needs implementation (has to be different for every driver).

PR #56113 (2023-09-29): Bluetooth: rpmsg: bt_rpmsg_close() implementation 
Added bt_rpmsg_close() implementation, any comments appreciated, addressing #52907 
In short, function invokes reset CMD on network core, cleans stuff related to IPC and puts network core in reset.
Checking for freed endpoints changed due to situation I observed: when network core creates two endpoints, the IPC on the app core has listed endpoints with names even if they are not bounded.

Fixes #52907
closed 2023-09-29 by carlescufi