zephyrproject-rtos/zephyr #52909: usb: usb don't work after switch from Zephyr 2.7.3 to 3.2.99 on i.MX RT1020

MikhailSiomin, 2022-12-08:
I have a proprietary board with a processor i.MX RT1020. After switch from Zephyr 2.7.3 to 3.2.99, USB-OTG1 CDC/ACM does not work.
Also the cdc_acm example does not work on this board.

Maybe I need to add some config?

**Environment:**
- OS: Windows 10 Pro (21H2)
- Toolchain Zephyr SDK 0.15.1
- Zephyr version 3.2.99 (commit a9ed11ab8dcd6b7f6f677036694f9b9fa5dba74d)

desowin, 2022-12-13:
What do you mean by "does not work"? Is the device enumerating at all? Does the device show in Device Manager? Are there any URBs captured when capturing newly connected devices with Wireshark/USBPcap?

DerekSnell, 2022-12-14:
Hi @MikhailSiomin ,
I tested the sample `samples/subsys/usb/cdc_acm` using commit a9ed11ab8dcd6b7f6f677036694f9b9fa5dba74d on the MIMXRT1020-EVK board, and this sample works fine for me, see details below.  I also tested with the latest commit in main (2d3365200c9bb0ebdd0c3dc95f0ac04ca8ff046d), and that commit works fine for me too.

Do you have a MIMXRT1020-EVK board to compare with?  Are you able to review the differences of board files of your custom board with this board?  Since the in-tree board has no issue with this sample, I am going to move your post here to a discussion, and we can continue to investigate why it does not work on your board.  

Below is the debug console output for me.  The RT1020 enumerates in Windows as a VCOM port, and connecting a terminal to that port, it echoes back the characters as expected.
```
*** Booting Zephyr OS build zephyr-v3.2.0-2387-ga9ed11ab8dcd ***
[00:00:00.000,396] <inf> cdc_acm_echo: Wait for DTR
[00:00:05.305,023] <inf> usb_cdc_acm: Device configured
[00:00:20.469,970] <inf> cdc_acm_echo: DTR set
[00:00:21.474,060] <inf> cdc_acm_echo: Baudrate detected: 115200
```

Best regards