zephyrproject-rtos/zephyr #52921: fs: util: add `mkdir -p` equivalent helper functions

mniestroj, 2022-12-09:
**Is your enhancement proposal related to a problem? Please describe.**
When working with files/directories in file system in a POSIX system, there is often a need to use `mkdir -p` to create all parent directories before creating desired file. The amount of invocations of `mkdir -p` in various shell scripts is a good advocate of how useful such command is. In Zephyr there is no shell scripting, but instead everything is coded directly in C. However, the need for `mkdir -p` equivalent still applies. One of such use cases is file storage backend for settings subsystem, which is discussed in PR's comment https://github.com/zephyrproject-rtos/zephyr/pull/52866#discussion_r1043324105.

**Describe the solution you'd like**
Add new "FS utils" layer on top of "FS" layer, which will implement high-level FS operations:
- ```mkdir -p /some/dir``` equivalent
- ```mkdir -p `dirname /some/dir/path` ``` equivalent (see https://github.com/zephyrproject-rtos/zephyr/pull/52866#discussion_r1043324105 for use case)

This is just an initial set of functionalities. However, this will grow over time to prevent Zephyr-based applications and Zephyr subsystems to implement common functionalities all over again.

**Describe alternatives you've considered**
Always implement common FS operations at the subsystem and/or application level.

ptg4, 2026-10-18:
Hi @de-nordic @mniestroj 

I'm part of a group of four students, and we are interested in taking this on as part of a six-week course project.
It doesn't seem like it's still being actively worked on, but we wanted to leave a message here just to double check.

lilithxxx, 2026-10-18:
Oh sorry @ptg4 I didn't see your message there but I just created a PR to address this: https://github.com/zephyrproject-rtos/zephyr/pull/122598

Edit: I didn't create the mkdir -p `dirname /some/dir/path`  equivalent part yet, maybe thats something you can work on?