zephyrproject-rtos/zephyr #52924: ESP32 get the build message "IRAM0 segment data does not fit."

sam131208, 2022-12-09:
**Describe the bug**
When updating to the latest version of Zephyr, and build my project get the message "IRAM0 segment data does not fit."
However, the branch before November 10 is normal. 

So, I compared the compilation information of this project "samples\subsys\mgmt\mcumgr\smp_svr" in the two branches and found that the latest version of the branch, iram0_0_seg , is 52KB less

 The project "samples\subsys\mgmt\mcumgr\smp_svr" build message:
The branch before November 10. 
Memory region         Used Size  Region Size  %age Used
     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          32 B         32 B    100.00%
             ROM:      345892 B    4194240 B      8.25%
     iram0_0_seg:       72356 B       128 KB     55.20%
     irom0_0_seg:      214884 B      3264 KB      6.43%
     dram0_0_seg:       37668 B     124580 B     30.24%
     dram0_1_seg:       71312 B     110032 B     64.81%
     drom0_0_seg:       29148 B    4194240 B      0.69%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         4 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%

the latest version of Zephyr.
Memory region         Used Size  Region Size  %age Used
     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          32 B         32 B    100.00%
             ROM:      344620 B    4194240 B      8.22%
     iram0_0_seg:       71016 B        76 KB     91.25%
     irom0_0_seg:      213612 B      3264 KB      6.39%
     dram0_0_seg:       33220 B     124580 B     26.67%
     dram0_1_seg:       72208 B     110032 B     65.62%
     drom0_0_seg:       27104 B    4194240 B      0.65%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         4 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%

**Logs and console output**
my project build message:
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map D:/zephyrproject/zephyr/samples/sam/ZooPondClock/build/zephyr/zephyr_pre0.map
cmd.exe /C "cd . && C:\Users\sam\zephyr-sdk-0.15.0\xtensa-espressif_esp32_zephyr-elf\bin\xtensa-espressif_esp32_zephyr-elf-gcc.exe  -gdwarf-4 @CMakeFiles\zephyr_pre0.rsp -o zephyr\zephyr_pre0.elf  && cmd.exe /C "cd /D D:\zephyrproject\zephyr\samples\sam\ZooPondClock\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E echo ""
c:/users/sam/zephyr-sdk-0.15.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.exe: zephyr\zephyr_pre0.elf section `text' will not fit in region `iram0_0_seg'
c:/users/sam/zephyr-sdk-0.15.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/users/sam/zephyr-sdk-0.15.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.1.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 22560 bytes

sam131208, 2022-12-13:
@sylvioalves  
   I redid it under linux to confirm that the compilation results were consistent.
  Under the “zephyrproject\zephyr\samples\subsys\mgmt\mcumgr\smp_svr”directory：
  west build -b esp32 -- -DCONFIG_ESP_HEAP_MEM_POOL_REGION_1_SIZE=8192  -DOVERLAY_CONFIG=overlay-bt.conf

  Here is all the compilation information:

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr
-- Found Python3: C:/Python38/python.exe (found suitable exact version "3.8.10") found components: Interpreter
-- Cache files will be written to: D:/zephyrproject/zephyr/.cache
-- Zephyr version: 3.2.99 (D:/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: esp32
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (C:/Users/sam/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (C:/Users/sam/zephyr-sdk-0.15.2)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: D:/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/dts.cmake
Parsing D:/zephyrproject/zephyr/Kconfig
Loaded configuration 'D:/zephyrproject/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration 'D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/prj.conf'
Merged configuration 'D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/overlay-bt.conf'
Configuration saved to 'D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/.config'
Kconfig header saved to 'D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/include/generated/autoconf.h'

warning: BT_CTLR_DATA_LENGTH_MAX (defined at subsys/bluetooth\controller/Kconfig:431) was assigned
the value '251' but got the value ''. Check these unsatisfied dependencies: BT_CTLR_DATA_LENGTH
(=n), BT_CTLR (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_CTLR_DATA_LENGTH_MAX and/or look up
BT_CTLR_DATA_LENGTH_MAX in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/sam/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: D:/zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build
-- west build: building application
[6/281] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (D:/zephyrproject/zephyr), build: zephyr-v3.2.0-2540-ge2742af6ed11
[271/281] Linking C executable zephyr\zephyr_pre0.elf

[275/281] Linking C executable zephyr\zephyr_pre1.elf

[281/281] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          32 B         32 B    100.00%
             ROM:      344620 B    4194240 B      8.22%
     iram0_0_seg:       71016 B        76 KB     91.25%
     irom0_0_seg:      213612 B      3264 KB      6.39%
     dram0_0_seg:       33220 B     124580 B     26.67%
     dram0_1_seg:       72208 B     110032 B     65.62%
     drom0_0_seg:       27104 B    4194240 B      0.65%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         4 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%

sam131208, 2022-12-13:
@sylvioalves 
  I'm more interested in why the new mcu_boot have fewer iram0_0_seg：
  iram0_0_seg:   72356 B  128KB    55.20%        //The branch before November 10.
  iram0_0_seg:   71016 B   76 KB    91.25%       //  latest version of Zephyr
  iram0_0_seg:   22932 B   76 KB    29.47%       //you build version

  As you can see, the region size of the old version of mcuboot compiles iram0_0_seg is 128KB。
  Of course, if my compilation result is the same as yours, it is also acceptable.
  I followed the Getting Started Guide（https://docs.zephyrproject.org/latest/develop/getting_started/index.html） to do it all over again, and the result of the compilation was still like this.

sam131208, 2022-12-14:
@sylvioalves 
  I checkout the branch Commits on Nov 29, 2022, build this project and get the region  size of iram0_0_seg is 128KB. 
  And the branch Commits on Nov 30, 2022, you have pull requests linker update then get the region  size of iram0_0_seg is 76KB .
   I guess, are you in a position to debug a new feature that modifies the local linker file? Perhaps, that's why your compilation results are different from mine?

Under is the information of Commits on Nov 29, 2022 and Commits on Nov 30.
 https://github.com/zephyrproject-rtos/zephyr/commits/main?after=17d3f3a5a05a0999c5b0f2d73a7aa7ad7ed9fef1+419&branch=main&qualified_name=refs%2Fheads%2Fmain

sylvioalves, 2022-12-14:
I see what you are referring now. MCUBoot uses lots of IRAM to make room for all the security stuff. With that in mind, the ESP32 `linker.ld` had wrong IRAM length prior the commit you mentioned above. Now it is fixed. It means that MCUBoot won't overlap ESP32 anymore.   
In case you are not using MCUboot security features, you can manually change IRAM length in MCUBoot to a smaller value and increase IRAM value in ESP32 linker, enabling you to compile that again.  We have been working out to make MCUBoot buildable in Zephyr's environment, which would optimize this matter by proper handling `CONFIG` x IRAM length.

sam131208, 2022-12-22:
@sylvioalves I've been testing compilation issues lately, including re-setting up Zephyr environments on other computers. Unfortunately, the compilation result has always been this iram0_0_seg: 71016 B 76 KB 91.25%。
  I don't know what to do to get your compilation result iram0_0_seg: 22932 B 76 KB 29.47%.

sylvioalves, 2023-01-23:
@sam131208 Do you still have this issue?

sam131208, 2023-01-23:
@sylvioalves       Yes, even if a new Zephyr environment is set up on many computers, it still doesn't solve this problem.

sam131208, 2023-03-01:
> 
I think I found the problem, when you compiled, you chose the serial port method, and I chose the Bluetooth method.
Please use this command “west build -b esp32 -- -DCONFIG_ESP_HEAP_MEM_POOL_REGION_1_SIZE=4096 -DOVERLAY_CONFIG=overlay-bt.conf” to compile.

closed 2023-04-28 by sam131208