zephyrproject-rtos/zephyr #52938: Generic TCP Client Module

bdesterBE, 2022-12-09:
**Is your feature request related to a problem? Please describe.**
Zephyr currently does not possess a generic TCP client code modules for use on applications. A code module for this would provide helper functions to aid in implementing simple TCP clients, similar to those of the C# TcpClient class (linked below).

**Describe the solution you'd like**
If it would be something of interest, I would like to propose to implement a C/C++ generic TCP client code module for Zephyr. This TCP client would be similar to the TCPClient class that exists for C#, and based off of the TCP drivers found in the echo_server and echo_client samples.

**Additional Context**
[C# TcpClient Class API](https://learn.microsoft.com/en-us/dotnet/api/system.net.sockets.tcpclient?view=net-7.0)
[Zephyr Socket Echo Server Sample](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_server/README.html)
[Zephyr Socket Echo Client Sample](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_client/README.html)

bdesterBE, 2022-12-12:
> You can certainly propose an API for that. I would suggest that the API would not be tied to TCP but support also UDP. For the applications it would be desired that TLS/DTLS is also supported.

Great, thank you for getting back to me. I will try submitting something then with your suggestions in mind.

I am a little unsure where it would be appropriate to put such an API within Zephyr. Any suggestions?

jukkar, 2022-12-13:
> where it would be appropriate to put such an API within Zephyr.

`subsys/net/lib/` directory contains already library code and this could be placed there.

jukkar, 2022-12-13:
cc: @rlubos

jukkar, 2024-02-08:
As I said earlier, patches are welcome but we do not really need this feature request for code reviews, so I am closing this one as there has not been activity for a long time.

closed 2024-02-08 by jukkar